grbl/grbl #791: Is possible to change kinematics?

owenlab, 2015-08-30:
Hi, does someone know if is possible to change de kinematics as it was for the corexy?
I would to try to use a grbl to drive a SCARA robot  at the start and it will be beautiful to reach to move also a Delta machine.
I searched a lot on web but I found some notice from someone that started to try adding these kinematics but no one show how to!

thaks a lot and sorry about my bad english

chamnit, 2015-08-31:
@owenlab : Unfortunately, no. Inverse kinematics are not possible in Grbl at this moment.

closed 2015-08-31 by chamnit

darkworks, 2020-09-24:
its 2020 any update on this ,  inverse kinematics is future , robotic arms are easy to build using 3d printer parts compare to cnc rails which are expensive so  gbrl need to evolve to support inverse kinematics