grbl/grbl #792: Probing error ("Bad number format")

thoralt, 2015-08-31:
I'm currently struggling with probing. While implementing probing functions inside grbl-panel, I've come across the following strange behavior:

1st case (without error): sending commands with quite a bit of time in between (individual button clicks)
* Reset, $X
* set position on current axis to 0 using `_serialPort.Write("G90G10L20P0X0" & vbLf)`: **OK**
* start probing using `_serialPort.Write("G38.2X-1F25" & vbLf)`: probing starts, **OK**

2nd case (with error): sending commands at once
* Reset, $X
* send `_serialPort.Write("G90G10L20P0X0" & vbLf & "G38.2X-1F25" & vbLf)`: **Bad number format**
* send again `_serialPort.Write("G90G10L20P0X0" & vbLf & "G38.2X-1F25" & vbLf)`: probing starts, **OK**

I have no clue why sending the commands the first time results in "Bad number format" and sending them again a moment later results in the expected motion.

I'm using grbl 0.9j, but 0.9i also had this issue. Am I using the gcode commands in a wrong way or is grbl not behaving as it should? Since the written data is much less than 127 bytes there should be no overflow in the receive buffer. The commands are separated by \n, so they should be executed separately. Any help is greatly appreciated.

thoralt, 2015-08-31:
For reference, this is the isolated code which reproduces the problem:

```
Imports System.IO.Ports
Imports System.Threading

Public Class frmMain

    Shared _serialPort As SerialPort
    Shared _continue As Boolean
    Dim readThread As Thread

    Private Sub btnOpen_Click(sender As Object, e As EventArgs) Handles btnOpen.Click
        _serialPort = New SerialPort()
        _serialPort.PortName = "COM12"
        _serialPort.BaudRate = 115200
        _serialPort.Parity = Parity.None
        _serialPort.DataBits = 8
        _serialPort.StopBits = StopBits.One
        _serialPort.Handshake = Handshake.None
        _serialPort.ReadTimeout = 500
        _serialPort.WriteTimeout = 500

        _serialPort.Open()

        _continue = True
        readThread = New Thread(AddressOf Read)
        readThread.Start()

        GroupBox1.Enabled = True
        btnOpen.Enabled = False
    End Sub

    Public Shared Sub Read()
        While _continue
            Try
                Dim message As String = _serialPort.ReadLine()
                Console.WriteLine(message.Trim())
            Catch
            End Try
        End While
    End Sub

    Private Sub btnClose_Click(sender As Object, e As EventArgs) Handles btnClose.Click
        _continue = False
        _serialPort.Close()
        GroupBox1.Enabled = False
        btnOpen.Enabled = True
    End Sub

    Private Sub btnReset_Click(sender As Object, e As EventArgs) Handles btnReset.Click
        _serialPort.Write(Chr(24))
    End Sub

    Private Sub btnStatus_Click(sender As Object, e As EventArgs) Handles btnStatus.Click
        _serialPort.Write("?")
    End Sub

    Private Sub btnUnlock_Click(sender As Object, e As EventArgs) Handles btnUnlock.Click
        _serialPort.Write("$X" & vbLf)
    End Sub

    Private Sub btnTestA_Click(sender As Object, e As EventArgs) Handles btnTestA.Click
        ' works
        _serialPort.Write("G90G10L20P0X0" & vbLf)
    End Sub

    Private Sub btnTestB_Click(sender As Object, e As EventArgs) Handles btnTestB.Click
        ' works
        _serialPort.Write("G38.2X-1F25" & vbLf)
    End Sub

    Private Sub btnTest_Click(sender As Object, e As EventArgs) Handles btnTest.Click
        ' works only every 2nd use and throws "Bad number format"
        _serialPort.Write("G90G10L20P0X0" & vbLf & "G38.2X-1F25" & vbLf)
    End Sub
End Class
```

chamnit, 2015-08-31:
@thoralt : Not sure what's happening here, but it could be something corrupting the communication between the Arduino and your computer. In config.h, you can uncomment the REPORT_ECHO_LINE_RECEIVED option (recompile and flash) to have Grbl report back the string it received right before executing it. It'll be partially parsed. No spaces, all caps, no comments.

109JB, 2015-08-31:
The echo is a good place to start as @chamnit mentioned.

In my GUI that I am writing, I too have found that sending commands one right after another can result in grbl not receiving the commands correctly. I am not sure why, but it happens even when sending separate serialport.write commands from the GUI. For this reason I implemented a wait for an OK response from grbl before sending another command. This has solved all instances of what you are describing in your post.

thoralt, 2015-08-31:
@chamnit Okay, I'll check that.
@109JB I can confirm the behaviour you observed. Even outside my problem described above grbl does not execute a movement every now and then (e. g., rarely the pull-off I implemented after probing is missing). This is especially sad since grbl-panel _is_ already using the "wait for OK for every single command" method which works well. I am trying to implement the recommended "buffer counting" method for a bit of extra performance with large numbers of small movements and this seems to cause these problems. But I will evaluate the `REPORT_ECHO_LINE_RECEIVED` option and additionally sniff directly at arduino's RX pin before throwing my work away.

AlexHolden, 2015-08-31:
I'm also seeing something a bit weird with the GUI I'm writing when I send a file in character-counting mode. Haven't got to the bottom of what's happening yet, but as an additional data point, I find that I don't get the 'Bad number format' error if grbl is in "check" mode.

thoralt, 2015-08-31:
I now did recompile grbl with `REPORT_ECHO_LINE_RECEIVED ` enabled and I'm seeing a similar, but slightly different behaviour:

```
ok
[echo: G90G10L20P0X0]
ok
[echo: G38.2X-1]
error: Undefined feed rate
```

The first command is parsed fine, while the second one is cut in the middle. Apparently the cut is at a different point than before and now grbl rightfully complains about an undefined feed rate. I recompiled without  `REPORT_ECHO_LINE_RECEIVED ` and got back the original error "Bad number format" which tells me that the cutting point changed a bit, but it is still reproducible.

I sniffed the signal directly off arduino's RX input, see screenshot below. It shows the decoded characters during the problematic transmission. The empty characters are not spaces, but line feed characters (I verified by zooming in). Apparently, the message is transmitted correctly by the USB serial converter. I tried another arduino to exclude any issue with that specific hardware and used a different brand/different layout, but it shows the same issue.

[image]

Taking the postings of @AlexHolden and @109JB into account, something in grbl's buffer management seems to be odd at the moment. How can I assist in tracking this down? I can offer skills in electronics hardware and AVR development.

chamnit, 2015-08-31:
@thoralt : Make sure that you count to 127, rather than 128 characters in Grbl's RX buffer. It's a common mistake. (Technically, I should make 128 mean 128, but it's been like this for a long time.)

AlexHolden, 2015-08-31:
Ah, I've been using 128. I think I probably got that from the line "RX_BUFFER_SIZE=128" in the stream.py example.

thoralt, 2015-08-31:
@chamnit I'm aware of the 127 bytes limit. We're talking about a string of 26 characters here and grbl cuts it after character 22. Before I send the commands, I wait several seconds, then reset/unlock, so grbl's buffer should definitely be empty. Interfering with commands 22+ bytes long is not a buffer overrun issue but something more tricky. As I said, I'm willing to help, but do not have insight into grbl's inner workings yet.

chamnit, 2015-08-31:
@thoralt : If it's a problem on Grbl's side, then current GUIs would exhibit this problem as well. So far, there has been no reported issues, as something like this would be a critical problem. So, it's not likely that Grbl is the problem, but the streaming or something to do with your setup. 

Grbl's source code up to the point to where it provides the echo is very simple. The serial read ISR is also very simple. Neither of these changed significantly in the last few months.

Please try a different GUI or Grbl's streaming script. Verify that the same g-code is being sent fine through other means.

thoralt, 2015-08-31:
To further dig into this, I've fired up my Mac (because I'm not sure how to quickly get a working Python with serial port under windows). I copied the latest version of stream.py from GitHub.

When just doing _one_ probe cycle, everything seems fine, even when doing another probe cycle shortly after. But since stream.py closes the serial connection, the Arduino gets reset after that and this is not the case in my VB.NET test setup, where I leave the connection open.

So I created a longer file containing multiple probe cycles. Since I'm probing offline with no steppers attached at the moment, grbl goes into alarm state (probing fail), so I need to unlock after every probe cycle. I'm not sure if it is legal to just $X and then probe and start over again, but didn't find any information against it. This is the file:
```
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
$X
G90G10L20P0X0
G38.2X-1F25
```
And here is the result:
```
> python stream.py probe-test.nc /dev/cu.usbmodem1411
Initializing grbl...
SND: 1 : $X BUF: 3 REC:
SND: 2 : G90G10L20P0X0 BUF: 17 REC:
SND: 3 : G38.2X-1F25 BUF: 29 REC:
SND: 4 : $X BUF: 32 REC:
SND: 5 : G90G10L20P0X0 BUF: 46 REC:
SND: 6 : G38.2X-1F25 BUF: 58 REC:
SND: 7 : $X BUF: 61 REC:
SND: 8 : G90G10L20P0X0 BUF: 75 REC:
SND: 9 : G38.2X-1F25 BUF: 87 REC:
SND: 10 : $X BUF: 90 REC:
SND: 11 : G90G10L20P0X0 BUF: 104 REC:
SND: 12 : G38.2X-1F25 BUF: 116 REC:
SND: 13 : $X BUF: 119 REC:
  Debug:  [echo: $X]
  Debug:  [Caution: Unlocked]
  Debug:  [echo: G90G10L20P0X0]
SND: 14 : G90G10L20P0X0 BUF: 116 REC: ok1ok2
  Debug:  [echo: G38.2X-10]
SND: 15 : G38.2X-1F25 BUF: 116 REC: error: Undefined feed rate3
SND: 16 : $X BUF: 119 REC:
  Debug:  [echo: 10L20P0X0]
  Debug:  [echo: G38.2X-1F25]
  Debug:  ALARM: Probe fail
  Debug:  [PRB:0.000,0.000,0.000:0]
SND: 17 : G90G10L20P0X0 BUF: 116 REC: error: Expected command letter4ok5
  Debug:  [echo: $X]
  Debug:  [Caution: Unlocked]
SND: 18 : G38.2X-1F25 BUF: 116 REC: ok6
SND: 19 : $X BUF: 119 REC:
  Debug:  [echo: G90G10L20P0X0]
  Debug:  [echo: G38.2X-1F25]
  Debug:  ALARM: Probe fail
  Debug:  [PRB:0.000,0.000,0.000:0]
SND: 20 : G90G10L20P0X0 BUF: 116 REC: ok7ok8
  Debug:  [echo: $X]
  Debug:  [Caution: Unlocked]
SND: 21 : G38.2X-1F25 BUF: 116 REC: ok9
SND: 22 : $X BUF: 119 REC:
  Debug:  [echo: G90G10L20P0X0]
  Debug:  [echo: G38.2X-1F25]
  Debug:  ALARM: Probe fail
  Debug:  [PRB:0.000,0.000,0.000:0]
SND: 23 : G90G10L20P0X0 BUF: 116 REC: ok10ok11
  Debug:  [echo: $X]
  Debug:  [Caution: Unlocked]
SND: 24 : G38.2X-1F25 BUF: 116 REC: ok12
SND: 25 : $X BUF: 119 REC:
  Debug:  [echo: G9]
  Debug:  [echo: G38.2X-1F25]
SND: 26 : G90G10L20P0X0 BUF: 116 REC: error: Unsupported command13error: Invalid gcode ID:3314
  Debug:  [echo: $X]
SND: 27 : G38.2X-1F25 BUF: 116 REC: ok15
SND: 28 : $X BUF: 119 REC:
  Debug:  [echo: G9]
  Debug:  [echo: G38.2X-1F25]
SND: 29 : G90G10L20P0X0 BUF: 116 REC: error: Unsupported command16error: Invalid gcode ID:3317
  Debug:  [echo: $X]
SND: 30 : G38.2X-1F25 BUF: 116 REC: ok18
SND: 31 : $X BUF: 119 REC:
  Debug:  [echo: G90G10L20P0X0]
  Debug:  [echo: G38.2X-1F25]
  Debug:  ALARM: Probe fail
  Debug:  [PRB:0.000,0.000,0.000:0]
SND: 32 : G90G10L20P0X0 BUF: 116 REC: ok19ok20
  Debug:  [echo: $X]
  Debug:  [Caution: Unlocked]
SND: 33 : G38.2X-1F25 BUF: 116 REC: ok21
SND: 34 :  BUF: 117 REC:
SND: 35 :  BUF: 118 REC:
G-code streaming finished!

WARNING: Wait until grbl completes buffered g-code blocks before exiting.
  Press <Enter> to exit and disable grbl.
```

I'm not sure how to correctly read all this output, but you can find truncated commands in grbl's `[echo...]` statements as well as I observed earlier in VB.NET. The first truncated echo seems to be after `SND: 14`.

P.S.: Did you look at my scope screenshot? Just telling me "It's an app related issue" after physically measuring valid gcode directly at arduino's RX pin makes me sad.

chamnit, 2015-08-31:
@thoralt : No need to be snarky. The fact that this is the first time a problem like this has been reported still leads me to think it's your setup(which is also hardware or connection related) or streaming dependent. I mentioned that it's likely an app-related issue because it's almost always caused by the app. If you don't believe me, check the closed issue threads. It's often caused by the streamer sending too many characters either by counting to 128 characters or it's sending more than it thinks it is, which causes Grbl's RX buffer to overflow.

Now that you have tested against another streamer, this rules out streaming for the most part. If you could try one more thing, alter the serial baud rate for Grbl. This is done in config.h. Change to something like 56700 baud or 250000 baud. These are considered more accurate due to timing inaccuracies of the 328p and the clock (see the 328p data sheet for details). There has been one report that 115200 baud can lead to some weird issues similar to yours and data may be lost. Even if you read the signal directly from the Arduino RX pin.

langwadt, 2015-08-31:
try to just a serial terminal and paste as much as will fit in the buffer

chamnit, 2015-08-31:
@thoralt : I think I see the problem. It's the EEPROM write that you call right before the probe. There is a write up in the Interfacing wiki on writing to EEPROM. Whenever this occurs, this disables all interrupts, including the serial RX, during the write. AFAIK, there is no way to avoid this issue in Grbl, as it's a fundamental problem with the 328p AVR.  Your streamer has to detect these EEPROM write commands and wait for an acknowledgement for that particular line of g-code before sending more data. 

So, yes, it's an app-related problem. Not Grbl.

closed 2015-08-31 by chamnit

thoralt, 2015-08-31:
@chamnit Sorry if you've got me wrong. English is not my native language and it is always difficult to write precise, polite and satisfying at the same time. I even had to look up "snarky" and definitely didn't want to sound like that. I understand your argument and grbl's statistics simply overrule my assumptions. Sorry again.

That being said, I tested your recommendation with the alternative baud rate. I didn't have the time to test hundreds of probe cycles (now with steppers connected again), but it seems the errors are gone. The rounding errors in the baud rate setting might contribute to the problem (I measured 119.000 output at the USB serial converter, and maybe the AVR is a bit off the other direction), but your second suggestion, the EEPROM write delay, is much more likely.

I'll try to implement something like waiting for the queue to drain whenever EEPROM writes occur before transmitting the actual command. Thanks for the clue.

chamnit, 2015-08-31:
@thoralt : No problem. I had a German professor as a graduate advisor. It took me a while to understand his very direct, and often terse, statements were meant to be constructive criticism, rather than demeaning or insulting remarks. He turned out to be my favorite professor in my entire time in school.