grbl/grbl #793: GRBL 3.5 for Mac OSX does not allow Choose file

thanhmeye, 2015-09-01:
Hi there,

I'm running grbl 3.5 on Yosemite to control my Shapeoko 2. Grbl allows me to connect to the machine, jog it in xyz directions and set Zero position. However, when I clicked on Choose file, nothing comes up that would allow browsing .nc file. I used to run an old version of Grbl on OSX Lion about 9 months back. That was before upgrading to Yosemite.

My grbl was obtained from http://zapmaker.grblcontroller.s3-website-us-west-2.amazonaws.com/GrblController35.dmg

Do you know what is causing it? In the mean time, I will downgrade to an older grbl version, hopefully that would work. Thank you!

109JB, 2015-09-01:
this isn't a grbl issue but rather an issue with the user interface. Grbl resides on the Arduono and what you see in your MAC is the user interface. They are separate entities that work together. Your question will have to be referred to the author of the interface.

109JB, 2015-09-01:
A quick search reveals that you are referring to GRBL Controller. The following are some links for GRBL Controller.

https://github.com/zapmaker/GrblHoming/issues?q=is%3Aopen
http://zapmaker.org/projects/grbl-controller-3-0/

closed 2015-09-01 by chamnit

LETARTARE, 2015-09-01:
hello,
I have grouped the latest releases on 'https://github.com/zapmaker/GrblHoming/releases'
Best regards

thanhmeye, 2015-09-01:
Noted. Thanks!

It seems that several others also experience my issue.
https://github.com/zapmaker/GrblHoming/issues/102

Cheers,
Mai

<http://www.aseanpreneurs.org/>