grbl/grbl #794: Homming - Define a switch for each axis and reverse when switch is triggered

metropt, 2015-09-03:
I have two end stop switch on my corexy and I saw that when running home I can stop it using any of the endstop switch. I would like to know if it possible to tell grbl that sw1 is for axis A and sw2 is for axis B per example, in my case both endstop switch could be triggered by B axis.

Another question, is it possible to make the motor reverse if the switch is already triggered when homming started?

thank you

chamnit, 2015-09-03:
@metropt : Grbl's CoreXY implementation was designed to home the workspace X and Y axes, not the individual A and B motors (not to be confused with A and B rotation axes). To me, it doesn't make much sense to home A and B motors.

metropt, 2015-09-03:
forget what I said about A and B axes, I meant XY axes.

metropt, 2015-09-04:
Re-editing my question: can I assign a specific trigger to a specific axes? In my setup one of the triggers could be hit by both axis. 

My second question is if it is possible to make the axis reverse motion when homing process start and end switch is triggered.

109JB, 2015-09-05:
There are already specific triggers for each axis. Digital pin 9 is for x, 10 is for y, and 11 is for z..

You should be able to send the unlock command and then jog off of the switch before homing.

ecncshop, 2015-09-10:
Is it possible to zero the McPos ? I dont have limit switches in my machine. But I want to jog to the hard stop and set that as my McPos zero. So that I can use some software limits : till I install limit switches. I understand Wpos can be made zero using G10 or G92 etc. Is there any command to make MPos zero ?

langwadt, 2015-09-10:
reset ;)

metropt, 2015-09-10:
There is any cmd to know if the sensor is triggered?

If I connect the X trigger to pin 10 and y trigger to pin 9 it works the
same way.

Cumprimentos,
Jose Xavier

ecncshop, 2015-09-10:
@langwadt : Software reset ( CtlX) does not make Mpos zero. Which reset r u talking about ? Ofcourse I can disconnect and reconnect or recycle power to arduino, but I was wondering if there is a software command to do this.

chamnit, 2015-09-10:
@metropt : Yes. In more recent versions of Grbl, you can enable limit switch state in the status reports. You can also enable control pin states, but only from the config.h file. Please read the wiki on how to enable these.

@ecncshop : There is no set MPos to zero command, because adding one would lead to misuse (which almost always happens). As @langwadt stated, the only way to do so, other than homing, is to hard reset your Arduino when you are at your hard stop. Fortunately, the Arduino has a hard reset pin that you can easily wire to a button so you can do this quickly at your machine.

ecncshop, 2015-09-11:
Ok : regariding zeroing the Wpos: I understand G92 or G10 command can be used. But work offset set using G92 is getting reset by software reset CtlX. So that wont work.  So Iam currently using G10. But this writes to EEPROM everytime and Iam worried about life as only limited write cycles r allowed to an EEPROM . So : is there a better way to set work offset without writing to EEPROM ?

chamnit, 2015-09-11:
@ecncshop : The Arduino 328p's EEPROM has a 100,000 write cycle life. It'll take 27 years if you rewrite EEPROM, 10 times a day, everyday. This is well beyond an expected life of a machine tool.

Also, G92 being reset is part of the g-code standard. It's not persistent.

109JB, 2015-09-12:
@chamnit 
In a previous post you wrote "There is no set MPos to zero command, because adding one would lead to misuse (which almost always happens)."

I don't understand the reservation to allowing resetting of the MPos or what "misuse" you are concerned about. Both LinuxCNC and Mach3 allow resetting of MPos when home switches are not installed on a machine. As a matter of fact in LinuxCNC it is required. I am not sure if it is mandatory in Mach3, but I do know that manually resetting MPos is achievable and with similar benefits. I will describe below the Linux|CNC procedure and the benefits of doing it this way.

When LinuxCNC is started the machine MUST be homed before you can use it, but you don't have to have home switches. If home switches are installed the homing sequence is much like GRBL's homing sequence. If home switches aren't installed the machine axis is manually moved, or jogged to a reference position and then the home button is pushed. If no home switches are installed LunuxCNC knows it and simply resets the MPos for that axis to zero. This procedure is repeated for all axes and the machine is considered "homed".

So what are the benefits of this?  The big one in my mind is that soft limits can be used. Without a "home" position, soft limits won't work, which is why in GRBL soft limits can't be used unless the machine has switches. The following quote is directly from the LinuxCNC "Getting Started" manual:

"A machine can be operated without home switches. If the machine has limit switches, but no home
switches, it is best to use a limit switch as the home switch (e.g., choose Minimum Limit + Home
X in the pinout). If the machine has no switches at all, or the limit switches cannot be used as
home switches for another reason, then the machine must be homed "by eye" or by using match
marks. Homing by eye is not as repeatable as homing to switches, but it still allows the soft limits
to be useful."

I myself used the soft limits in LinuxCNC without having limit or home switches for over a year on my first CNC mill, and for several months on my current one and the soft limit feature proved very useful even though I had to use non-switch references to home. GRBL could also have the usefulness of soft limits without having physical switches, but not without being able to reset the MPos.

As for misuse, I again do not see what concern there would or should be. Certainly GRBL, or LinuxCNC, or Mach3 can be misused in many ways in their present form. Even Fanuc, Okuma, or any of the commercial controllers can also be "misused".  Just about everything in life can be "misused". The accelerator on my car can be misused, but the car manufacturers still put them in there. Just because misuse is possible does not mean a feature should not be considered for implementation, particularly when it could provide a useful benefit.

I would be curious as to what you would consider "misuse" of a feature allowing resetting of the machine position.

chamnit, 2015-09-15:
@109JB : TLDR. In short, Grbl will not let a user set the machine position outside of homing or hard reseting the Arduino. For two reasons, to keep things simple and make that GUI's mature correctly. In the past, early GUIs really did a number on how a machine should be run. Everyone had their own interpretation on it based on nothing but whats popular. One example is the use of G92, rather than work coordinates. Or the importance of homing. Or how to do tool changes. Many decisions with Grbl up to this point has involved making sure GUIs are written to use a machine controller in the accepted standard way. By keeping it fairly restrictive and simple at the same time. As an experienced machinist, I think you'll find the value in that.

That said, I've done my research on how LinuxCNC, Mach3, and Haas controllers work. Settings. Configurations. I plan on installing all that for the Pro version. Because, it's for pros or advanced users who can use additional things like that.

DarklyLabs, 2015-09-19:
Hi Chamnit,
I think the issue @metropt is reporting is that when homing with a coreXY machine that has an x & y limit switch installed, either switch will trigger the x-axis or y-axis limit being detected.

For instance, if the machine is currently performing the x-axis homing sequence, triggering the y-axis limit switch will cause it to register the x-home position has been reached.

There are situations where this causes problems. It is possible for a machine to trigger the y-axis limit switch during the x-axis homing sequence.

Is there anything in the latest GRBL code to help overcome this?