grbl/grbl #795: Advice before I port to Leonardo / SLA galvos

unorthodox-engineers, 2015-09-04:
Greetings, not sure where else to post this. I'm finding it hard to contact Sonny, or any of the developers.

I'm (re)building my new UV laser/galvo system which aficionados of the Form-1 would find weirdly familiar. All the electronics work - the galvos are driven by a dual 12-bit DAC in turn by an Arduino Pro Micro (Leonardo compatible). I'm just working out the build platform mechanics, prior to experimenting with different types of photogoo.

I've got it happily drawing test patterns using custom code, but I'd like to drive it with more standard 3D printing software, and it looks like the best way to achieve that is to port GRBL. The minimum-jerk motion planner is quite nice, and In theory I just have to rip out the X/Y stepper pulse generators and replace them with the SPI DAC routines I already have.

I've done a _lot_ of microcontroller / robotics / g-code, so I'm not after coding help. I'm just after advice on whether this is already being done, and whether I should do it to GRBL. I've looked at Sprinter and Marlin-SL, but neither of those projects have been updated in years. Then again, perhaps bringing one of those up to date would be better.

While I'm at it, perhaps I could even fire up the old Taig CNC and make sure any Leonardo port still does steppers properly. The 32u4 has some differences in the timers and loses a few small chunks of memory to the USB interface, but is pretty capable. The original Leonardo I can take or leave, but I love the Pro Micro - they fit anywhere.

Finally, is there interest in folding either of these features (Leo support, or Galvos) into the main tree? If so, I'll be a little more soft-eng about everything. If not, I'll just fork and hack with no expectations of merging again. I don't have a lot of time to spend on this, so I need to be efficient.

you can reach me at unorthodox.engineers@gmail.com if this isn't the right place.

chamnit, 2015-09-04:
@unorthodox-engineers : Emailing you now.

closed 2015-09-15 by chamnit

hilam8899, 2018-07-15:
hi,chamnit 
I do that same work,can you sender me about gavlo base grbl ,thank you!
hilam8899@gmail.com