grbl/grbl #796: position lost if step mode changed

J-Dunn, 2015-09-08:
If I send an axis

J-Dunn, 2015-09-08:
Further more , if I switch back to half step and new "zero" ( half way down the axis ) and send G0X0 it  correctly does not move, but a subsequent move to a negative position ( in an attempt to return to true zero )  runs into the buffers at full speed and seems to keep going until reset.

chamnit, 2015-09-08:
@J-Dunn : Grbl was not designed for non-fixed stepping modes. When a step mode is changed, it's advised to only do so when there are no motions in queue and between jobs.

closed 2015-09-08 by chamnit

J-Dunn, 2015-09-08:
I realised it was not designed to cater for this, as noted in #1 

However, even if this is done between jobs it still loses where it is and will require a fresh homing cycle. I don't see anything warning that this needs to be done when change $100...$102 . That probably should documented if it is not already.

I was thinking that this could be useful for large moves when running a machine on small micro-steps for accuracy. Flip to full step mode in gcode to get there more quickly.

What would need addressing to cater for this? It seems a simple ratio needs to be applied to position counters and recalculate any moves already in the buffer.  There is probably already a way to trigger that for other reasons. 

If you can give me some pointers, I'll probably have a look at adding this.

langwadt, 2015-09-08:
I assume you are still using the l6474, to  change the stepping mode the outputs have to be disabled and the internal state will be reset so you can't just change mode without risking lost steps

J-Dunn, 2015-09-08:
I don't need the internal ABS_POS counter since it is grbl that is maintaining postiion, not the chip.
It's true that I'd been doing testing manually and changing step mode when the machine was in idle , ie with the buffer empty and outputs set to HiZ state in st_go_idle().
Equally changing $100 .. 102 will only be accepted when idle o'w "[not idle]" is output.

It would still be nice if this did not destroy the position counters. If that is simply a mult and a div it seems minimal overhead.

These variables can currently bet set when machine is idle. If position is lost when setting these variables, it probably should be flagged by outputting [axis posn. lost]  , or similar. If it can be handled by a simple multiplication, that would seem better than a warning. 

Having noted that to change $100 the machine has to be idle and buffers will be empty, this seems trivial, unless I'm over looking something. Do you see any complications?

langwadt, 2015-09-08:
it is not only the abs_pos count that gets reset, it is also the internal state, i.e if you are in 1/16 mode 
the 0-15 count gets reset. 

afaict grbl keeps track of the position in steps, so I'd think you have to scale the position and possibly offsets to change steps/mm on the fly

or maybe do some trickery simular to a switch between mm/inch if that can be done on the fly

J-Dunn, 2015-09-09:
Yes, it seems obvious from what I reported that GRBL is counting "STEP" pulses ( which can be micro-steps for the motor ) not mm or inches. This raises the question of whether it loses home position when switching units too. I'll have to check. 

L6474 does not have a separte counter for micro-steps, abs-pos is in micro-step units and that is why the count becomes meaningless after a change. This is effectively the same thing I reported happening in GRBL. 

I had not considered that offsets need checking too. Thanks for pointing that out.

langwadt, 2015-09-09:
what I meant by internal state is that when micro stepping it counts through a sine table of currents to set position, when you change stepping mode it always starts from step zero, first position in the table. so depending on what quadrant and step in the table you are at when you change you lose steps

J-Dunn, 2015-09-10:
You are correct about the sine table but since GRBL will reject any change to $100...102 if the machine is not idle, this is not really an issue. Obviously, changing step mode or steps/mm while a motor is running is likely to lead to missed steps and positional inaccuracy. 

What I was envisaging was switching to , say, full step mode for a large G00 move but cutting at step_by_4 , for example.

BTW, I'm having problems with L6474 overheating on theirX-NUCLEO-IHM01A1 demo board,

It seems that the default config switches way too fast and is unusable in practice. I have slowed it down but it still gets hot after about a minute of continuous movement. I also found a comment from an ST guy suggesting some configuration can be unstable and oscillate. Do you have any info about reliable configs that don't overheat? 

Thanks.

langwadt, 2015-09-10:
look at fig.6 in the datasheet, if the motor is not at reset position when you change stepmode it will power up in a different position. 

I'm not convinced the IHM01A1 is thermally very good, or ST just exaggerated what the driver can do, to do 2A on my stack of 3 i put a fan on, a 40x40mm fits nicely in the cutout blowing through the space between the boards.  

at work we use the l6474 on our own board with the back of the pcb bolted to an alumium block, it runs 3A with no problems

I've never seen any info on how to determine the config

J-Dunn, 2015-09-11:
Thanks for the input. 

That was the sort of conclusions I was coming to, it's useful to have confirmation. A small PCB like the IHM01A1 can only sink the heat from the chip. It's ability to actually get rid of it is limited. Forced convection is probably necessary for sustained use unless the dissipation in the chip can be reduced significantly. 

It looks like the motors are not going to work reliably above about 3.5kHz pulse rate so I really don't need the chip to be chopping that fast. It just needs to remain ultra-sonic. In principal, if I can get a slower chopping freq. the dissipation will probable come down notably. 

The problem is that it's quite a complex patented algo with very little into on how to configure it and suggestions that some configs are not stable. From an engineering point of view suck-it-and-see design sucks. 

The support forum software at st.com is a dysfunctional mess at the moment so not much hope there. 

Are the boards running at 3A using the default configuration? 
thanks.

langwadt, 2015-09-13:
I'll correct you ;)  TON_MIN is a minimum time, it'll be on until the set current is reached 
and you can't deduce the RMS current from the duty cycle, when ON the current will rise at a rate depending supply voltage and motor inductance, when OFF the current will decay depending on the motor inductance and whether fast or slow decay is used.

from that you can also see that you can't just pick your chopping frequency, it'll depend on speed, voltage, and motor inductance and how much current ripple you can live with

J-Dunn, 2015-09-13:
Obviously the response is not trivial in an inductive circuit and a complex control algo,  but I don't see sect 7 suggest anything like fast/slow decay cycles during TOFF_MIN. That regime applies to  controlled ON state and on step changes which alter the target currect value. TOFF_MIN is an uncontrolled off state.

With default TON_MIN=TOFF_MIN=21us and say 100us pulse steps the will be several TON_MIN TOFF_MIN cycles during each step period, with TON_MIN being PWM switched to maintain target current around an RMS of TVAL, once it is reached.  With a low inductance cct that will give something close to a 50% duty cycle with on currect from TVAL.

With higher inductance, both rise and fall will be slower. Still roughly 50% duty cycle. 

So setting TVAL=3A will be more like 1.5A RMS. This is presumably why the chip allows a max TVAL of about 6A 

I've fiddled around and got a config that is getting more power into the motor ( less stalling ) but it still throws a TH_WRN after about 20s of continuous motion even at TVAL of 3A.

power disipation is given as 22 C/W but the heatsinking described is far more than provided on the EasySpin boards:
2. POWERSO36 mounted on EVAL6474PD Rev 1.0 board: four-layer FR4 PCB with a dissipating copper
surface of about 40 cm2 on each layer and 22 via holes below the IC.

I'm losing hope of getting anything more out of these demo boards, than I could out of a Polulu. I have a couple of those, so I may rig up test curcuit to compare. 

Getting a pre-emptive TH_WRN signal that allows a controlled stop is nice, and it seems to work well with GRBL. I wait a bit then release the hold and it continues the rest of the  move without losing position. But beyond that I don't see any advantage.

langwadt, 2015-09-13:
from section 7.1: during the OFF state both slow and fast decay can be performed; the better decay combination is automatically selected 

it works similar to  buck converter so the duty cycle will be mostly dependent on the ratio between the supply voltage and the voltage required to reach the current in the DC resistance of the motor
at speed the supply voltage will be countered by the back-emf

looking at the on resistance of the switches they are quite similar to the other drivers and comes out to roughly 5W at 3A, that would require ~20C/W

J-Dunn, 2015-09-13:
Thanks but if you look at 7.1 it refers to the OFF state of the PWM during TON and the transition from one step level to another, not the unregulated TOFF_MIN state. 

As with all FET drivers, the problem is not so much the on state but the switching transients. That is why the fast switching with timing parameters in the range of 1 to 20us are causing heating problems. The PWM during TON will be in range of several 100kHz.

Now there may be some artful way of configuring TFAST &co. to reduce that but it's going to be a lot of work unless ST produce some guidance notes. I suspect improvements will be limited.

From the gerber files they provide, it's two layers of 35micron Cu. The main area inside the pins is about 25cm^2, a bunch of vias under the chip.

J-Dunn, 2015-09-18:
OK, I got this constant holding torque on the scope and it is switching pretty fast and cleanly, so your calculations should be about right. 35micron copper ( 1oz ) gives 43C/W per sqr inch. So if the board layout is good, it should be just about enough. 

TVAL=64 ( 2A ) throws a TH_WRN but seems to stabilise under TH_SD . 

I'll try your suggestion of forced ventilation, I have a small 40mm GPU fan.