grbl/grbl #797: how to calculate first value

kahyaoglumert, 2015-09-08:
Hi all,

How can I calculate x,step/mm , y,step/mm etc.. Please let me show that one way. I am using trying and error method.  

thank you so much

Norwegian-Gadgetman, 2015-09-08:
Generally, you need to know how many steps per turn of the steppers, the microstepping setting on your driver, and the circumference of the pulleys(if using belts, or the 'rise' if using soe sort of threaded rod)

Take the number of steps/rotation and multiply with the microstepping selected in the driver, then divide that by the circumference of the pulley.

20tooth GT2 pulleys have a circumference of 40mm, most NEMA17 steppers are 200step, and a common microstepping setting is 8. That gives (200*8)/40 = 40 steps/mm.
Of course, as belts can stretch a bit, you will probably need to adjust it slightly.

On threaded rods, you need to know how much it 'rise' for one full turn. 
with M8 threaded rod the nut will move 1mm as it is rotated a full turn.
Then you take the number of steps(multiplied by whatever microstepping is used) and divide by the 'rise'

kahyaoglumert, 2015-09-13:
Thank you for your reply,

There's something confusing my mind. I calculate all settings for my mechanism. X axis and Z axis are correct, but I Y axis did not go to 10mm, It went to 5 mm when I ordered 10mm. How can I fix this problem? 

[image]

thank you so much your attention.

Norwegian-Gadgetman, 2015-09-13:
1. Why are you using the old v0.8 of GRBL?

2. Why are you getting different numbers for the X and Y axis?

3. Why a 50uSec on $3?

Can you describe your machine in some detail?

kahyaoglumert, 2015-09-13:
Thank you for your help in this matter :) 

1. I encountered some problems v0.9j with the controller grbl and I used the old version to get a fast way.

2. We are using 2 stepper for one direction which is calling Y axis and reduction 1 to 2 percent.

[image]

3. I've seen similar value in a case study. I use different values when test.

________________

You can see X and Z axis in this picture. 

[image]

Norwegian-Gadgetman, 2015-09-13:
I'm not certain what your machine is, but...
This value is the length of the pulses going to the driver. 
Please try setting $3 to 10 uS and try...
Actually, try anything you want between 5 and 30uS.

kahyaoglumert, 2015-09-13:
Dear Norwegian-Gadgetman thank you for your reply, 

I will try change 10uS and I will use 0.9v. 

I will write again tomorrow.

kahyaoglumert, 2015-09-20:
Hi again,

I changed 10us and it doesnot solve this problem. Then I upgrade grbl version, this way solve my problem.

Thank you Norwegia-Gadgetman.

closed 2015-09-20 by kahyaoglumert

gunerg, 2015-09-20:
selam
gönderilerinizi
türkçede  yayınlıyabilirmisiniz

kahyaoglumert, 2015-09-20:
Merhaba,

Takıldığınız sorunlar için türkçe yazabilirsiniz, bende öğrendiklerim kadarıyla size yardımcı olabilirim.