grbl/grbl #798: pulse jitter at certain speeds

J-Dunn, 2015-09-09:
In monitoring the STEP output on an oscilloscope, I notice certain pulse rates show considerable jitter. An example:

200 step/rev ; 5mm pitch lead screw = 40 step/mm or 160 pulse/mm in quarter step mode, hence $101=160

setting $111=1400 produces steady, regular pulse train.
setting $111=1200 produces considerable jitter

The period is 300us with 75us high pulse, the jitter is nominally equal to pulse width, leaving
a shadow trace starting where the regular pulse ends. I'm falling edge triggering so it seems
to be the timing between pulses that is not stable.

STEP_PULSE_DELAY is not defined

Are there certain pulse rates that mix badly with GRBL timers ? Is this expected?
Thanks.

J-Dunn, 2015-09-09:
Anotherr, probably related oddity:

$111=2300 produces either a roughly 50% M/S ratio or a very narrow pulse. Period nominally the same.This can happen in either forward or reverse direction or neither. There seems to be some semi-random condition determining whether it takes one form or the other. This may be an AMASS boundary value but may be being flipped on side or the other by the cause of the  above jitter.

Visually the trace is pretty stable at 2300 but small jitter may be flipping AMASS :?

chamnit, 2015-09-09:
@J-Dunn : The falling edge of the step pulse can sometimes be delayed by external circumstances, like the serial RX or TX ISRs firing around the time of a step pulse, but this is on the order of a few microseconds.

If you are seeing something around 75us, it's probably AMASS. 300us/4 = 75us. The 4 is the AMASS overdrive level at 3333Hz.

For falling edge stepper drivers, use the step invert setting. This should help with your problem.

J-Dunn, 2015-09-09:
Thanks. I was wondering whether this was AMASS related but had not dug into it. 

The falling trigger I mentioned was the scope not the driver. To explain that it was the low level part of the cycle that was variable.

Does that mean that AMASS is flipping between different levels? 300/2 and 300/4 ? Do some step pulse frequencies close to an AMASS boundary cause jitter? 

Does this also explain what I reported in #2, where two very different M/S ratios were being produced? That did not seem to produce any irregularity in the motor since the pulse cycle was apparently the same. It just seemed odd that it was not one thing or the other. Either direction could get the short pulse waveform, sometimes neither did.

J-Dunn, 2015-10-14:
I have confirmed that this effect is due to AMASS by recompiling without the feature. 

It seems to happen at EVEN multiples of 1200mm/min at 80 steps/mm in half-step mode. 
ie 1200, 2400,4800 but not 3600.