grbl/grbl #799: Z-Axis Limit not working

Lexxar, 2015-09-14:
Hi everyone,
my GRBL-Project ist running fine so far. Only thing that is bothering me is my Z-Axis Limiting. (X-Y-Limits working fine). I use an Arduino UNO328p with the newest grbl-firmware and an CNC-Shield which needs the old PIN-Layout (Z-Limit PIN 11). I don't need PWM etc. for the spindle. So i downloaded the .zip package disabled the line #define VARIABLE_SPINDLE in config.h and flashed the Arduino with Arduino IDE. When i try to stop the machine from working by closing the switch nothing happens (software wise). The switch has 5V unclosed and 0V closed, so i guess the source code has to be modified somewhere else too. Any suggestions?
Best Regards,
Lexxar

kahyaoglumert, 2015-09-20:
Hi,

I have same problem. I am using grbl v0.9j and CNC shield. Z-limit has a pull-up resistor 4.7k ohm. The problem is Z-limit pin which is PIN number 11 always set LOW so I am not using homing position. How can I solve this problem or any suggestions?

Thank you

kahyaoglumert, 2015-09-20:
Hello again,

I have solved my problem, I have to change Z limit pin to Spindle Enable for CNC Shield. " Z Limit pin and spindle enabled swapped to access hardware PWM on Pin 11.  " 

I solved it, but I have faced a new problem :). I working on it.

trietdao95, 2018-06-22:
hi, i'm using grbl 1.1. Where is the code i need to change to get back D11 Z limit pin? because i'm using CNC shield v.3