grbl/grbl #800: grbl 0.9j Homing fail after upgrade

KonatsuNoKo, 2015-09-14:
Hi, I'v upgraded my Grbl from 0.8 to 0.9j and now I have problem with homing.

I moved z-limit do D12 and hard limit is working fine.

When i hit $h z-axis go up hit limit switch, go down 1mm and give Alarm: Homing fali

My config

>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=110 (dir port invert mask:01101110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=200.000 (x, step/mm)
$101=200.000 (y, step/mm)
$102=200.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=340.170 (x max travel, mm)
$131=528.180 (y max travel, mm)
$132=26.700 (z max travel, mm)

any advise for beginner?

chamnit, 2015-09-14:
@KonatsuNoKo : This usually means that you haven't cleared the homing switch when it tries to move off of it. Try increasing the homing pull-off value to something like 3 or 5 mm.

KonatsuNoKo, 2015-09-14:
tried 10mm, no luck

KonatsuNoKo, 2015-09-14:
ok, the problem was to big cap on limit switches

chamnit, 2015-09-14:
@KonatsuNoKo : Great. Thanks for reporting that you found the problem. You can also increase the homing debounce delay. It'll force Grbl to wait until it reads the pin.

closed 2015-09-14 by chamnit