grbl/grbl #801: G28 (and probably G30) ignore axes

nathanwosborn, 2015-09-14:
The spec says (in part):

G28 - makes a rapid move from the current position to the absolute position of the values in parameters 5161-5166.

G28 axes - makes a rapid move to the position specified by axes including any offsets, then will make a rapid move to the absolute position of the values in parameters 5161-5166 for axes specified. Any axis not specified will not move.

G28.1 - stores the current absolute position into parameters 5161-5166.
<end of spec>

So "G28 Z0" should move only the Z axis to the position specified by G28.1.  As it stands any axes specified are ignored and all axes (x,y and z) move.  I think this may have resulted in the confusion over whether G28.1 should take axes parameters.

This arose because the post that Fusion 360 uses (grbl.cps) uses "G28 G91 Z0" to retract at the end of a program and then "G28 G91 X0 Y0" to move x and y.  The idea is to retract z before moving x and y.

I have replaced grbl.cps with a custom post that uses "G53 G90 Z-5" to retract z. But we should still fix it.

chamnit, 2015-09-14:
@nathanwosborn : Thanks for the report. I'm pretty sure its what G28 does, but a lot has changed since I last worked on that part of the code. I'll check to make sure it's operating as it should.

chamnit, 2015-09-14:
@nathanwosborn : Ok. I think I tracked down the issue. This is a difference between the original NIST g-code standard and LinuxCNC's. The NIST standard, which Grbl was originally written from, specifies to travel to the programmed position first than to the stored G28.1 position. With all axis. It looks like LinuxCNC's definition is different with it only moving axes that are specified.

Grbl has used only the NIST standard in the past, but has recently started to shift to LinuxCNC's. I'll fix this discrepancy shortly, but during testing, I did find a weird bug when sending a G91 with a G28. Fixing this bug is higher priority than converting G28/30 to LinuxCNC's definition.

nathanwosborn, 2015-09-14:
I did use the LinuxCNC page as a reference.  Thanks for looking into this so quickly.

109JB, 2015-09-15:
This is not correct for LinuxCNC.  the following command will have the results listed below in LinuxCNC

G28 Z0 (Rapid first to Z=0, then move all axes to that specified in G28.1) 

This is the same as the NIST standard

109JB, 2015-09-15:
For reference, below is the example straight out of the linuxcnc user manual:

G28 Example Line
G28 Z2.5 (rapid to Z2.5 then to location specified in the G28 stored parameters)

109JB, 2015-09-19:
Just got home from another trip and checked my machine running LinuxCNC and I was wrong. the following line:

G28 Z2.5 

Results in a rapid Z move to Z=2.5, followed by a rapid move to the z position stored by G28.1

I believe, from my feeble memory, that the Cincinnati VMC's I used to run 25 years ago worked the way the NIST standard states (ie: G28 Z2.5 - move z to z=2.5 then all axes to the G28.1 position). On those old Cincinnati's, G28 was the tool change position.

chamnit, 2015-09-24:
Ok. There is no bug with Grbl's NIST implementation of G28/30. I had forgotten that the g-code mode was still in G91 incremental (I never use incremental).

@109JB : Thanks for verifying with LinuxCNC. I did look back at the G28/30 g-code descriptions in older LinuxCNC versions. It's been non-NIST for a while. 

I'll update G28/30's behavior in v1.0, but not v0.9. This is because I don't know what the implications would be if I do. Some GUIs may assume the old NIST behavior and depend on it. If there is a compelling reason to do it for v0.9, I'd like to hear it though.

chamnit, 2015-09-24:
@nathanwosborn : The v1.0 beta in edge branch has the updated g28 and G30 behavior. Clsoing this issue.

closed 2015-09-24 by chamnit