grbl/grbl #802: Compiling with make ATMEGA2560

vlachoudis, 2015-09-15:
I've tried to compile the latest version 0.9j using the CPU_MAP ATMEGA2560
from the command line, but it fails with a lot of error messages. Probably some include is missing?

avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/spindle_control.c -o build/spindle_control.o
In file included from grbl/cpu_map.h:37:0,
                 from grbl/grbl.h:47,
                 from grbl/spindle_control.c:22:
grbl/spindle_control.c: In function 'spindle_init':
grbl/cpu_map/cpu_map_atmega2560.h:135:28: error: 'DDRH' undeclared (first use in this function)
   #define SPINDLE_PWM_DDR  DDRH
                            ^
grbl/spindle_control.c:30:5: note: in expansion of macro 'SPINDLE_PWM_DDR'
     SPINDLE_PWM_DDR |= (1<<SPINDLE_PWM_BIT); // Configure as PWM output pin.
     ^
grbl/cpu_map/cpu_map_atmega2560.h:135:28: note: each undeclared identifier is reported only once for each function it appears in
   #define SPINDLE_PWM_DDR  DDRH
                            ^
grbl/spindle_control.c:30:5: note: in expansion of macro 'SPINDLE_PWM_DDR'
     SPINDLE_PWM_DDR |= (1<<SPINDLE_PWM_BIT); // Configure as PWM output pin.
     ^
grbl/cpu_map/cpu_map_atmega2560.h:82:33: error: 'DDRE' undeclared (first use in this function)
 #define SPINDLE_DIRECTION_DDR   DDRE

vlachoudis, 2015-09-15:
I forgot to mention that using the arduino IDE it compiles with no problem

chamnit, 2015-09-15:
@vlachoudis : You'll need to change the cpu type in the makefile to the 2560. I think this will solve this.

vlachoudis, 2015-09-15:
Thanks @chamnit  I didn't think to look the makefile. Changing the DEVICE to atmega2560 works perfectly!

closed 2015-09-15 by vlachoudis