grbl/grbl #803: 2-axis homing with grbl

Zakattacks, 2015-09-16:
Hi,
I am attemting to make a laser cutter with grbl, and I need to skip the z-axis for homing (and return to 0 too?)
I found this old thread (https://github.com/grbl/grbl/issues/224) , but I have not been able to get this working in .9j firmware, some of the codes seem different. I should change it to:

#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))
#define N_HOMING_LOCATE_CYCLE ((1<<X_AXIS)|(1<<Y_AXIS))

instead of

#define HOMING_SEARCH_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
//#define HOMING_SEARCH_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))
#define HOMING_LOCATE_CYCLE ((1<<X_AXIS)|(1<<Y_AXIS))

Right?

Also, any tips for driving a 250mw laser (maybe through spindle settings?) using grbl would be appreciated!
Thanks!

chamnit, 2015-09-16:
@Zakattacks : HOMING_CYCLE_0 is correct. Or you can split them so CYCLE_0 homes X and CYCLE_1 homes Y. I've heard this works better in some cases, not sure why though. N_HOMING_LOCATE_CYCLE should remain 1.

If it doesn't work, let me know, and I'll look into the source code to see if there are any problems.

Shanjaq, 2015-10-24:
Compile and Flash?  this seems like it should be a setting

chamnit, 2015-10-24:
@Shanjaq : o rly? If you hadn't noticed, Grbl has very limited flash space remaining. Things like non-standard build options, like this, need to be compile-time macro. Not sure why you are complaining, because it isn't all that hard, given that users can "easily" upload via the Arduino IDE.

Shanjaq, 2015-10-24:
So just modify the config.h file and do the "include library" trick with the upload sketch?  Or is it the horribly convoluted and troublesome process described in this megathread?
https://github.com/grbl/grbl/issues/293

chamnit, 2015-10-25:
@shanjaq : It seems you deleted your post, but I’ll include what you said here I got through Github’s automatic emailing. Because I feel that I need to respond to it.

```So just modify the config.h file and do the "include library" trick with the upload sketch? Or is it the horribly convoluted and troublesome process described in this megathread? #293 <https://github.com/grbl/grbl/issues/293> ```

Please remember that Grbl is an open-source project and is actively supported by the free-time of myself and others in the community. Try to be a little more understanding of this fact and know that not everything is perfect. In open source, it never is.

First, yes. Modify config.h and recompile and upload. There’s a lot there, so read the comments on what they do and how to configure them.

Second, the homing cycle setup was designed to be able to adapt to different machine and switch types. You can’t have simple and configurable at the same time. It’s always a middle ground and it’s not that complicated, if you read the comments on how to set them up.

As for the mega-thread you mentioned, if you read the the first few posts, the 2-axis homing setup was cleared up immediately. The rest of the post involved a common problem with electrical noise in the limit switches and wiring. Not related to Grbl directly, but the community helped solve this user’s problem anyway.