grbl/grbl #804: Feed hold command "!" doesn't respond with "ok"

milesingrams, 2015-09-16:
I noticed that when a feed hold command is sent there is no "ok" response until a resume is sent. Upon a resume all buffered commands including the initial pause and resume commands are processed and respond with "ok". This makes it difficult when writing a GUI to ensure that the feed hold command was received without subsequently performing a status check.

chamnit, 2015-09-16:
@milesingrams : This is the intended behavior for feed holds. It puts the system in a suspend state. So you will not have an ok response sometimes. You can know when Grbl receives a hold command by monitoring the machine state in the '?' status reports.

closed 2015-09-16 by chamnit

chamnit, 2015-09-16:
@milesingrams : After thinking about it, I'll add a short confirmation message for real-time commands, like feed holds and resumes, in the next major version of Grbl, since there will be an updated communication protocol for it. It'll likely remain the way it is for the current Grbl to keep all current GUIs compatible.

milesingrams, 2015-09-17:
Thanks! That would be awesome to be able to register the receipt of the real-time commands without having to poll. But I can understand why they weren't included given that it's a suspend action.
Additionally, have you considered adding the option for job-done confirmation responses for GUI's? For instance I have GRBL running together with a automatic fluid sampler on the end of my Z axis. In order to move to the next step I have to make sure that both the stage and the fluid sampler have completed all queued actions. Currently I do this by polling the status of the stage and waiting for it to say "IDLE", but it could be nice to have the option for GRBL to automatically send a "Job Done" status when all buffered moves are complete.

chamnit, 2015-09-17:
@milesingrams : The simplest way to force Grbl to empty it's queue is to send it a G4 dwell command. When you receive the 'ok' for this, you will know that it's complete. The M2/30 program handling probably should behave this way too, but right now it doesn't. I'm pondering whether or not to not force a pause.

milesingrams, 2015-09-17:
That's definitely a nifty trick to know. However in my case I just want to know exactly when the stage has reached the desired location. For instance if I send a G0 command to tell the stage to go to X100 Y100, I recieve an "ok" that the command has been processed but that doesn't tell me that the stage is at X100 Y100. In order to know that I need to poll with "?" and wait for IDLE. since polling is effectively limited to 5Hz I might lose 200ms before I know that the move has completed.

chamnit, 2015-09-17:
@milesingrams : I mean that if you send a G4 dwell command after the G0 X100 Y100, you'll know that the move has completed when you receive the 'ok' from the G4 command.

milesingrams, 2015-09-17:
Ahhhhh. Gotcha. That's very useful. I'll give that a try. Thanks!