grbl/grbl #805: Using G43.1 Dynamic Tool Length

zhivko, 2015-09-17:
Can this gcode be used on grbl while gcode program is running and it will be taken into account as soon as grbl move axis that is specified by G43.1?
I am thinking using this feature for plasma torch height controller - in this process height sensor constantly (10Hz) reports distance of workpiece from torch, and I would like to correct Z with G43.1 Z[offset], to adapt height while gcode is executing on grbl.

chamnit, 2015-09-17:
@zhivko : Yes. G43.1 can be used in a middle of a gcode program. See LinuxCNC's description on how it works (http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G43_1).

closed 2015-09-17 by chamnit

zhivko, 2015-09-17:
Can you confirm if i understand it right please.
In the middle of gcode program means while gcode is executing?
So actual gcode (for example 1000line of gcode) that was sent to grbl have
not contained that code and then while line 500 is executing G43.1 is sent
and controller immediately acknowledges it.

Please confirm.