grbl/grbl #807: Need a way to simulate home switches for programming aid

109JB, 2015-09-25:
I have finally had some time to work on my GUI, but I am running up against a little issue. Not with grbl, but because my machine is in my barn and inconvenient for access while programming. For some parts of my program debugging I need to be able to use the $H homing feature and test parts of my program. I have been using an Arduino that isn't connected to a machine and it has been working well for most of the programming. Does anyone have an idea as to how to simulate the having homing switches hooked to the machine without actually having a machine right there. I have been having to go back and forth from the house to the barn to test things and it is getting a bit tiresome.  Thanks.

John B.

chamnit, 2015-09-25:
@109JB : I just have a prototyping shield on an Arduino Uno that has momentary buttons, leds, and a mini breadboard. It's worked for me. I've seen dedicated test systems that have a panel with motors and limit switches that you hand activate.  Or simply comment out the homing cycle stuff in the code, so you can skip it.

109JB, 2015-09-25:
@chamnit
I didn't even think about that simple of an approach. I had just assumed the homing cycle would have coding that would not like a manually activated switch like that since the steps driven during what GRBL would think was the pull off would be vastly different than the steps sent for the second approach.  If a simple switch works I can breadboard that in just a couple minutes.  Thanks.

closed 2015-09-25 by 109JB

Protoneer, 2015-09-25:
@109JB <https://github.com/109JB> : I have been using my RPI CNC Board
<http://wiki.protoneer.co.nz/Raspberry_Pi_CNC>  + [Arduino + Relay Shield]
for testing things like this.(I have mostly tested hardware this way and it
even emails me with test results :)

The hard part with this testing is getting your app code to send stuff to
GRBL and activate the Relay in the background. I would suggest a
batch/shell script that starts a Python timer script( that switches the
relay after a certain period of time) and your app in sequence.

Other option is to have some kind of messaging/log system in your app that
writes to a file. This file then gets checked for changes by the Python
script and reacts to it.