grbl/grbl #808: Compiling killed me!

stribuda, 2015-09-25:
Hi everyone,
i try since 1 Month to compile the example code of grbl, i tried all Arduino versions und put all grbl libraries, choosed Arduino Uno and all Programmer, changed the version of java und tried this on many computers... i recieved all possible error messages, the last message was: 
Arduino: 1.6.5 (Windows XP), Board: "Arduino/Genuino Uno"

grblUpload.ino:27:18: fatal error: grbl.h: No such file or directory
compilation terminated.
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Please help me, tell me whitch version of arduino, of java and of grbl i should install.

chamnit, 2015-09-25:
Make sure to import only the source code directory named 'grbl' into the Arduino IDE, not the download folder, usually named 'grbl-master'.

stribuda, 2015-09-25:
Haha it works xDDD

Done Compiling.
Sketch uses 28,790 bytes (89%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,485 bytes (72%) of dynamic memory, leaving 563 bytes for local variables. Maximum is 2,048 bytes.

Thank you!

closed 2015-09-25 by chamnit

cducvn, 2017-10-08:
I got the message "Invalid library found in C:\Users\Mylaptop\Documents\Arduino\libraries\grbl-master: C:\Users\Mylaptop\Documents\Arduino\libraries\grbl-master", when I tried for compiling more time with this source: grbl-master-v0.9i. Please help!

chamnit, 2017-10-08:
@cducvn: Follow the directions exactly as written. The "grbl-master" directory name should be a big clue.

lalo-uy, 2017-10-09:
Reneme the dir to just grbl

cducvn, 2017-10-22:
It was done very good, thank you so much.

dineshvyas, 2019-11-25:
there is no such thing IMPORT anywhere??????????

dineshvyas, 2019-11-25:
can someone post screenshot???