grbl/grbl #812: GRBL loosing position

CreateiveRobotics, 2015-10-08:
I've had a couple of occasions now where GRBL will change the Z offset during a job - I'm confident its not just missing steps due to loading or speed.

The first time it happened it lost position by a few mm (+) at the start of a job, it did this consistently and I couldn't see anything wrong with the G code. It was a pocketing operation and despite zeroing on the work-piece top it would always end up about +2mm by the time it reached the first pocket position. Each time I would reset the position and try again with different feed rates and it always happened. The second time it shifted by 0.4mm in the middle of an engraving job. I saved that file and marked the point at which the Z offset changes. Again it did this consistently every time I ran the job, and even after altering feed rates. I was able to complete the job by creating a new MOP from the point at which it looses position (it was engraving text and lost position after moving on to a new letter) - it all worked fine with the new MOP.

My machine is a shapeOKO, GRBL build 0.9i.20150329 Machine settings are:

$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.050 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=10.000 (homing feed, mm/min)
$25=1500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=2.000 (homing pull-off, mm)
$100=43.801 (x, step/mm)
$101=43.839 (y, step/mm)
$102=320.000 (z, step/mm)
$110=14000.000 (x max rate, mm/min)
$111=14000.000 (y max rate, mm/min)
$112=700.000 (z max rate, mm/min)
$120=1200.000 (x accel, mm/sec^2)
$121=1200.000 (y accel, mm/sec^2)
$122=80.000 (z accel, mm/sec^2)
$130=210.000 (x max travel, mm)
$131=332.000 (y max travel, mm)
$132=90.000 (z max travel, mm)

G code: (the text file up-loader wouldn't work ..? so I have cut and pasted the first part of the file, up to the first section after the problem appears - it was a fairly large file):

( Made using CamBam - http://www.cambam.co.uk )
( calvert 10/8/2015 12:17:37 PM )
( T0 : 0.0 )
G21 G90 G40
G0 Z3.0
( T0 : 0.0 )
T0 M6
( VEngrave2 )
G17
M3 S1000
G0 X33.4766 Y17.9144
G0 Z0.5
G1 F300.0 Z0.0
G1 F1200.0 Z-0.4586
G1 X33.4204 Y17.7698 Z-0.3597
G1 X33.3539 Y17.6251 Z-0.2601
G1 X33.3524 Y17.622 Z-0.258
G1 X33.2676 Y17.4684 Z-0.1431
G1 X33.2667 Y17.4667 Z-0.1417
G1 X33.2548 Y17.4455 Z-0.1218
G1 X33.2812 Y17.4925 Z-0.1626
G1 X33.2865 Y17.5012 Z-0.1695
G1 X33.3229 Y17.5653 Z-0.2178
G1 X33.3557 Y17.6285 Z-0.2625
G1 X33.3608 Y17.6391 Z-0.2698
G1 X33.3972 Y17.7164 Z-0.3231
G1 X33.4045 Y17.7329 Z-0.3344
G1 X33.4382 Y17.8118 Z-0.3884
G1 X33.4464 Y17.8322 Z-0.4022
G1 X33.4827 Y18.1169 Z-0.5508
[... 688 lines cut ...]
G1 X15.8217 Y34.4482 Z-0.4698
G1 X16.0107 Y34.4753 Z-0.4524
G1 X16.199 Y34.4758 Z-0.4375
G1 X16.2027 Y34.4755 Z-0.4372
G1 X16.3554 Y34.4563 Z-0.4277
G1 X16.504 Y34.4209 Z-0.4203
G1 X16.5126 Y34.4184 Z-0.42
G1 X16.6444 Y34.3728 Z-0.4173
G1 X16.921 Y34.2391 Z-0.4192
G1 X17.069 Y34.1421 Z-0.4264
G1 X17.2073 Y34.0296 Z-0.4379
G1 X17.2089 Y34.0281 Z-0.438
G1 X17.3515 Y33.8736 Z-0.4559
G1 X17.4682 Y33.6936 Z-0.4785
G1 X17.551 Y33.4908 Z-0.5021
G1 X17.5523 Y33.4863 Z-0.5026
G1 X17.587 Y33.3323 Z-0.5184
G1 X17.6012 Y33.1739 Z-0.5302
G1 X17.5965 Y33.0139 Z-0.5401
G1 X17.5407 Y32.7234 Z-0.5537
G1 X17.5006 Y32.5932 Z-0.5593
G1 X17.4524 Y32.4654 Z-0.5651
G1 X17.3965 Y32.3404 Z-0.5708
G1 Z0.0
G0 Z3.0

109JB, 2015-10-09:
Are you sure grbl is changing the work offset.  You can check the offset before and after using the $# command to grbl

CreateiveRobotics, 2015-10-09:
I will check next week but I was having to re zero the z axis on the workpiece every time and reset the offset.

chamnit, 2015-10-09:
If Grbl is reporting the correct programmed position after an incident, the problem is not likely Grbl. This is because Grbl reports the running total of steps executed by the interrupt process that sends pulses to your stepper drivers. So, the problem has something to do with the stepper drivers, the steppers, or Grbl motor settings.

If Grbl is not reporting the correct position within a step per mm, then there is likely a communication problem or a data corruption issue. 

As far as I can tell, you have no unusual gcode that would cause any issue like the one you are reporting.

langwadt, 2015-10-10:
could it be something really simple like a stepper driver overheating and shutting down?

tmpvar, 2015-10-11:
lasersaur has a guide for this stuff (they also use grbl). http://www.lasersaur.com/manual/troubleshooting

In every situation I've encountered it was user error, configuration, gcode, or hardware issues.

CreateiveRobotics, 2015-10-12:
Thanks. I would agree that user/machine errors are probably the cause but what caused me to doubt that is the consistency of the errors. I have calibrated my machine and done repeatability and accuracy tests with no problems - it is nice and square, there are never any lost steps even at top speed and I have run plenty of jobs on it with no problems.

The two issues I had were all entirely repeatable - running the problematic MOP without a work piece and at different absolute Z heights still produced a position error at exactly the same point in the MOP.

So ... I ran a whole sequence of tests with the problematic MOP and edited the MOPs G1 feed rates - Lower feed rates removed the issue but strangely they were still above the Z max feed rate. In the end I reduced the machine Z acceleration by 10 and now it works ... DOH!

I guess the lesson there is that running lots of position and speed tests with G0 won't always show up an acceleration issue ... which I now suspect is down to me topping my PSU current rating under certain conditions.

closed 2015-10-12 by CreateiveRobotics

tmpvar, 2015-10-12:
@CreateiveRobotics happy you figured it out!