grbl/grbl #814: Feature suggestion for homing cycle (detect lost steps)

109JB, 2015-10-10:
A feature I think would be useful for grbl is a "Lost steps" reporting feature. Generally, a stepper based system is open loop and the controller has no idea if steps are lost. Although there is no way to detect lost steps while running, it is possible to detect then during a homing sequence. At least to the repeatability of the switches. Here is how I would envision this working.

1. The GRBL controlled machine is homed to the switches
2. If homed again, GRBL could report the difference in machine position from the previous homing cycle before resetting the machine home position.

The position difference would essentially be the lost position. The user would have to interpret this based on the repeatability of their switch setup.

Not sure if this is feasible since I don't know how the homing feature actually works, But it would be useful, especially in terms of setup and troubleshooting.

controlzone, 2015-10-13:
I think this would be great, and surely not too hard to implement or even increase the size (that much). Maybe as a compile time option during commissioning. It would be interesting to know if you are losing a step every job or only very occassionally.
I would imagine something like: GoHome, If LimitHit Then CurrentPos-OldHome = LostSteps. 
The other option would be to put it in the GUI rather than grbl.

109JB, 2015-10-13:
The problem with having the GUI do it is the gui would have to move to trip the switch. If soft limits are enabled, this wouldn't be allowed by grbl. Also, even if soft limits aren't enabled, each axis homing sequence has it approach fast -- trip switch -- back off -- approach slower -- trip switch -- reset machine position . This is repeated for each axis. That is 2 trips per axis, 6 trips total, so 6 times that GRBL would go into alarm mode and need to be reset.

chamnit, 2015-10-14:
@109JB :  Grbl's homing routine is not compatible with this idea. During homing, Grbl doesn't track position, since it assumes it's undefined. It also doesn't bother with smooth decelerations upon detecting the limit. To do something like this would require a complete overhaul of the routine to behave more like the probe cycle routine (which is possible). 

At this point, I'm not interested in overhauling homing yet again and trying to cram it into v1.0 along with everything else. But, I'll add this to the development list for the next project. Even so, I'm not sure how useful this is, except in the case of setting up a machine. Once setup, it's likely this would only be used on occasion to help figure out problems with a job. 

That said, it's completely possible to do this same thing via probing or using a separate measurement tool, like a dial indicator. Wire in the homing switches, temporarily, to the probe pin, disable hard/soft limits, and probe the limit pin. The reported position will tell you the position relative to homed position. With a dial indicator, just fix it to somewhere in your machine, tell Grbl to move to it, and zero it out. During or after a job, move back to that same position. If the dial indicator is not zero, then the machine lost steps.

109JB, 2015-10-15:
@chamnit  
Thanks for the explanation on how homing works. I wasn't sure if it tried to keep track of position during the homing cycle. I was hoping it did. If so, it probably would have been an easy thing to implement, but I guess it doesn't. Too bad.

Using dial indicators or probes is fine for setup, but in the case of troubleshooting, when you suspect lost steps, you would not be able to find out unless you had the dial indicators set up before hand. Since the limit switches are there all the time it would make for an easy way to check whenever you had suspicion of lost steps.

Having said that, I don't think it is worth it if it is a whole lot of work, so I'm going to think about a GUI implementation some more, which brings up a few questions about soft limits. 

1. With soft limits enabled, and say a 1mm pull off, will grbl allow a command to only the pull-off position, or will it let it fully travel to the switch?

2. if the answer to #1 is that it can only go to the pull off position, can the pull-off position be set to 0?

3. Finally, can homing be used if soft limits are disabled? I think the answer is yes, 

I am on the road for work travelling right now and not near my machine to try things out.

Thanks.

chamnit, 2015-10-15:
@109JB : If you enable HOMING_FORCE_ORIGIN in config.h, Grbl will set the pull-off position as zero (and not force negative machine space). Otherwise, it sets the limit trigger point as zero. And homing can be used without soft-limits, but not vice versa.