grbl/grbl #815: Steppers only holding Torque, but don't move

thomasveit, 2015-10-12:
Hi there,
I have a problem and I hope you can help me. 
I'm building an own CNC Maschine and used GRBL 0.8 Library from this link https://github.com/Protoneer/GRBL-Arduino-Library

Now I have connected one of my easy driver with stepper pin to Arduino Pin 2 and direction pin to arduino pin 5. All other cables are connected correctly. 

If I use an own sketch the stepper moves and everything is good. But if I use grbl the stepper only holds torque and is not moving. I'm new to CNC technology and used the G-Code Command "G91 X10" to test the stepper movement but it only holds still and the easydriver gets hotter and hotter.

Could you please help me? I don't know what is wrong with it. Maybe the configuration have to be changed? Is there a way to disable the holding torque with the configuration so if it doesn't burn while I'm testing?

Kind regards
Thomas

PicEngraver, 2015-10-12:
Try: G1 X10.
G91 is not a move command.

thomasveit, 2015-10-12:
Hi,
I also tried this command and nothing happened.

Do I have to connect the enable/disable stepper pin? Because in my test sketch this was not necessary.

Kind regards
Thomas

Norwegian-Gadgetman, 2015-10-12:
That's Protoneer's GRBL library, not the GRBL G-code parser.
There's a bit of difference between them(such as one hasn't been maintained in 3 years)

Do you have any way of verifying that there's any pulses on the 'Step' pin?

Easy Driver?
Links to the model? 

I'd suggest turning the current down until you need it, so that it doesn't heat up too soon.
(Also you may want to fit bigger heatsinks or a fan)

EliteEng, 2015-10-12:
Are you 100% sure you have wired the motors correctly? Most of the problems like this are caused by incorrect wiring.

Use a multimeter to check the resistance of the wires ( do NOT go by colour coding as different motor manufacturers use different colour codes ).

You shouldn't have to connect the enable/disable pin as most of the drivers are enabled=LOW ,check the datasheet for that driver.

thomasveit, 2015-10-12:
So the Stepper is connected correct. Because if I use my own sketch where I turn the stepper with pin 2 for the step and pin 5 for the direction it works without problems. Only the grbl software makes problems. I used this library because until now I only worked with the arduino IDE with my arduino mega and I wasn't sure if grbl would overwrite the normal arduino bootloader and it is possible that I will need the Arduino for other projects sometimes. 

The driver is an EasyDriver v4.4 with A3967 IC
Stepper motors are 42BYGHW609

I've also checked the resistance on all connections and so on and as you can see with my own sketch the stepper motors turning correct. 

I have already a fan lying here and only have to connect it to my built board. Until now I only have evenrything connected for about 30 seconds because no stepper is turning. I also have only one stepper connected at this time to find out why it is not working. My board is built so that I can eject my EasyDriver to change them or disabling an axis.

thomasveit, 2015-10-12:
Vendor of the easydriver should be sparkfun. I have the Version 4.4 of these: http://www.schmalzhaus.com/EasyDriver/

Norwegian-Gadgetman, 2015-10-12:
You know that the EasyDriver doesn't do more than 750mA, while a NEMA17 like the one you list may require up to 1.7A depending on application. 

If your own code seems to work, but the GRBL Library code doesn't, I'd say the problem is in the code.
(Please note; I absolutely HATE the word 'sketch' used this way... I want to do unspeakable things to whoever decided on that)
I would suggest downloading GRBL 0.9 instead, and trying that.

thomasveit, 2015-10-12:
Ok, when I'm home from work I'll try this today and give a feedback if it worked or not. 

PS: I also just using "sketch" because it's the official name for this. I'm Java Developer and for me every "sketch" is an own "project" or simply "source file".

X3msnake, 2015-10-12:
Sily question but, did you unlock the GRBL with a $X first?

https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

thomasveit, 2015-10-12:
No I didn't because as I said, I'm new to this and somehow I never found it that I have to send this command :)

PicEngraver, 2015-10-12:
Also, Try: G1 X10 F1270.

I don't know if all versions of GRBL have a default feed rate set, but the latest version does not, so an Fx command will be needed for a test.

109JB, 2015-10-12:
The default feedrate should be F=0. So you will need a feedrate with a G1 command. The other question is what interface are you using and do you have the baud rate set right. You can sometimes get a connection with the baud rate set wrong, but the computer and Arduino won't understand each other. What interface are you using to send the commands?

Norwegian-Gadgetman, 2015-10-12:
Seems you replied in the wrong thread....

Overheating while holding. 
This is caused by the fact that the drivers deliver more power to the motors when holding than when moving, and that also results in more power expended(and more heat) by the driver itself.
(actually, it delivers the same current, but now it's continuous, not in short pulses)

There's only three solutions to this:
1. Reduce the current if possible, 
2. Adding heatsinks and/or fans.
3. Disable the driver when the machine is standing still.

thomasveit, 2015-10-13:
is there a way to disable the driver the driver while there is nothing to do? Maybe with the enable/disable pin? If it is so then does grbl set this pin automatically before starting a command and after it is finished?

109JB, 2015-10-13:
You have to connect the enable pin and then set $2=(something less than 255)

this will disable the drivers when idle, but note that without power, the motors can drift given the right circumstances for example, the Z-axis on my mill will slowly lower the head if power is take off and the gibs are a bit loose..  Not sure about the easy drivers, but many stepper drivers have idle current reduction where they go to a lower power output state during idle when compared to running. The advantage here is that the motors still have some power applied at all times so that they can maintain their position.

EliteEng, 2015-10-19:
Does your driver actually overheat or just heat up.

In normal operation these drivers get hot ( to hot to touch ) but they will happily run all day if you dont draw to much current from them ( more than 750mA for this driver ).

I think these chips have thermal protection ( around 165 deg C ) so if they overheat they will just cutout until they cool down. I have managed to get the drivers into thermal shutdown a few times from driving them way to hard but they have all survived.

Heatsinks help and are cheap.

Reducing the current helps a lot. Just because the driver can output a certain current dosn't mean you actually need that amount of current to move/hold the steppers.

It all comes down to what your use for the CNC is and how hard you are pushing it.

thomasveit, 2015-10-19:
They only got hot. I've solved all Problems until now with latest 0.9 Version of GRBL, Arduino UNO instead of my MEGA and using copper heat sinks in combination with a big fan on top of my interface board. So thank you all very much for your help. Your're great :)

cheers
Thomas

ankr, 2017-01-14:
@thomasveit Do you mind share what you did in order to get GRBL to work correctly? I'm in same situation with my motors holding torque but not working.

KotieSmit, 2017-02-09:
I had struggled for some time as well.
Using Arduino Duemilanove, protoneer cnc shield v3, grbl 0.9.

Initially the motors would hold torque, but when it should move, it was loosing torque, would then move and regain torque.

The fix was:  jumper over the grnd/en pins on the shield, and making sure of $4=0.

Now it working like charm.

Here is a dump of my settings:

`Grbl 0.9j ['$' for help]
>>> $$
>>> $G
(step port invert mask:00000000)
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (0000000)
$4 = 0    (step enable invert, bool)
$3 = 0    (dir port invert mask:0vert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$5 = 0    (limit pins in02 (arc tolerance, mm)
$13 = 0    (report inches, bool)
$12 = 0.0ft    (limits, bool)
$21 = 0    (hard limits, bool)
$20 = 0    (sole, bool)
$23 = 0    (homing dir invert mask:00000000)
$22 = 0    (homing cyc (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$24 = 25.000=250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 250.000    (x, step/mm)
$101 = 250.000    (y, step/mm)
$260.000 (z, step/mm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$102 = 25^2)    (Z-axis travel resolution, step/mm)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$121 = 10.000    (y accel, mm/sectravel, mm)`

duronebis, 2017-02-09:
Is there a simple way to limit the max frequency of the stepper timer in the source code?
I applied 2 more axes and now for high speed the position data (and other...) get corrupted.
I know that I can limit the max speed of the motor, but if the user then try a too high value now the application almost crash.
I just ask for a simple way to limit for example to 1/2 the max frequency of the stepper interrupt, without touching the speed computation and other complex things.
Thanks to all!

mbe2014, 2017-10-13:
Hope this will help someone:
I have had a similar problem. Engine hold torque, made noises, jugged but did not rotate.
It turned out I used an off the shelf motor cable (with plugs on both sides).
It turned out the wires were: (as came out from my engine) a1,a2,b1,b2, 
However the Shield order is a2,a1,b1,b2 - I switch the wires and all worked.

kot9pko, 2020-01-05:
I'm using cnc shield v4 with Arduino nano. In my case step pins were connected to direction pins and vise versa. I solved this issue by switching it programmatically in `grbl/cpu_map/cpu_map_atmega328p.h` (left in comments their original values):

```
// Define step pulse output pins. NOTE: All step bit pins must be on the same port.
#define STEP_DDR        DDRD
#define STEP_PORT       PORTD
#define X_STEP_BIT      5  // Uno Digital Pin 2
#define Y_STEP_BIT      6  // Uno Digital Pin 3
#define Z_STEP_BIT      7  // Uno Digital Pin 4
#define STEP_MASK       ((1<<X_STEP_BIT)|(1<<Y_STEP_BIT)|(1<<Z_STEP_BIT)) // All step bits

// Define step direction output pins. NOTE: All direction pins must be on the same port.
#define DIRECTION_DDR     DDRD
#define DIRECTION_PORT    PORTD
#define X_DIRECTION_BIT   2  // Uno Digital Pin 5
#define Y_DIRECTION_BIT   3  // Uno Digital Pin 6
#define Z_DIRECTION_BIT   4  // Uno Digital Pin 7
#define DIRECTION_MASK    ((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT)|(1<<Z_DIRECTION_BIT)) // All direction bits
```

gururajkashikar, 2020-03-24:
> I'm using cnc shield v4 with Arduino nano. In my case step pins were connected to direction pins and vise versa. I solved this issue by switching it programmatically in [...]

Hi kot9pko,

I observed the same. Dir and Step pins were swapped in my "Cnc nano shield V4" with DRV8825 drivers. I modified as you have suggested above in the cpu_map_atmega328p.h file. 
Opened Arduino Studio -> Navigated to grblUpload example, Compiled and Uploaded it to Nano. But the issue did not go away. How can I make sure the changes made in cpu_map_atmega328p.h is actually flashed into Arduino Nano board.

Thanks in Advance.
-Regards
Gururaj

whitetd, 2020-03-24:
Are you selecting 328, or nano for your board?

gururajkashikar, 2020-03-25:
Thanks for the reply whited. I am sure I am selecting Arduino Nano for the board. What I am asking here is, if we make changes to any of the grbl file ( like *.h file)  will it be picked up by Android studio when we compile grblUpload project. Does the hex file generated, will include our changes and be flashed to target board. In fact I have another nano board with grbl 0.9. if I insert that to the CNC shield everything works fine.

whitetd, 2020-03-25:
If you selected nano then the changes to the cpu map needs to be in the
nano section.

as far as studio no idea.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gururajkashikar, 2020-03-25:
I found a way to know if the changes made by me is picked up. The idea is to purposefully make some wrong entry in the file you want to be flashed. In my case I wrote text "Gururaj" into a random line of  cpu_map_atmega328p.h file. Then when I compiled in  Arduino IDE (sorry for earlier post where I mentioned this as Android Studio.) I did not see any error. That showed its not picking my modified file kept at C:\Mydir\grbl-master\grbl

Solution :
1) In  Arduino IDE, Under the "Sketch" menu, click  "Show Sketch Folder". This took me to another directory as shown below. 
      **C:\Users\Lenovo\Documents\Arduino\libraries\grbl**
2) I made the all the changes (including the error line) in the cpu_map_atmega328p.h under this directory. 
3) When I tried to compile, I could see lot of error starting from the line where I had inserted "Gururaj" text.
4) Removed the erroneous line, re-complied grblUpload project in Arduino IDE, hex file got generated.
5)  Uploaded the program/hex file to Arduino Nano and everything was working fine.

Thanks to all.

Regards,
Guru