grbl/grbl #816: Possible response problem from GRBL

109JB, 2015-10-12:
I was running a rather large file (1,049,930 lines) through my GUI for testing  and about 800,000 lines in, I got an error. What it turned out to be is the status response from GRBL came back with the echo (I have echo enabled in config.h) on a single line with no end of line character. The line I got was as follows:

<Run,MPos:3.1925,3.5375,1.1634,WPos:3.1925,3.5375,1.1634,Buf:17,RX:[echo: X2.5546Z0.8951]

I was running in character counting streaming mode, so the planner buffer was full or nearly so.  That was actually what I was testing, was how full the buffer stayed in a long program with itty bitty moves at high speed. The re-run showed that except for the start and end of the program where the buffer is initially filling or emptying the last lines, the buffer remains at 16 or 17 over 99% of the run.

 My GUI reads the serial stream using the readline function in VB.NET, so it reads until it sees an EOL character.The GUI choked because the line it read was not in a format that it recognized. Also, since the string received had the RX buffer info cut off, it looks like for some reason GRBL may not have finished the status report before sending the echo. The other possibility is that the VB readline function experienced a glitch. Since it happened 800,000 lines into a 1+million line code file, it is pretty impossible to reproduce but I thought I would bring this to attention. I looked over my code and I don't see anything in the GUI code that would be a problem. As a side note, I reran the program and it completed without problem, so I guess it could have been a really coincidental problem related to the timing of the status request, and where grbl was at in relation to running a new line, responding to the echo, etc.

chamnit, 2015-10-12:
@109JB : This doesn't seem like the size of the file is relevant for this problem, but rather, it allowed more time to pass to uncover a rare, random error. The serial write code has a wait-if-full statement when writing anything back to a GUI. So this makes no sense in terms of how Grbl's source is written. 

But, I did recently find an issue with placing volatile variables inside a struct. The volatile is undefined in this case, so really weird things started to crop up on the most recent development builds when flash and memory were getting close to maxed out (the default master build seemed to be ok though). For example, status reports would just print garbled values, occasional random soft-resets, g-code parser states not updating correctly, etc. I issued a fix for this last week. If you haven't updated, you may be running into this problem with Grbl's echo enabled.

109JB, 2015-10-12:
Thanks.  I did see some weird characters on a $I request in the previous version I had loaded and I loaded a new version. What I was seeing was the version response with a bunch of ????? after it before the closing bracket.

I just ran a $I request on the version currently loaded and get this:

[0.9j.20150930:4G0]

Is this correct for a version response? The G0 at the end looks like the tail end of my $N0 which is:

$N0=G90G20G54G0

chamnit, 2015-10-12:
@109JB : Strange and stranger. This shouldn't be happening either. If you send an `$I=` command to clear the $I build info string, does the N0 get truncated? This would mean that the EEPROM mapping is off for some reason. 

Secondly, are you still using an Arduino Nano? Did you end up updating the boot loader to Optiboot on it? I wonder if its an issue with boot loaders and remaining flash space.

thomasveit, 2015-10-12:
Solved the problem. 
Switched from my new Arduino MEGA to my old Arduino UNO and now it is working :)

One last question: The holding torque function extremely heats my easy driver. When moving they are not getting so hot. Is it possible to regulate this function so that it is not so strong?

109JB, 2015-10-12:
$N0 doesn't get truncated. It looks fine.  I just tried a couple things. First, I decided to try it in the Arduino IDE serial monitor to eliminate my GUI as a possibility.  Of course on the IDE, it performs a soft reset automatically. I then ran $I with the same results above. I then sent a $N0= to clear that, and ran $I again. Same result. Then I performed a hard reset, and ran $I with the same result. I checked $N0 and grbl reports it is empty now, as is $N1.

I am still using a nano, but I never did get around to doing the optiboot loader for it. I will download the most recent grbl and reflash and check again. I'll report back. I'll also check what the IDE says for flash space used too.

109JB, 2015-10-12:
I downloaded and reflashed and this is what I get now from a $I followed by a $N with echo turned off.

Grbl 0.9j ['$' for help]
[0.9j.20150930:4G0]
ok
$N0=
$N1=
ok

I then used the Arduino IDE to completely clear the EEPROM and reflashed GRBL. That appears to have corrected it as now from a $I after a soft reset, I get:

Grbl 0.9j ['$' for help]
[0.9j.20150930:]
ok

I resent my $N0=G90G20G54G0 string and reran $I and all looks good now.

What could cause this eeprom data to become corrupt like this?
:

chamnit, 2015-10-12:
Not exactly sure how this could have happened. This EEPROM string remains there until over-written or cleared, so it could have happened at anytime in the past. It's possible that a prior $N0= write command wrote the incorrect place in EEPROM, either via this bug I spoke about, or something else. Or, somehow Grbl received a `$I=4G0` command at some point. That said, I didn't install too many proper input checks into Grbl, because they take up so much flash space. With the $Nx and $I command especially. These aren't used very often and once setup, users don't typically mess with them.

closed 2015-11-12 by 109JB