grbl/grbl #817: Invalid gcode ID: 33 - should stop further processing

faultylee, 2015-10-14:
Hi guys, great work on grbl, been using it for my shapeoko2 for a while and love it. 

Similar to grbl/grbl#619, I get the same error on a similar line of code, but I believe grbl should stop processing instead of skipping ahead, it end up cutting a totally wrong shape, further more offset the 2nd part after this by a lot. 
```
X34.431 Y97.611 Z-0.4 I28000609.344 J2449736.304 
G02 X42.057 Y10.446 Z-0.4 I-27059770.111 J-2367507.587 < error: Invalid gcode ID:33
G03 X43.379 Y10.03 Z-0.4 I0.747 J0.065 < error: Invalid gcode ID:33
G02 X54.697 Y23.53 Z-0.4 I67916.435 J-56925.766 < error: Invalid gcode ID:33
G03 X66.016 Y37.03 Z-0.4 I-167619.993 J140551.402 < error: Invalid gcode ID:33
X66.056 Y37.082 Z-0.4 I-0.574 J0.481 
```
I'm on 0.9g, with UGS 1.0.8 and I checked the setting and output, the gcode was not altered in any way. Verified with the visualizer in UGS as well as here http://fablabamersfoort.nl/gcodevisualizer/ that the gcode is correct. Gcode produced by Heekscnc using modified emc2 post-processor modded for grbl+shapeoko2. Been using it for a year now and have not notice much issues.

In the picture below, to the left is what's supposed to come out, the right is the result after `Invalid gcode ID: 33` and continuing. (Please ignore the weird cut line, my vacuum pull up the piece and the spindle jammed onto it :), I added wider tags after that)

[image]

109JB, 2015-10-14:
The problem is not GRBL. Quite frankly, it is the program you are using to send to GRBL. Once GRBL sends an error response, the interface should immediately pause GRBL, stop sending lines of G-code, and then reset GRBL which will clear anything in the planner buffer.  

As far as the error, the wiki description is below. You said you used a modified postprocessor in a cam program. Likely the I and J values are in G90.1 absolute coordinates. GRBL doesn't support that. The I and J have to be in G91.1 incremental values. A quick check with my calculator shows that this is what is happening.

33: The motion command has an invalid target. G2, G3, and G38.2 generates this error. For both probing and arcs traced with the radius definition, the current position cannot be the same as the target. This also errors when the arc is mathematically impossible to trace, where the current position, the target position, and the radius of the arc doesn't define a valid arc.

faultylee, 2015-10-14:
@109JB thanks for the reply. Probably need to feedback to UGS' guys here too. I was expecting a halt much like hitting the limit. If the planner buffer is not cleared in time, GRBL will still carry on.

As for the cam postprocessor, I didn't change anything important (I'm not expert enough to make sure it doesn't break anything). I'm only removing the support for a few G-code that GRBL doesn't support so an alternative build-in method is use to generate a proper G-code.

This problem only happened when I rotated the piece by 5deg to fit into my work piece. I just cut it again straight, it cut fine.

chamnit, 2015-10-14:
@faultylee : Grbl's streaming protocol can't distinguish what the intent of the g-code is. For example, manually typing in commands or streaming something. In either case, it's up to the user and the GUI to determine what to do when encountering an error in the g-code. It's possible for Grbl to halt everything automatically upon an error, but this will greatly decrease the usability of it (and annoy hordes of users).

The easiest thing to do is to have the GUI stop streaming when GRbl returns an error. Even if there are motions queued in Grbl's motion planner, these motions are all valid and the machine will stop at the point where the stream stopped.

109JB, 2015-10-14:
@chamnit  
I'm either misunderstanding, or I disagree with this statement:

"The easiest thing to do is to have the GUI stop streaming when GRbl returns an error. Even if there are motions queued in Grbl's motion planner, these motions are all valid and the machine will stop at the point where the stream stopped. "

More must to be done than simply stopping streaming or disastrous results could happen. For example, I hand code a lot. lets say I type a g-code program that has the following block of code:

=== some previous G-code here  that has Z at Z=  -1 inch ===
N011G90G0Z1  <<< Raises Z to 1 inch above the part to clear my vise
N012 G0X5         <<< Rapid to the other end of my part
=== More code here =====

But I mistype the code and it comes out like this:

=== some previous G-code here  that has Z at Z=  -1 inch ===
N011G900Z1  <<< Missing the second "G" keystroke
N012 G0X5         <<< Rapid to the other end of my part
=== More code here =====

This would be an extremely simple error to make.  What is going to happen if you simply stop streaming is an error will be thrown at N011, but the GUI had already sent N012 and it is in the GRBL planner buffer. What will happen is the error will be thrown, but the movement to the clear height of Z=1 will not happen due to the error. Then, the N012 line will be run with the tool below safe height and crash it into my vise. So, the GUI just stopping streaming in this case will still have the machine crashing into my vise.

What needs to be done is stop motion as fast as possible, stop streaming from the GUI, and clear the GRBL buffers. What I have done in my GUI is as follows:

1 - error reported by GRBL and received by GUI
2 - GUI issues an immediate "!" hold command that stops machine motion. This was the fastest way I could find to halt motion since "!" is picked out of the stream and interpreted immediately by GRBL.
3 - GUI exits the streaming loop to stop transmission of any more lines of G-code
4 - GUI issues ctrl-X command to reset GRBL thereby clearing both the planner and receive buffers.

Really, my parsing routine would actually catch the above error and never even send it to GRBL, but in case there are some error cases that I missed or didn't consider in my parser, I still have the above enabled. The above steps probably aren't the only way to do it, but in my view it is imperative that an error halts motion immediately and prevents all further motion, whether that is from the GUI sending more commands, or from commands already in the GRBL buffers.

faultylee, 2015-10-15:
May be there should be a distinguish distinction for error, or rather fatal error, like soft/hard limit. There's errors which is safe to be ignored and some not. When I first turn on the soft limit, I face usability problem at first, the machine halting, and I still have to go figure out what causes it and eventually learnt to place the work piece properly and setting the limit correctly. I don't see how ignoring an arc which is part of a complete Gcode.

I think it's the same here with other potentially catastrophic error, at first it will introduce usability issue  as the user have to find out where the problem is, but eventually we'll learn to correct our mistaken (in my case the CAM software) and usability will not be an issue anymore. I would rather be safe then sorry.

Probably for those Gcode that can be safely ignored, it can be responded as "warning" so as to differentiate from fatal/error which should halt. I didn't go through all the discussion here and in the shapeoko forum, but I bet there's been debate around this. I did see some debate on thing that should be handle by GRBL vs the GUI.

Just my honest feedback, no offence on the great work and time you guys have put into GRBL :)

faultylee, 2015-10-15:
Any currently known GUI that handle the error properly or at least user can choose to halt?

109JB, 2015-10-15:
@faultylee 
I disagree with the premise that there are any G-code errors that can be ignored. There is no way to know if a g-code is potentially catastrophic or not. EVERY G-code error should be treated as catastrophic. 

Bottom line is that there is nothing in  GRBL needs to be changed. Plain and simple, the GUI's have to handle the errors.

As for a g GUI that handles errors correctly, I believe mine does. You can try it out by downloading it at the following link. 

https://dl.orangedox.com/EqqZYCIOLOeHIrjLc6/109JBs%20GRBL%20Interface.zip

If you decide to try it, please read the manual I put together before asking questions, and don't ask questions about it here. This place is for GRBL, not my GUI. If after reading the manual you have questions, you can e-mail me using the link in the help pull-down in the program.

chamnit, 2015-10-15:
@109JB : Sorry I should have been more clear. The simple send-and-wait-for-response protocol works fine by not sending any more commands after an error is returned. This is because the GUI only sends more commands when it receives an acknowledgement. The character-counting protocol can cause the commands stored in the serial RX buffer to execute automatically. I suppose this part needs to be fixed.

@faultylee : @109JB is correct in stating that ALL g-code errors must halt. It's because g-code is highly modal and if one is missed, the whole program is comprimised. I believe bCNC does have an option for error handling as well.

That said, Grbl also has a check g-code mode, invoked by a `$C` command. You can stream the entire program and Grbl will parse it as it would normally, but not move anything. It's there as a way for users and GUIs to check for errors before running a job.

As for error states, Grbl already has a fatal error. It's Grbl's alarm mode. This is used when things go really bad.

109JB, 2015-10-15:
@chamnit 
Personally I think any automatic error detect - halt within GRBL would be annoying from a user perspective. For example a mistype in a MDI window triggering a halt which requires a user interaction before being able to re-issue a corrected MDI command would be annoying to me.

What you describe though sounds like it could be easily written into a GUI. Actually it sounds like just issuing a feed hold "!" after an error. A feed hold would stop the machine and if the user decided the error could be ignored, they could just hit "~" cycle start/resume and off it goes again.

Even if implemented within GRBL, the GUI still has to be written properly. Lets say the GUI is in the middle of streaming a G-code file and GRBL detects an error and halts. To stop the streaming the GUI author still has to exit the streaming loop, otherwise once another ok response is received, it will start the streaming again.

My opinion is that the proper place to handle error response is in the GUI. It is quite easy and the GUI authors need to step up and handle the errors correctly. In my GUI, the meat of it is handled by only 9 additional lines of code. GUI authors don't really have an excuse for not having it in there.

Here are the important lines in my code (Visual Basic)

========Define the global variable to use as a flag for loop exit =======
Public BreakLoop as Boolean = False

==== In the serial port data received event ====
If string.contains("error") then           <<< Checks incoming stream from GRBL for the word error
        SerialPort.Write("!")                    <<< Send the feed hold command
        BreakLoop= true                       <<< Set the flag for exiting the GUI's streaming loop
        Sleep(100)                                 <<< Wait for 0.1 seconds to prevent an alarm from GRBL
        SerialPort.Write(Chr(24))          <<< Send the ctrl-X reset command
End If
=======================================

====== in the GUI's streaming loop =============
BreakLoop = False                            <<< Initializes this variable so that the loop doesn't exit right away
(Start of streaming loop)                               <<< the existing streaming loop. 
          (Other commands here)
          If BreakLoop = True then Exit For      <<< the if statement that stops further streaming
          (Other commands here)
(End of streaming loop)
========================================

faultylee, 2015-10-15:
@109JB Thanks, too bad I'm running Linux in both the main laptop and the one that stream the gcode to GRBL. No wonder I didn't notice your application before.

faultylee, 2015-10-15:
@chamnit $C makes sense now, so I'll in-cooperate that into my CNC workflow, thanks again.

109JB, 2015-10-15:
If it had USB access you could probably run it in a virtual machine.

faultylee, 2015-10-15:
The streaming laptop is an EeePC. Now that I know what to look for, I'll find an alternative, or worse case modify UGS myself

closed 2015-10-15 by faultylee

LETARTARE, 2015-10-15:
Hello,
I think the first work to be done on a gcode file is to make it FULLY COMPATIBLE with Grbl.
Then can be treated transmission errors.
The simple 'call-response' protocol seems the most robust for transmitting gcode file.
The character-counting protocol  should be used only for mode' Check '.

109JB, 2015-10-15:
I disagree.  Character counting can be just as robust as the send response. I have tested my GUI with character counting on gcode files over 1 million lines in length and it performs flawlessly. There it's no reason that if should be relegated to check mode only.

As for error checking, yes the user should try to make sure the code is good, but when grbl throws an error, the GUI should not ignore it.

chamnit, 2015-10-15:
@109JB : I think @LETARTARE means that the character counting protocol has some * stipulations associated with it. Meaning that you don't have control over what's in the serial RX buffer. If Grbl throws an error, Grbl will still parse and execute what's in there, even if you stopped streaming. Not so with the send-response protocol. So, it's best to check the g-code for errors prior to streaming it to Grbl via the character-counting protocol.

LETARTARE, 2015-10-15:
@109JB :
I use both protocols interchangeably. The protocol 'Call-response' is therefore easier to write more robust in terms of programming.
I am interested in a code of more than 1 million lines: we can I get it ?
Second, I propose to define a compatible file extension Grbl , for example  : 'grnc' or other, to distinguish more easily made compatible Grbl other files.

109JB, 2015-10-16:
@LETARTARE - I guess we have a different definition of "robust".  I view it in terms of "a more robust" program is one less likely to experience errors, or can handle those errors better than a program that is "less robust".  In this sense, there is no difference between a properly implemented character counting protocol and a send response protocol. One can be made as robust under this definition as the other. While it is true that the send-response protocol is easier to write the code for, that in and of itself doesn't by itself make it more robust. I would judge my character counting protocol as being equal in terms of this definition of robustness  to my send-response protocol.

As for the 1 million line g-code file, The link below is to the file in my dropbox. It is a shorter g-code file that was pasted as many times as it took to get over 1 million lines. I think it started out as a 40,000 lines or so.

https://www.dropbox.com/s/etq8tdtpaszlzd1/1-million%20line%20file.nc?dl=0

@chamnit - What you said about not having control over what is in the RX buffer during the character counting portocol is true, but not entirely so. If you just stop streaming, then the commands remaining in the RX buffer will be executed. I guess this is what you mean by having no control over that, But this is why I said that the GUI can't JUST stop streaming. A hold, streaming stop, and a reset, in that order stops motion fast and flushes the planner and RX buffer of ALL lingering commands, so there is some control that can be had over what is in GRBL's buffers including the RX buffer.  In terms of an error detect, this is all the control a GUI really needs.

AlexHolden, 2015-10-16:
It seems to me that when using the call-response protocol, it makes the most sense to stop streaming when we see the error response and allow grbl to finish executing everything up to the error line, rather than aborting part-way through the current move because we've detected that a future block has an error. Maybe also automatically send commands to turn off the spindle and coolant (I haven't implemented that yet but it seems like a good idea).

The character-counting protocol is more problematic because if you just stop streaming when you see the error, there will probably be more blocks waiting to be executed in grbl's serial buffer and we don't have a way to flush them other than rebooting grbl.

In my GUI I implemented a hybrid protocol mode where I use character-counting when grbl is in check mode (it massively speeds up the check time) and call-response otherwise. This seems a good compromise to me because at 115200 baud the call-response is generally fast enough when actually cutting.

LETARTARE, 2015-10-16:
@109JB :
thanks for the file, I'll test it according to the two protocols.
Defining a robust software :
https://en.wikipedia.org/wiki/Robustness_%28computer_science%29
@AlexHolden :
I do the same.
@chamnit :
what do you think define a compatible file extension Grbl?

109JB, 2015-10-17:
@LETARTARE  - Be advised that the 1 million line code I put up may have some code that is not compatible with native GRBL allowed G-code commands. I can't remember how I made it. My GUI supports some things GRBL doesn't natively, so it may throw an error if simply streamed as is to GRBL.

Thinking about things as I was on a plane back home tonight, I thought that you may be considering send-response "more robust" because you prefer the way simply stopping streaming works for error detection. I am thinking about it from the point of "Is character counting more prone to crash the GUI during a G-code run?" From this perspective, it is not, at least not in the way I have it implemented. I think either way of looking at it fits the definition of "robust", just had to figure out where each was coming from.

@AlexHolden While I agree that the send response is quite adequate most of the time, there are occasions where the character counting will provide benefits. One example is when running high feed rates while running 3D machining operations created in CAM. These operations often include very short movements stacked together with this repeated over and over in the program. I have done a lot of testing in this regard on my machine and have found that although run time benefit isn't great, there is some, but the biggest benefit is obviously smoother operation of the machine. In one of my test programs, a feature that creates a bunch of these short moves will results in starving GRBL's planner buffer using the send response resulting in perceptibly "jerkier" motion due to the lack of look ahead, and on slower computers, even having GRBL go idle for a some period of milliseconds because it is waiting for data. I can see this on my status monitoring on my GUI. Using the character counting protocol on this same CAM generated file results in a bit shorter cycle time, but the big benefit is obviously smoother operation through the sections where there are a lot of short moves. The character counting allows the GUI to catch up on sections with longer moves and the result is that GRBL doesn't go idle and actually maintains a planner buffer that is at least 1/2 full and usually at 16 or 17 commands in the planner buffer. Granted that my GUI has a lot more going on than just getting the commands from the file and sending them to GRBL due to my implementation of canned cycles, tool changes and the like. A GUI with a simple streaming scheme might not notice these effects like mine does.  In any case, I leave it up to the user which protocol they want to use as it is selectable in my "User Settings" window.

As for error detection and the send-response vs character counting, I agree that simply stopping streaming in send response is fine. I think I will actually change it in my GUI. In terms of character counting, I don't think it is a big deal to halt at the first indication of an error even though the planner has commands left in it.  I say this because if there is an error, no matter where the program is halted, the user will have to correct the error and re-run the G-code. This small detriment may be outweighed by benefits of the character counting protocol, but is probably something that the user should be able to decide.

@chamnit This discussion (a really good one in my opinion) has brought up one thing that could be done in terms of GRBL error detection, and that would be to simply have GRBL flush only the RX buffer on encountering an error, or provide a way that GUI authors can flush that buffer. If this were to happen, then the GUI could, upon detecting an error, simply stop streaming in either the send -response or character counting protocols with the same result, that the planner buffer moves would still run up to the line that had the error. 

Thanks guys. going to work on implementing G90.1 now.

chamnit, 2015-10-19:
@109JB : Not sure if this is the right solution either. If you automatically flush the RX buffer, the GUI has to know exactly what it needs to re-send, either by the GUI tracking the replies properly or Grbl replying with a special message notifying which line has the problem. For the latter, this would require some modification to the protocol further, like appending a line received value to each ok or error statement. 

Not sure that I want to go down that rabbit hole now, but I think this will be needed eventually. 

@LETARTARE : There is no reason for a file extension for Grbl, because it doesn't have file dependencies. You just stream g-code to it. It's the GUI that manages the files.

109JB, 2015-10-19:
@chamnit 
Correct me if I am wrong, but if GRBL detects an error, then all of the commands that remain in the RX buffer are commands that followed the line with the error. Correct?

 It may just be my opinion, but once an error is encountered, streaming should stop at (or before) the line with the error. The GUI can stop sending, and if GRBL flushed the RX buffer, that would take care of the the lines after the error that had already been sent by the GUI.

chamnit, 2015-10-19:
@109JB : You're absolutely right. But like I said before, the GUI needs to know where it needs to resend because what's in the RX buffer can be several blocks ahead. The GUi will need to track that and I'd like to offload as much as possible from what a GUi has to do.