grbl/grbl #819: Homing issues with CoreXY kinematics

schlotzz, 2015-10-14:
I've used a fork of grbl 0.8c to control my CoreXY kinematics based laser cutter for a while now without problems, but I had some issues with the homing routine in the new grbl 0.9i.

In my config.h file `#define HOMING_CYCLE_0 (1<<X_AXIS)` and `#define HOMING_CYCLE_1 (1<<Y_AXIS)` were set. After the first bump into the homing switch on X axis the effector gets retracted not straight backwards, but in a strange diagonal way. Further moves were wrong/diagonal, too.

I tried to apply a small fix with commit #818 . Hopefully it's within grbl coding guidelines...

chamnit, 2015-10-16:
@schlotzz : Thanks. Fixing the CoreXY stuff has been on the list of things to do. Please be patient. I've very little time to work on code for the past few months.

SloMusti, 2015-10-28:
I am experiencing the exact same problem with grbl 0.9i

schlotzz, 2015-10-28:
@SloMusti : Did you try my little fix? Maybe it does work for you. I'm looking forward to hearing if it works for others, too.

schlotzz, 2015-10-28:
@chamnit : Thanks a lot for maintaining this great piece of open source software, even if your time is very limited at the moment...

SloMusti, 2015-10-28:
@schlotzz works for me when applied to master branch
@chamnit thank you for grbl :)

brunobasto, 2016-04-15:
hey guys, I noticed this issue is still open. Is the fix officially applied to master?

roglio, 2016-07-25:
Hi, with my fellows at Arduino User Group - Cagliari (http://www.augc.it) we built an AxiDraw clone based on a design found on thingiverse (http://www.thingiverse.com/thing:1514145). Based on grbl compiled for a CoreXY machine, it works like a charm but homing (with switches on MIN X & Y) . Does there is a chance that this fix will be merged soon?

chamnit, 2016-07-25:
@roglio : I'll take a look at this now. It's been on the to-do list for a while, but I don't have a CoreXY machine to test it on. So, please report back on whether or not it works.

roglio, 2016-07-25:
Hi @chamnit, I'll be glad to test it on my drawing machine. I just merged the @schlotzz patch to the latest grbl version and I'm going to test it. Btw I'll wait for your patch because my vision of the project is limited only on this issue.
Thanks.

chamnit, 2016-07-25:
@roglio : I posted a bug fix on a temporary branch called  'master-corexy'. Please download that version and test it. I found another oddity in the homing and don't remember if it was intentional or not. Let me know how it works out.

Please note, that the CoreXY configuration is extremely important to it working right. Make sure your motors are placed and rotate exactly how the [CoreXY theory diagram](http://www.corexy.com/theory.html) shows it. Otherwise, your machine will move in different directions than you expect. Please double check this.

roglio, 2016-07-25:
Hi @chamnit, I just tested it and doesn't work properly. First I modified your config.h to comply to the homing procedure related to CoreXY, changing the two defines in this way:
// #define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
// #define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)

In first instance seems to work for X axis, but once homing the Y axis grbl ignores my $23=3 configuration (invert homing direction). I also set $22=1; other config are defaults.

My patch surprisingly worked! Merging the @schlotzz code to the latest, behaved as expected. I modified only the limit.c (see attachment).

Btw, your patch is advisable because mine is just a "hack".

[grbl_roglio.zip](https://github.com/grbl/grbl/files/382146/grbl_roglio.zip)

chamnit, 2016-07-25:
@roglio : Thanks for testing it out. Not surprising that @schlotzz's code works, since it's already been tested to work. I took a look at the code and found a mistake. I flipped the axes on one of the computations. Try it now. It should work.

roglio, 2016-07-26:
@chamnit sorry for being late. It works!!! Thank you so much :)

chamnit, 2016-07-26:
@roglio : No problem. Thanks for helping out. I'll merge this update and close this issue.

closed 2016-07-26 by chamnit

sefatek, 2017-07-18:
thanks a lot this topic really helped me