grbl/grbl #820: Why does $$ require idle condition?

J-Dunn, 2015-10-14:
Issuing a $$ request gets a "not idle" response if the machine is moving. Why?

There does not seem to be anything in there that is not valid and independent of axis movement. OTH, any parameter setting commands beginning with $ could legitimately be refused during movement of the related axis. 

During manual control it would be useful to be able to interrogate the machine parameters during a possibly long movement as a confirmation or as  a prelude to composing a further command. Is the $$ refusal just a blanket block on all $ commands, or is there a reason?

chamnit, 2015-10-14:
@J-Dunn : Yes. There is a very good reason. It takes too long to print and send the message.

closed 2015-10-14 by chamnit

J-Dunn, 2015-10-15:
```
      if ( sys.state & (STATE_CYCLE | STATE_HOLD) ) { return(STATUS_IDLE_ERROR); }
         // Block during cycle. Takes too long to print.
```
Ah yes, I was forgetting I'd read that comment. 

```
  // Wait until there is space in the buffer
  while (next_head == serial_tx_buffer_tail) { 
```
Yes, I can see that would rather load the processor if the 128 byte buffer is full !
It's going to lock up the chip except for the basic GRBL timers. You may as well simply poll UDRE0   if this is all it's doing.

It has all the overhead of calling the ISR just to write one IO register.

 In fact, having taken off the trap on $$, and doing a quick test with a slow step rate, it seems to stop the motor all together, which is a bit odd since USDE int is lower priority than the timers.

Perhaps this could be done in a more efficient way.  There's some pretty sharp,  optimised code in GRBL, this bit seems a bit of a kludge. ( Plenty of  TODOs around there so my be wants some help. )

Is there some non interrupt stepper code that is getting blocked by this loop? 

Thanks

J-Dunn, 2015-10-15:
digging around to understand the TX blocking, I found a small code size gain:

```
  // enable rx and tx
// UCSR0B |= 1<<RXEN0;
//  UCSR0B |= 1<<TXEN0;
	
  // enable interrupt on complete reception of a byte
//  UCSR0B |= 1<<RXCIE0;
  UCSR0B |= 1<<RXCIE0 | 1<<TXEN0 |1<<RXEN0;
```

12 bytes. Every little counts ;)

chamnit, 2015-10-15:
@J-Dunn : Thanks for the code tip.

J-Dunn, 2015-10-15:
You're welcome. ;)
Can you point out what I'm missing with the serial blocking?  It don't see why ISR(TIMER1_COMPA_vect) et al don't seem to get called when interrupts are re-enalbed coming out of ISR(SERIAL_UDRE). Is there something in the main thread that needs to happen to keep the pulses going?

A quick pointer would be helpful.

chamnit, 2015-10-15:
@J-Dunn : See line 91 in serial.c.

J-Dunn, 2015-10-15:
Thanks, I realise this is currently slow and inefficient and I'll probably write something to at least handle the PGMstrings in one hit. But I'd still expect to get at least one pulse from the Timers in between each TX byte, however it seems block completely.

Once it's in constant speed segment, shouldn't the timers just keep firing and pulsing the STEP output?

J-Dunn, 2015-10-15:
```
The number of segments "checked-out" from the segment buffer and the number of segments in
   the segment buffer are sized such that no operation in the main program takes
   longer than the time it takes the stepper algorithm to empty the segment buffer. 
```
EXCEPT for serial output ! 
```
   Currently, the segment buffer conservatively holds roughly up to 40-50 msec of steps.
```
So st_prep_buffer() needs to be called every 40ms and is in the main thread getting blocked by while (next_head == serial_tx_buffer_tail) 

I'll have a think whether it's worth  the effort of refactoring that.