grbl/grbl #821: Documentation comments.

J-Dunn, 2015-10-15:
GRBL are generally copious, clear and well-written, however sometimes they can be confusing. Some time back I'd said I'd provide some suggestions once I got a clearer understanding of the code base. 

I'll use this issue as drop box for suggestions as I come across the them. 

protocol.c
   // cycle reinitializations. The stepper path should continue exactly as if nothing has happened.   
   // cycle reinitializations. The stepper path should continue exactly where it left off.   

Since there is decel, stop and accel the cut will not be "exactly as if nothing has happened. "

chamnit, 2015-10-15:
@J-Dunn : I suppose I am human after all. :) Just as a note, all comments don't get updated for every minor change. Some get over looked and may not mean the same thing in the new context. Anyhow, these comments are generally for me. I copiously comment every piece of code I write, especially at my day job.

J-Dunn, 2015-10-15:
Me too. Saves loads of time trying to decipher what I thought was obvious when it was fresh in my mind six months ago. However, the level of comments is what makes GRBL such a pleasure to work with.  

Hopefully some of the things that got overlooked or seem obvious to someone with an intimate knowledge of the code will become clearer to others. The major bits that confused me were in stepper.c , generally very helpful but some confusion of terms. I'm still not fully sure of it to the point that  I've got it clear enough to make fully valid suggestions on the main blocks. I'll probably further edit what I have now.

Sometimes sentences get a bit hard to follows and I have to read several  times to get what was intended. I've been trying to clarify the  language in some cases. Here's one from motion_control.c

```
// NOTE: This is the primary gateway to the grbl planner. All line motions, including arc line 
// segments, must pass through this routine before being passed to the planner. The seperation of
// mc_line and plan_buffer_line is done primarily to place non-planner-type functions from being
// in the planner and to let backlash compensation or canned cycle integration simple and direct.
```
```
// NOTE: This is the primary gateway to the grbl planner. All line motions, including arc line 
// segments, must pass through this routine before being passed to the planner. The seperation of
// mc_line and plan_buffer_line is done primarily to place non-planner-type functions outside
// the planner and to make backlash compensation and canned cycle integration simple and direct.
```

chamnit, 2015-10-15:
@J-Dunn : Well I can't make it too easy for businesses to blatantly copy things into closed-source controllers.

J-Dunn, 2015-10-15:
Yep, that's a downside, like the widespread counterfeit use of Linux in ADSL routers but it's a double edged sword. If you want to leverage those who are likely to contribute to OSS, there's not much point in putting them off of making them spend most of the their time trying to get to grips with the code.

Was that tongue in cheek, or do you wish to leave the comments with their defects to confuse the enemy?

J-Dunn, 2015-10-15:
mc_homing_cycle
```
// Perform homing cycle to locate and set machine zero. Only '$H' executes this command.
// NOTE: There should be no motions in the buffer and Grbl must be in an idle state before
// executing the homing cycle. This prevents incorrect buffered plans after homing.
void mc_homing_cycle()
```
But mc_probe_cycle says:

```
  // Finish all queued commands and empty planner buffer before starting probe cycle.
  protocol_buffer_synchronize();
```

The description suggests they should be doing the same but $H does not synchronize. Over-sight?

chamnit, 2015-10-15:
@J-Dunn : As you have already found out, the code is already very tightly wound and fairly complex. I don't think any amount of comments will help decipher what its doing without an intimate knowledge of how everything works together. I would just leave things alone until the next project when there is more room to breathe and abstract things out.

J-Dunn, 2015-10-15:
Well my experience FWIW is that I've spent a lot of time picking through code line by line where clearer comments in stepper.c would have saved a lot of effort. What's there is pretty good but lacks a clear outline of the blocks/segments/steps division in the structure. I was often unclear which of the several buffers was being referred to and there seemed to be some lack of clarity between steps and segments at times. 

But if you want to leave it, I'll just make my notes locally

X3msnake, 2015-10-15:
I think clearer comments are a improvement. Anything wrong in what J-Dunn
is trying to do?

chamnit, 2015-10-16:
@X3msnake : Grbl's current plentiful and concise comments are rare in the open-source community already. There's nothing wrong with what @J-Dunn is doing, but on my end, it's not a good investment of time and energy. There is very little gained from making the comments perfect.

closed 2015-10-16 by chamnit

X3msnake, 2015-10-16:
Ok.

No dia sexta-feira, 16 de outubro de 2015, Sonny Jeon <
notifications@github.com