grbl/grbl #822: Startup blocks execute immediately?

AlexHolden, 2015-10-16:
It appears that when you change the contents of a startup block, the new code executes immediately. Is that the intended behaviour? It took me by surprise because I expected them to only execute at startup.

For example:
$G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
G20
ok
$G
[G0 G54 G17 G20 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
$N1=G21
ok
$G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok

chamnit, 2015-10-16:
@AlexHolden : Yes. When writing a startup block, Grbl will execute it. This is so that Grbl can quickly ensure that the startup block is a valid line of g-code. It's possible to refactor things to not do so, but you only set a startup block once and never during a job. So, it shouldn't be much an issue if it does execute it. At least, that was my thought when I implemented it.

closed 2015-10-16 by chamnit

AlexHolden, 2015-10-16:
OK, thanks.