grbl/grbl #823: variable spindle speed PWM o/p at pin D11

ecncshop, 2015-10-17:
Iam using GRBL 0.9I version with variable spindle PWM enabled.

My doubt is pin D11 : does it output analog 0V to 5V or actually outputs PWM digital pulse train with changing duty cycle?

cri-s, 2015-10-17:
Pwm train.

ecncshop, 2015-10-18:
If it is PWM pulse train: then what is the frequency? Is it user settable ? 
This page says we will get 0V to 5V in pin D11 
https://github.com/grbl/grbl/wiki/Connecting-Grbl

109JB, 2015-10-19:
Not sure if grbl uses a different method, but the standard Arduino pwm is 490 Hz.

The 0v to 5 v is for the pwm equivalent. The output is not true analog. The output is a 0v to 5v square wave and the equivalent analog voltage is dependent on the duty cycle of the output. The pins are 5 v, so 100% duty cycle is 5 v, 80% is 4v, 60% is 3v, etc.

EliteEng, 2015-10-19:
Variable Spindle PWM is "PWM" so yes 5v pulses at different duty cyles to give you an average voltage over time.

The Mega uses a 16bit timer and the Uno uses an 8 bit timer to generate the pulses.

The best source of information is the source code itself (spindle_control.c lines 85-96) which says it has a prescaler of 8 so it should have a frequency of 2MHz.

If you want to use it as a analog voltage you can add smotthing (capacitors,etc.)
if you want to use it as pulses you can change the frequency to suit your needs.

109JB, 2015-10-19:
@EliteEng 
I don't think 2 MHz is correct. The Arduino clock is 16,000,000 Hz, and the prescaler is 8 as opposed to the default Arduino prescaler of 64, but according to config.h, there are 256 intermediate levels  plus the 0% duty cycle for the speed output (0-256).  Also, the frequency would the duration of pin high plus the duration of pin low so you have to divide by 2. 

So, 
16,000,000 / 8 / 256 / 2 = 3906 Hz <<< GRBL's PWM frequency

with the default Arduino prescaler of 64, it is
16,000,000 / 64 /256 /2 = 488Hz <<< Standard arduino PWM frequency

EliteEng, 2015-10-20:
@109JB Yes It has a 2MHz pulse width not frequency
Those calculations dont look right.

fPWM = fCPU / (PRESCALER * (1+TOP))
fPWM = 16,000,000 / (8 * (1+255)) = 7812Hz for the UNO
fPWM = 16,000,000 / (8 * (1+65535)) = 30Hz for the Mega

**Note With the default Arduino you will find they use phase correct PWM ( not Fast PWM like GRBL ) so that is why you divide by 2 at the end.

closed 2016-03-04 by chamnit