grbl/grbl #824: Advises for adding punch custom command to grbl fork

frett27, 2015-10-17:
Hi, in a team project we use grbl for handling a CNC machine that will be used for punching music card (barrel organs/streets organ). as seen here : https://www.youtube.com/watch?v=aS7PNQ5tNPs .
we successful set up the motion part, but we'd like to add a new "M6" custom command to permit the punch, that involve a modal motion stop.

this will also be provided under the GPL3 license, 

i started to understand the source code and found that the parser can easily be extended, and the command added, but as the motion is handled by the interrupt, i wonder if there is a proper way to put the (¨PUNCH Actuator Down, sensor down check, PUNCH actuator up, sensor up check) cycle ?

how can i wait for an ended motion of the punch head before activating the punching cycle ?

thank's for the reply.

frett27, 2015-10-17:
ok, found : 	

protocol_buffer_synchronize(); // Sync and finish all remaining buffered motions before moving on.

from my understanding, i can add a punch_control file and synchronousely launch the command from the gcode execute section.

chamnit, 2015-10-17:
You're on the right track. The implementation you have outline is how I would have done it. 

Nice job on your machine. I've never seen your application before but looks quite well made!

tmpvar, 2015-10-18:
I second @chamnit's commending of your machine, it looks totally professional and to be quite honest, made me want to put down what I'm doing and start building something!

Nice work!

frett27, 2015-10-18:
Hi, sorry for the confusion, 
 this is not our machine but the kind of machine we aim to build, also, 
jean pierre who's working on the hardware part, has done the same quality building, 
here are the currently in progress machine : https://www.youtube.com/watch?v=3HnxPN0Fs90

i'll continue to advance on this topic in the M6 implementation, 

Patrice

109JB, 2015-10-19:
Sounds like you have a handle on things, but may I suggest using a different code than M6.

M6 is a code for tool change which doesn't sound like what you are after. M100-M199 are used for "custom" M-codes, so would be more applicable to what you are doing. 

Just trying to prevent confusion regarding an M6 command that does not fit RS274 standards.

frett27, 2015-10-23:
Hi 109JB, thank's for your valuable feedback, i'll do change the M6 code toward the M100,
i'm not an GCode expert, and still learning.

thank's very much.

frett27, 2015-10-25:
i've done the early modifications on my fork, https://github.com/frett27/grblPunch , left to manage the hold state, and probably some variable to invert inputs, 
i tried to not alter the core, but i had to make a bunch of #ifndef pragma to handle this.
maybe a "driver" architecture could handle the modifications we could have to make to grbl to permit a more readable view.
that's just a thought, and i'm not such confident, as the code is quite easily readable and simple as far as the understand of the architecture is done. I still have in mind the linus attitude over this.
very great job guys !

Patrice

frett27, 2015-10-25:
And took the M100 Gcode command !! :-)