grbl/grbl #826: Adding new buttons to GRBL interface

kucharzb, 2015-10-21:
Hello,

please redirect me if this is wrong place for such question:
is it possible to add some functions to the interface?

I would like to have a quick button for:
1)$H
2)G28 (going to predefined position)
3)probing down
4)inserting Z offset after measurement

I'm not a programmer, maybe somebody knows how to do it (if possible)?

Krzysztof

chamnit, 2015-10-21:
These would need to be covered by a GUI. Not Grbl. These commands are non-real-time commands and would require a lot of additional code to manage them. Also there is the obvious problem that there are no available physical pins for physical buttons either.

closed 2015-10-21 by chamnit

109JB, 2015-10-22:
I think you may be referring to my user interface. If so, please use the e-mail link in the help pulldown for requests like these. This forum is for GRBL, not the user interfaces for GRBL. I provided my direct e-mail for this express purpose.

Having said that, if you are referring to "109JB's GRBL Interface", there is a button for $H on my GUI labeled "Home".. For G28, and probing down, you can define your own buttons in the customizable macros. For setting the Z-offset, you click the Z axis readout on the DRO and it will bring up a window for setting it to any value you would like. All of this is in the pdf manual included in the download.

kucharzb, 2015-10-23:
Misunderstanding :) I confused GRBL and the controller which are separate things. I use GRBL controller by zapmaker (however it seems they stopped updates). I had a look on your interface and maybe we will like each other :) I will give it a try! Thank You!

LETARTARE, 2015-10-23:
@109JB
you can create an account on 'Github' and a deposit so you will have a wiki and discuss the possibility.