grbl/grbl #827: GRBL with drivers

Dave3891, 2015-10-22:
Would it be possible to change GRBL code around to use driver files? This would make it a lot easier to port to other processors. I think the parts that would need driver functions would be the timer interrupts/setting and the serial functions.

chamnit, 2015-10-22:
@Dave3891 : Yes. It's planned for the next Grbl project. The Arduino Uno's processor has limited room left and abstracting things to things like "drivers" would take up unnecessary space.

closed 2015-10-22 by chamnit

Dave3891, 2015-10-22:
Is this going to be part of 1.0? Or a much later version?

langwadt, 2015-10-22:
I think most of the things could be "abstracted" using no additional space in the current implementation. replace the current hw accesses with macros, define those macros to what the code does now. Other implementations can then replace those macros with what ever is needed for other implementations

chamnit, 2015-10-22:
@langwadt : If you're using a compatible MCU, like a Mega or 1280, yes. For ARM, this strategy wouldn't work as well. But FWIW, whenever I re-visit a piece of code, I try to isolate hardware level calls to make porting easier down the line. It's getting there, but not quite how I would consider easy, until you know what you are doing.

csdexter, 2015-10-23:
Also, the more generic you want to go, the uglier the macros :-) I know because I wrote such a HAL for grbl and I've seen how ugly it gets real quick. On the other hand, using functions to abstract things away works rather nicely -- but you can only afford functions on a platform with enough storage which the ATmega328P is not.

langwadt, 2015-10-23:
nothing prevents you from putting a function call in the macro if it is needed. 
and, at least in principle, inline functions should come out the same as a macro but depending on the compiler and optimizer level it doesn't always work that way