grbl/grbl #829: Parser state after alarm

109JB, 2015-10-27:
I notice that after an alarm, such as a hard limit trip, or a reset while in "Run" status results in all of the parser state values being set to what appears to be GRBL defaults.  For example, if the alarm happens when the state is:
G1 G56 G18 G20 G91 G94 M0 M5 M9 T0 F100.0 S0.0
Once unlocked, it is:
G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0.0 S0.0

The above occurs regardless of whether the user has any defined $N startup strings defined.  

Is there a reason that an alarm resets all the parser state info. I believe GRBL should retain the parser state values in effect before the Alarm and those should be applied after the unlock. Or, at a minimum, the parser state should reflect the $N values after unlock following an alarm.

chamnit, 2015-10-27:
@109JB : This is a known issue. Grbl will set the g-code standard defaults after an alarm, which is the second line you show, but if unlocked, it won't execute the startup lines. I've debated this before in some prior threads. The thought is if the user unlocks an alarm, it may not be the best idea to execute the startup lines. Most of the time it should be fine, but I've seen some users place G0, G1, G28/30 in their startup lines. If there is an unlock and machine position is not known, then there is a problem.

I suppose one solution is to make this a compile option to execute the startup lines after an alarm unlock.  But this adds one more option to a massively growing list of behaviors. Another solution would be to check if any motion commands are being written into startup lines and prevent them from being saved. 

The latter solution would be best, and should be much easier to do now with the recently refactored g-code parser. (Before it was hard to know/expensive to find out and check what commands are in a line.)

109JB, 2015-10-27:
@chamnit 
Is there a reason why GRBL couldn't just remember the parser state just prior to the alarm and keep those values, except for maybe M5 and M9?  Maybe I'm just not seeing the issue with doing this.

chamnit, 2015-10-27:
An alarm is a catch all state that something went wrong. As far as I understand, professional CNC controller always reset everything including the parser state. This is done to reduce potential human error when restarting a job(parser states is different)

It's possible to keep the gcode state I suppose but I'd rather have the system be as robust as possible, even if it enforce things that maybe slightly inconvinient.

109JB, 2015-10-27:
While true that commercial CNC's reset the parser state, I know of no commercial machine that doesn't allow the user to choose which of G20/G21 is the default. This can be set so that the machine boots or resets to the user defined units (inch/mm). In fact, on some commercial machines, if the default is G20 and the program includes a G21 command, an error is generated (Fanuc and Haas controls behave this way).

In the case of GRBL the default is always G21 and is not user selectable and sending a G20 command does not result in an error. This is not so much of a concern, but considering that the $13 units report mode retains the user setting, but the G20/G21 reverts to G21 after an alarm is concerning. This can result in confusion on the part of the user. when the machine will act in metric units, but the display showing inches.

At least this one setting (G20/G21) should be selectable by the user as a default behavior. Since discovering this, I have already implemented this in my GUI and a user selectable G20/G21 is sent right after any $X command, but I feel this is important enough that it should be addressed in GRBL.