grbl/grbl #830: Homing Y axis moves twice wrong dir and says OK

Shanjaq, 2015-10-29:
So I had homing working a little earlier, but today when I call $H the X axis will home correctly but the Y axis will make two short movements in the wrong direction and the console output will say "OK", as if it successfully nudged into the limit switch, but it totally didn't.

What would cause it to make two brief movements in the wrong direction then assume all is well?

I tested the limit switches by turning on hard limits $21=1 and manually nudging the platform into the limit switch, which triggered an alarm in the software (bCNC.)

Here's what I did to the config.h before compiling and uploading using the "load library" method:

#define HOMING_CYCLE_0 (1<<X_AXIS)                // REQUIRED: First move X
#define HOMING_CYCLE_1 (1<<Y_AXIS)  			  // OPTIONAL: Then move Y

X3msnake, 2015-10-29:
You probably have the homing switches wired the wrong way.

You can always change the mask of the direction in the settings.

Shanjaq, 2015-10-29:
Homing switches are wired correctly, quintuple-checked and probed for good measure.

direction mask is correct, as it was homing just fine the other day.

The odd and interesting behavior is that it Moves Twice in the wrong direction.  Like, it moves for a half second, pauses for a half second, then moves another half second, then says "OK".

maaadbob, 2015-10-29:
DIR wiring has come loose ?  This will result in your motor only driving in one direction.

> The odd and interesting behavior is that it Moves Twice in the wrong direction.

michaelwuehr, 2015-10-29:
I think what you describe is probably the backlash compensation of the end stops?
 
In this case this a usual behaviour of homing cycles.

euchcat, 2015-10-29:
I have thinked the same way than maadbob, it's seem a wiring temporary false connection, or temporary short circuit...

chamnit, 2015-10-29:
@Shanjaq : The behavior you are describing is commonly caused by electrical noise on the switch or wiring to the switch. The homing cycle is detecting the switch noise as a triggered switch and moving only the pull-off distance away from the switch. It also sounds like you have two homing locate cycles configured, which would explain the double motion. 

You can verify this by removing the Y switch and using a short wire you manually bring hi and low into the Arduino pin.

Shanjaq, 2015-10-30:
Ok I think I found the problem:  The Instructable I used said to use pin 12 as the laser tottle/enable pin.  Looking at the grbl pinout, that's the Z-axis limit pin!  Removing the laser module corrected the homing cycle.  Now to figure out which pin to use for the laser..

closed 2015-10-30 by Shanjaq

TadyTheFish, 2015-10-30:
I have it connected to spindle pwm... M3 turns it on and M5 off... And in the beginning of the program S8000

Shanjaq, 2015-10-30:
Ok, I got it working like I want by commenting out this line in config.h:

#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

Now pin 12 acts as Spindle Enable/Disable and I don't have to desolder anything.  Yay!  :)

TadyTheFish, 2015-10-30:
May i ask which program do you use for generating gcode? I use flatcam but the post processor isn't editable. So i have to edit the gcode (replace Z- with M3 and Z+ with M5)

Shanjaq, 2015-11-01:
This Python script may help with converting FlatCAM gcode output to control the laser with M3 and M5:

[code]
import sys, getopt

helpmsg = 'gcode_laser_convert.py -i <inputfile> -o <outputfile>'
z_negative = ('G01 Z-', 'G00 Z-')
z_positive = ('G01 Z', 'G00 Z')

def main(argv):
	inputfile = ''
	outputfile = ''

	try:
		opts, args = getopt.getopt(argv,"hi:o:",["ifile=","ofile="])
	except getopt.GetoptError:
		print helpmsg
		sys.exit(2)

	for opt, arg in opts:
		if opt == '-h':
			print helpmsg
			sys.exit()
		elif opt in ("-i", "--ifile"):
			inputfile = arg
		elif opt in ("-o", "--ofile"):
			outputfile = arg

	with open(outputfile,'w') as o:
		with open(inputfile) as f:
			content = f.readlines()
			for r in content:
				if any(s in r for s in z_negative):
					o.write('M3\n')
				elif any(s in r for s in z_positive):
					o.write('M5\n')
				else:
					o.write(r)

	print "DONE"

if __name__ == "__main__":
	main(sys.argv[1:])
[/code]