grbl/grbl #831: asymmetric stalling

J-Dunn, 2015-10-30:
hi, 

in testing to find the max step rate that my system will achieve using GRBL, I find that it has a higher tendancy to stall in the formward direction ( increasing axis coordinates) . If it reaches max displacement it will return without error. 

I an curious as to whether this could be some asymmetric variation in the pulse train proudced by GRBL or possibly a magnetic asymmetry in the motor design.  Is there any reason why there could be small differences in the timings produced by GRBL dependent on direction of motion that could trip an error condition preferentially in one direction in near critical conditions.? 

I an testing on a bare motor with a just a small inertial load in the form of a cylindrical coupling. 
I am varying $101 with a run lenght sufficient to have a ' trapezoidal'  speed profile.

When at critical step rates I can determine by ear that it is failing after running a fair section of the max speed section. I have not determined whether or not it is just at the point of hitting decel when it fails. The motor stalls then picks up again when well down the decel curve when it gets back into sync with the pulses. 

If I push it hard enough it will fail in both directions but it will always fail preferentially in the forward directin.

PicEngraver, 2015-10-30:
Does it repeat with different motors?

chamnit, 2015-11-01:
@J-Dunn : There is no difference between directions during step pulse generation. The only **minor** effect would be position to step conversion, but this effect would be extremely minimal. It's most likely stepper motor related. Shaft mechanisms often don't have the exact same friction in both directions. Either from uneven wear or debris.

closed 2015-11-01 by chamnit

J-Dunn, 2015-11-03:
Thanks for the reply Sonny. It would have taken a long time to get such a definative answer by parsing the code. 

I have several, nominally identical, motors so I will do some experimentation swapping coil directions and changing motors. There are probably small inaccuracies in the magnetic structure that could cause preferencial failure in one direction.

You have previously commented that there are small adjustments made at the virtices of the 'trapezoidal' speed curve to smoothe the jump from max accel/decel to contant speed or stop conditions but I have not been able to locate this. 

Can you point we to where to look. Thanks.