grbl/grbl #832: Homming Issue

JoseGiusGius, 2015-10-31:
Hi, My name is Jose.

I have a problem with the homming cycle and I hope that you can help me to fix it.

I use Windows 8.1, GRBL version 0.9j and chiliPeppr as g-code sender.

I have 3 optic switches and I run a homming cycle, the Z axis do it perfect, but X and Y axis start to move just like 5mm and stops.  And print a message "ALARM: Homing fail", if I vary the $27 "pull off" parameter, the x and y axis moves further but it stops after "X" mm (X is the value on $27)

I use capacitors 10uF to stabilise the signal and the parameters like this:

$0=5
(step pulse, usec)
$1=1
(step idle delay, msec)
$2=7
(step port invert mask:00000111)
$3=0
(dir port invert mask:00000000)
$4=1
(step enable invert, bool)
$5=0
(limit pins invert, bool)
$6=0
(probe pin invert, bool)
$10=2
(status report mask:00000010)
$11=0.01
(junction deviation, mm)
$12=0
(arc tolerance, mm)
$13=0
(report inches, bool)
$20=0
(soft limits, bool)
$21=1
(hard limits, bool)
$22=1
(homing cycle, bool)
$23=0
(homing dir invert mask:00000000)
$24=200
(homing feed, mm/min)
$25=500
(homing seek, mm/min)
$26=1
(homing debounce, msec)
$27=5
(homing pull-off, mm)
$100=410
(x, step/mm)
$101=410
(y, step/mm)
$102=2048
(z, step/mm)
$110=1000
(x max rate, mm/min)
$111=1000
(y max rate, mm/min)
$112=1000
(z max rate, mm/min)
$120=500
(x accel, mm/sec^2)
$121=500
(y accel, mm/sec^2)
$122=500
(z accel, mm/sec^2)
$130=200
(x max travel, mm)
$131=200
(y max travel, mm)
$132=200
(z max travel, mm)

if you can help me it would be great, thanks you.

TadyTheFish, 2015-10-31:
It looks like grbl thinks that the machine is already on the limit switch. Try inverting the limit pins. $5=1

TadyTheFish, 2015-10-31:
Optic switches dont need capacitors. They dont have debounce. If you didn't put a resistor before the capacitor you cam damage the switches. I advise you to remove those caps! The caps can pull a lot of current when chraging or discharging. 
Debounce capacitors are used with mechanical switches.

JoseGiusGius, 2015-10-31:
I have 4.7K ohms resistor before the cap, I use it because the digital input is so sensitive that the noise from the motors trigger them. 

I inverted the pin as you tell me, and now the home at Z axis doesn't work.

I would continue testing, if you have another idea it would be great, Thanks you.

TadyTheFish, 2015-10-31:
What do you mean the z axys doesnt work? Alarm?

JoseGiusGius, 2015-10-31:
When I move the machine out of the limit switches they work perfectly, the problem that I have is Just at homing cycle.

JoseGiusGius, 2015-10-31:
Yeah. Alarm on Z axis,

TadyTheFish, 2015-10-31:
Can you draw me a schematic how you connected the switches and the type of switches

TadyTheFish, 2015-10-31:
What is you $5 setting?

JoseGiusGius, 2015-10-31:
If you want I will send you a video or some photos to evaluate the behavior.

TadyTheFish, 2015-10-31:
Send it to tady.the.fish@gmail.com . please post me your settings

eduardsik, 2016-02-17:
Same here, proper connected switches on all axes, xy works great, but z gives hard limit error just on multiple pushes. (not sensig) All microswitches. I reversed x and z, and turns out, z switch working great. no noise on wires. It seems some software error on Z home switch.

JoseGiusGius, 2016-02-18:
It wasn't a software problem. I already fixed it, it was because I had 2 NC switches and 1 NO switch...

JoseGiusGius, 2016-02-18:
Tady the fish was the one who helps me to solve it. Thanks tady :)

eduardsik, 2016-02-18:
Jose, now working all limits in 0.9j and can use hard limits?

JoseGiusGius, 2016-02-18:
Yep ! All working, also I have used GRBL on arduino One and Arduino Nano