grbl/grbl #833: homing/limit & status  issues

valpoder, 2015-10-31:
i am using grbl 0.9j , fresh build

question 1:
when i do the current status command "?" i get nothing ...
is possible to see limit status with this command???

question 2:
i have 2 N.O  switches , and 1 N.C switch ...     
i know i can invert the behavior using $5 but :
 is there a way to make a diferent conf for each pin ???

question 3:
when performing $H command:
shuld i place each axis near the limit switch ???
i dont understand the meaning of the pull-off parameter...

i tried values such as:
 $27=100.000 (homing pull-off, mm)
but i get odd behavior...

note 4:
i suggest adding more info on the capacitors for the limit switches , it took me long hours before i figured that out...

Thanks for GRBL !!!  its GREAT !::::!!!!

TadyTheFish, 2015-10-31:
Answer1: how do you send the command? If it is with the arduino IDE program you have to set the "new line". When you send something to GRBL it expects to see car

TadyTheFish, 2015-10-31:
Cariage return. This is mandatory!
Answer 2. Here lies you problem. And no you cant configure a single homing pin in reverse. If you can't replace the NC switch with a NO you can wire the NC switch from the pin to +5v but you must wire a 1k or 500 ohm resistor from the pin to the ground.
Answer 3: it doesn't matter where the axus is it should move to the switch by itself.
The $27 parameter tells the grbl how many mm or inches it moves avay from the switch. 
When you press homing button GRBL starts to move to the switches. When it detects a switch it stops goes back, hits it again with slower speed and then it "pulls of" or move away from it with the mm or inches that you specified in parameter $27. This way the switches are not active when homing is complete. 
Then grbl uses those sam switches (if enabled) for limits
Comment on your note4: the capacitors differ from each machine. No one can tell what kind of switches you are using. If you google "switch debounce" you can get alot of info how to chose the right capacitors. If i have problems with debounce i use a 1k resistor in series with the input pin and 1uF cap or 10uF in parallel with the input and ground

Hope tgis helps:)

valpoder, 2015-10-31:
question 1:
im using grbl-controller-3-0 in windows 7 64bits to send the commands..
i think the Cariage return is already there ... but im not sure ... most commands such $$ $X , seem to work though...  

i will try a diferent program , and sending ASAP

question 2:  
perfect , i will try a pullup or rewire my switches..

question 3: perfectly clear now .. pull-off = move away from the limit switch !!!

all very good , thanks so much for the support !!!!

TadyTheFish, 2015-10-31:
I just remembered. You don't see the returned string. The program uses the data to update the current position. If you open the arduino IDE and type ? You will see the response...
I use grebrlControl and i cant see the returned data... But values of Wok X and Y and values of Machine X and Y change;) dont worry about the response from "?" the program doesn't show it ;)
As for the switches. If you are using NO contacts you need a PULL-DOWN resistor:)
All three inputs must be HIGH when none of the switches are pressed. And they gow LOW when you press them ($5=0)

TadyTheFish, 2015-10-31:
Sorry i nade a mistake.... If you are using switches with NO contacts... You wire them like this: http://dangerousprototypes.com/docs/images/0/06/Pullup-pin.png

And if you are using NC switches you wire them like this:http://www.horrorseek.com/home/halloween/wolfstone/Controllers/cioinp_PullDownSwitchClosed.bmp

109JB, 2015-11-03:
A few points to clarify.

First, the realtime commands sent to grbl do not need any end of line character. They are picked from the stream and acted on. So a carriage return isn't needed for these. An end of line character is needed for other non-realtime commands though.

Second, Pull-up resistors are not needed if using NO switches. The arduino has built in Pull=up resistors that are implemented in GRBL. If using NC switches you will need pull-down resistors though.

You cannot use a mix of NO and NC switches without modifying the GRBL code. I haven't looked so can't even say if it is feasible at all. 

When using $H, the axis can be started anywhere which is what the homing sequence is for in the first place.

You can enable limit pin reporting in the status reports by setting $10 as described in the wiki. However, this may not be supported in the GUI you are using..

I would recommend getting this to work using a simple serial monitor such as the one in the Arduino IDE and typing in commands.

valpoder, 2015-11-03:
home seem to work correctly now..

I found a small issue thought:
-my Z axis has a spring that pulls the spindle back up....
-this activates the limit switch when machine is idle..

if the pull-off value is too small , the switch does not change and the homing process is aborted
if the pull-off value is large enough , the switch changes state.. and homming process continues to other axis...

it seems homing algorithm is not suited for my setup...

TadyTheFish, 2015-11-03:
Can you work with a big pulloff?

langwadt, 2015-11-04:
can't you just move the spindle down a few mm before doing homing?

109JB, 2015-11-04:
@valpoder 
From what you are describing, the spring is pulling the Z-axis up past the point where the switch changes state from open to closed. There will be some distance from where the switch state changes to the where the Z-axis is pulled up by the springs. That distance has to be less than the pull off distance. This will affect the other axes as well. If you can adjust the Z axis switch to reduce the distance from switch state change to where the springs pull the Z to then you should be able to reduce the pull off distance. You could also further limit the z-axis up travel to reduce this distance.