grbl/grbl #835: Changing frequency for pin 11 does not seem to work

mladegaa, 2015-11-01:
Hello everyone,

I have been trying to get a PWM controller I have to run with grbl 0.9j and uploaded with Arduino 1.6.5.

I have been having issues with my PWM controller not reading my PWM signal, and I was told that it will only read below 1khz and preferably in the 100hz range.

So I changed line 94 in spindle_control.c to this
TCCRB_REGISTER = (TCCRB_REGISTER & 0b11111000) | 0x06; // set to 1/256 Prescaler
in order to get 122hz.

After I compile and upload I hooked my oscilloscope up and it still reads 7.8khz which I understand is the standard grbl pwm output.

I haven't been able to figure out how to fix this, and was wondering if anyone could shed some light on the issue.

chamnit, 2015-11-01:
@mladegaa : Make sure you are editing the Grbl source files in the Arduino library directory.

EliteEng, 2015-11-02:
@mladegaa the code is correct, that will set the prescaler to 256. You should get 244Hz

As @chamnit said make sure you are editing the files in the Arduino Library folder.
If you are editing the correct files then it is somewhere in the compiling or flashing.

mladegaa, 2015-11-02:
@chamnit @EliteEng Thanks for the advice, I wiped my Arduino folders and reinstalled and added grbl again, applied my changes and it compiles with the changes now. Maybe my Arduino library was pointing somewhere else and my changes was applied to the wrong place. Had an old install that got patched recently.

@EliteEng just verified that i'm getting 244hz, maybe I read it wrong, thought that phase correct pwm would give 122hz

Thanks again!

EliteEng, 2015-11-02:
@mladegaa GRBL is set up as Fast PWM not phase correct that is why you are getting 244Hz

closed 2016-03-04 by chamnit