grbl/grbl #836: Having an issue with G3 on grbl 0.9J

slapers, 2015-11-02:
Hi,

Its my first time trying to mill something, so please excuse me if i'm making some stupid mistake...
I'm using fusion 360 and had some weird behaviour when it was doing G3.
So i extracted the gcode that behaves weird and surrounded it with the G1 X0 Y0 Z0 to also show that grbl seems to have lost it positioning ? This is what happens : https://www.youtube.com/watch?v=5qLB040t4w0

```
G90 G94
G17
G21
S25000 M3
F400
G1 X0 Y0 Z0
G1 X14.537 Y17.048
G3 X35.463 Y32.952 I10.463 J7.952
G3 X14.537 Y17.048 I-10.463 J-7.952
G1 X0 Y0 Z0
```

maaadbob, 2015-11-02:
Interesting.
CamBam shows a circle being cut.  (see attached pic)

chamnit, 2015-11-02:
@slapers : There shouldn't be a problem with Grbl with this g-code (I also tested it to make sure). This is likely caused by your steppers losing steps (Grbl should show the correct location, even though your machine isn't at it). This can be caused by incorrect settings, particularly the junction deviation, max speed, and acceleration, or inadequate shielding or power to the stepper drivers. If Grbl is not reporting the correct location (in position after the job), its historically been something to do with the g-code sender.

slapers, 2015-11-02:
@chamnit : i created a test file with some linear movements and upped the speed. And indeed it seems like its loosing steps... I had tested the machine before building a housing for the electronics..so something must have gone wrong (i recabled everything in the process...). Thanks for pointing me in the good direction !!

slapers, 2015-11-04:
Just wanted to give an update, found out that i had bad ground connection between the arduino and  stepper drivers. Must have happened when i mounted everything into the case ..
Thanks again for pointing me in the right direction !!

Feel free to mark case closed :-)

chamnit, 2015-11-04:
@slapers : Great! Thanks for letting us know.

closed 2015-11-04 by chamnit