grbl/grbl #837: Baud rate issue?

109JB, 2015-11-05:
While doing some torture testing of my GUI I came across a problem that I am not sure of 100%. I am using an Arduino Nano clone  that uses the ch340g chip with the baud rate set to 115200. The pertinent GRBL settings were 250 steps/mm, 7200 mm/min max rate to achieve 30 kHz step rate. Then what I did was take one of my 130+ thousand line test files and modified it to only have G0 rapid moves. This file has lots of little short moves so there is fast communication required between the GUI and GRBL. Occasionally an error would be returned from GRBL, but I know the file is good having been run many times with the only change using a text editor to do a replace of G1 with G0. While investigating the error I turned on echo from GRBL to investigate

An example of what was happening occasionally is instead of getting the following 2 lines:

X3.8212 Z1.0742 
X3.8173 Z1.0765 

GRBL would echo something like:

X3.8212 Z1.X3.8173 Z1.0765 

which is a truncated first line with the second line tacked on the end. The errors were not happening at the same spot in the file but the errors were always similar in fashion with a truncated line having the next line tacked at the end.

After some searching on the internet I found some information regarding the Atmega328P  that indicates that at 115200 there is a higher chance of errors than at 250000 when at a 16MHz clock.

[image]

So, I switched my GRBL install to 250000 baud rate and no more errors. I ran the same code file about 10 times with no errors. Then for an additional test, I switched back to 115200, and ran the same code file and experienced an error on the first run.

Having said all of this I must say that my testing means are limited and the test file I am using is not a realistic file that anyone would run in a real life situation. Also,with G1 commands in it there were no problems due to slower communications I suppose, and the errors never showed up at less than about 60,000 lines into the test program. Most people won't get to this issue anyway, but it does show a potential problem.

What this brings up is whether 115200 should be the default baud rate, or if 250000 should be due to the lower error rate indicated in the above table.

TadyTheFish, 2015-11-05:
I am also running grvl on a nano clone and i have no problems. Could you send me your g-code file so i can test it? What GUI arw you using?

blinkenlight, 2015-11-05:
I find it hard to adequately stress how unlikely I find that your problem would be baud-rate related. The very same datasheet explains nicely that majority-voted multiple sampling allows problem-free signal reconstruction up to a maximum theoretical error of about +/- 4.0%, which is two full percent more and/or twice the 2.1% resulting for 115200 baud. The connection itself is board-local and any other source of rate error could only originate from the xtals driving the ATMega and the ch340g, but as the datasheet itself notes xtals simply just don't have the sort of imprecision that could interfere with baud rates. That said, obviously less error is always better (except when it means twice as many interrupts per second for an already overworked MCU), but I just can't see it matter in this instance.

kfoltman, 2015-11-05:
Don't forget we deal with imperfect physical devices, often manufacturing rejects that end up on aliexpress and the likes - so you can't rule out a dodgy CH340, out-of-spec xtal or something similar. Manufacturers may be lying in the specs, too, or the datasheet may be badly translated. I've heard about people having problems with 115200 but not 250000 before, in context of 3d printers.

chamnit, 2015-11-05:
I've particularly seen here timing issues being resolved when switching from 115200 baud to 250000 baud with Arduino clones. @kfoltman may be correct that manufacturing specs may not be as exact with clones compared to official boards. But, I would like to add that this issue is still fairly rare.

FWIW, there is no problem using 250000 baud. In fact, I would probably recommend using that over 115200. The only reason 115200 baud is standard on Grbl is that it is a universally supported baud rate for all terminal programs.

109JB, 2015-11-05:
@TadyTheFish  - I am running my own GUI that I am developing. I can put the file on my dropbox for you, but as I said it is an unrealistic code that serves no purpose other than testing.

@blinkenlight  - You may find it hard to believe, but all I have to do to reproduce the error is set 115200, and all I have  to do to make it go away is set 250000. That's all. Nothing else. So explain to me how the reproducability of the error at 115200 but not 250000 is not evidence of these errors being baud rate related?  If it isn't baud rate related, give me something else to look for because I discovered this trying to make everything as robust as possible from my GUI to GRBL. If I have a problem in my GUI I want to find it, but from a deductive point of view with what I have tested, everything is pointing to it being tied to baud rate.

@kfoltman - That is a good point. I have another Nano and I may flash it and match the eeprom settings and test that one to see if it is a fluke with this one Nano board or if it can be reproduced on another one.

@chamnit  - Ahhh. I see where 250000 is problematic due to terminal support. I realize now that even the Arduino IDE only goes to 115200.

I don't want anyone to think it is a GRBL problem per se, because I don't think it has anything to do with GRBL's coding.  I believe it has more to do with the hardware,  If you do a search over on the Arduino forums you will find that there are numerous posts regarding problems at 115200 on the ATmega 328 boards. If you read through enough of them you find a similar theme. That the errors go away when you set something other than 115200. Granted there are posts about serial errors at other baud rates too, but from what I read and inferred, 115200 is particularly troublesome on the ATmega 328. I am not an expert on this by any means. Just reporting what I have found. 

As I said, this is an error that is going to be exceedingly rare. Heck, I just now found it and have been working with GRBL and programming my GUI for hundreds and hundreds of hours without ever seeing this issue. It only appeared when I decided to push everything to absolute limits (full rapids, short moves, 30kHz step rate, etc.). For a "normal" user it would probably never show up. Having said that, unless I find a reason not to, I will be using 250000 from here on out.

109JB, 2015-11-05:
@TadyTheFish My test file is at:
https://www.dropbox.com/s/sljlq0q486022uz/139K%20file%20all%20rapids.nc?dl=0

Also, after realizing that I didn't have a basic serial terminal program that went above 115200 baud, I made one and uploaded it to my dropbox. If anyone would like it it is at the following link. I has some extra buttons for grbl specific commands added.
https://www.dropbox.com/s/gs2axrkbalc98fq/SerailMonitor_1_0_0_1.zip?dl=0

blinkenlight, 2015-11-05:
@109JB - Evidence...? Really now. It absolutely isn't. When you change the baud rate you change everything about how the timing of serial activity interacts with everything else - code inside GRBL, hardware in the ch340 chip, timings and/or message sizes between the USB host and the ch340, servicing of the host USB stack and drivers - you change **everything**. That cannot be called "evidence" of **baud rate error** in any meaningful sense. To humor you, a more convincing proof would be disconnecting the Arduino's xtal, having it switched over to external clock and feeding it 16MHz then 16MHz+/-2.1% and see the problem come and go. It would still not *exclusively* change the one thing under test therefore it would still be a flawed proof, but it would have tremendously more credibility.

Anyway, I never said it was impossible, only that it was quite unlikely - the theory simply does not offer a satisfactory explanation for the fault occurring, and a bug is not pinned down until the mechanism of its occurrence is adequately explained by established theory. If 115200 is indeed a widely-known problematic rate then of course it's rather likely that something dodgy is happening there, but all that tells us is some other baud rate would likely be preferable unless too inconvenient - let's not forget, most people don't seem to have much of a problem with it after all.

chamnit, 2015-11-05:
@blinkenlight : I think this is an example of the simplest solution is often the right one. The fact that changing the baud rate from 115200 to something else will successfully solve problems like this. Given that it's not a problem isolated with Grbl, but with other firmwares, I would say that the timing error associated with 115200 baud is the likely culprit. My guess is that's its external clock related, as you mentioned, where the clocks for clones may not be as precise as official boards.

Either way, serial RX/TX comm is handled on a hardware level on the 328p. As long as the serial ISR is serviced before the next RX byte comes in, Grbl shouldn't lose any data.

109JB, 2015-11-05:
@blinkenlight  - You words seem very harsh**,with all of the bold screaming text.** First of all, IT IS evidence that there is a problem using 115200 versus 250000 even if the exact mechanism of the error is not known. Could it be in GRBL code? Or the ch340g chip? Or something else? Sure, but the FACT that it happens on 115200 and not on 250000 is absolutely evidence that the baud rate setting is suspect. I don't see how in the world you can argue otherwise. Is it a "baud rate error"? I don't know, but it is an error link to the baud rate setting which is my point. Everything else is semantics. Whatever the reason, switching the baud rate fixes it.  I am only trying to bring to light a possible error that can occur however unlikely that may be.

blinkenlight, 2015-11-05:
@chamnit - sure, as long as the doubled amount of ISR being executed (and that right now there doesn't seem to be any senders with 250000 support) is not a problem, why not. As I said - if it's indeed a widely known issue, there's clearly something funny about 115200 baud on the Arduino, no argument there.

@109JB - whatever tone you read into a piece of text is your own interpretation. Either way, last time I checked bold or italics was accenting a point, screaming LOOKS LIKE THIS (merely as a demonstration, of course). Kinda like what you do up there above. At any rate, nobody argued the baud rate itself couldn't possibly be the problem, I merely pointed out I have issues with it being attributed to the precision of the baud rate. I'm happy to keep playing 'he said / she said' if you insist, but I do believe the point is long moot by now.

109JB, 2015-11-05:
You are simply arguing semantics of whether the error is from baud rate precision or something else. I personally don't care if it is actually from baud rate precision or something else related to the baud rate selection. I only care how to fix it. Here are the facts:

1. A error was reported by GRBL while streaming at 115200 baud - fact
2. The error is reproducible at 115200 baud - fact
3. The error goes away at 250000 baud and I haven't been able to reproduce it at this setting - fact
3. Research from other forums shows that 115200 baud may be  problematic on 328P arduinos - fact
4. The theoretical baud rate error at 115200 is 2.1% - fact from datasheet
5. The theoretical baud rate error at 250000 is 0.0% - fact from datasheet
6. The 328P datasheet lists the recommended maximum receiver baud rate error as + or -2.0% - new fact from datasheet
7. 2.1% is greater than 2.0%
8. 0.0% is less than 2.0%

Taking points 6, 7 and 8 into consideration, 115200 baud exceeds the maximum recommended baud rate error as specified by Atmel, whereas 250000 does not. The datasheet describes the 2.0% value as "the maximum receiver baud rate error that can be tolerated." Those are Atmel's words, not mine. On top of that, the table listing for 8 data bits is only shown on the U2Xn = 0, normal speed mode. If the high speed mode is not available at 8 data bits then the baud rate error at 115200 and 8 data bits is not 2.1%, but rather -3.5%. This would put it far outside of Atmel's recommended maximum error.

The baud rate tolerance that can be accepted is dependent on both the transmitter and receiver baud rate precision, so the 4 percent-ish error threshold  that you referenced is the total error accounting for both, not just on the Arduino end.

The bottom line is that there is something going on, however rare, at 115200 baud. It could be from baud rate precision, hardware, GRBL coding, fudge ice cream, or whatever. The bottom line is that picking 115200 resulted in errors and picking 250000 did not.

109JB, 2015-11-05:
@chamnit - From your comment:
"Either way, serial RX/TX comm is handled on a hardware level on the 328p. As long as the serial ISR is serviced before the next RX byte comes in, Grbl shouldn't lose any data."

From this I infer that the error I experienced is a result of a "perfect storm" and would likely only rear its ugly head if a similar set of circumstances presented. I guess what I mean is that this would only be an extremely rare occurrence and not too much to worry about in real life machining applications, and that if wanting to err on the safe side just use 2500000 baud rate. Is this your take?

Thanks

chamnit, 2015-11-05:
@109JB : If it is indeed an external clock problem, I would say it's a fairly rare occurrence between manufacturers and productions runs at 115200 baud. And less so for well made ones. This is generally from observations over the years I've seen this issue crop up. It's almost always on a cheap clone board and changing the baud rate fixes it. But, this is all conjecture at this point as to what the root cause actually is. 

@blinkenlight did bring up a good point that the serial RX ISR has to be serviced twice as fast at 250000 baud. At this rate, you may run into some performance problems at the upper end of step frequency. I've measured the 328p's interrupt overhead once. It's not great so it'll eat into the available cycles quickly.

109JB, 2015-11-06:
So I did some more testing and decided to try another baud rate from the datasheet table that had low values for baud rate error. 76800 baud shows 0.2% error in both low and high speed mode, so I set the baud rate in GRBL and my GUI to this and ran the file that generated the error 3 times with no errors. It still errors out at 115200 so nothing changed there. Run times are  about 10% slower compared to 250000 baud, but again this is an unrealistic G-code file. So far I can't see a drawback to going to either 250000 or 76800 baud except for the case of maybe a router cutting soft material.  Oh, and all of this testing is with the GRBL setting such that G0 rapid moves should get to 30kHz. The 250000 doesn't seem to pose a problem for GRBL with these settings.

@chamnit : Do you see any reason 76800 should be avoided

blinkenlight, 2015-11-06:
@109JB : well if you must insist - the Arduino is not actually communicating with another Arduino (in which case the argument for staying under half the admissible 4.0% rate error would have some merit) but with a ch340g, with a declared maximum Tx error rate of less than 0.3%. So nice try, but no cigar (fact). Unfortunately the return maximum acceptable error is stated much more ambiguously, but either way even in case of actual incompatibility that shouldn't generate an error on its own as long as the data streamed down to the Arduino is intact. One other way to fix that specific board (if the rate error is indeed the issue) and stay on the standard baud rate would probably be to change the xtal to a 20Mhz and recompile GRBL for that frequency - it depends on what you feel more comfortable changing and what you'd prefer to keep. Anyway, if it works better at 76800 for you, great. One less problem I guess.

109JB, 2015-11-06:
@blinkenlight - I have tested at baud rates between 76800 and 250000 and the problem shows up at the baud rates listed in the datasheet table that exceed the Atmel  maximum baud rate error specification, and does not show up at baud rates with error percentages below the Atmel specification. So if it isn't a  timing issue somewhere, then you tell me what it is?

**For everyone else** I decided to do more testing as I have several Nano clones and a couple Mega 2560 boards. The problem encountered shows up when using any of the Nano clones I have at 115200 baud, but not at 250000 or 76800. These clones are not all from the same supplier and superficially some have a slightly different shade to the board color, and different screen printing of text. so, definitely from different batches. The problem does not show up when running on the Mega boards at 115200, 76800, or 250000 baud. The Mega boards don't use the ch340g chip for communication. I would be curious if the problem exists on the UNO clones as they also use the ch340g. Unfortunately I don't have an UNO clone to test.

AlexHolden, 2015-11-10:
I've just added support for 250000 baud to my GUI. This turned out to be slightly painful because it's a non-standard rate that isn't supported by the Posix termios API. Both Linux and OSX have their own weird poorly-documented ioctl() that lets you go behind termios's back and request an arbitrary non-standard rate (on Linux this involves directly including a kernel header file because it isn't supported by Glibc).

It depends on the hardware and the kernel driver whether any particular rate is actually possible. What I've discovered is that using an Uno board with the ATmega16U2 USB interface chip, 250000 baud works on both Linux and OSX, but on a Nano board with the CH340g chip it only works on Linux. The OSX CH340g driver returns an error when you request 250000 baud.

I don't have a board with an FTDI chip so I don't know if those would work. I did read somewhere that the Raspberry Pi's built-in serial port is capable of 250000 baud, but only if you first alter one of the boot parameters to change the baud rate generator clock rate.

109JB, 2015-11-14:
I have been doing some more testing. I decided to get a serial port sniffer program to see exactly what is being sent, and what is being received to verify if any of this is from my GUI. What I have determined is that the data sent to GRBL is correct, but the echoed return is not correct, meaning the error happens somewhere either in the arduino, or in the GRBL code.

As further testing, I made a small program to test with. This program has streaming code that is very streamlined with nothing extra except what is absolutely necessary for streaming and for detecting GRBL error messages. Nothing else. I am testing with a CAM generated g-code modified to have only G0 commands. This results in very fast streaming. If I slow the stream down a little by placing an intentional delay in the streaming code, it works as expected, but at full speed GRBL doesn't always echo back the same line that was sent and throws errors. I have verified that the lines sent are correct by using a serial port sniffer program. For example, in a recent run using the character counting streaming, the following lines were sent  at the very start of the run:

G01G21G90X-0.5375Y3.5375Z3.3598F4000
X3.8999Z1.0234
X3.896Z1.0262
X3.8645Z1.0475
X3.8566Z1.0527
X3.8212Z1.0742
X3.8173Z1.0765

These first 7 lines were immediately sent bu the GUI and the character count, including a LF at the end of each line, totals 127 characters and was verified using the serial port sniffer. The response from GRBL was:
 

 X2.8811Z1.167             
 X2.889Z1.1702             
 X2.9362Z1.1889          
 X2.9441Z1.192             

with these responses from GRBL

 [echo: X2.8811Z1.167]
 ok
 [echo: X2.889Z1.1702]
 ok
 [echo:]
 ok
 [echo: ]
 ok
 [echo: .9362Z1.1889]
 error: Expected command letter
 [echo: X2.9441Z1.192]
 ok

As i said, the serial port sniffer confirmed that the line that generated the error is sent correctly with a LF used as an end of line character. So, this indicates to me that  the problem lies either in the USB/Serial chip, the communication between the chip and the MCU, or within the GRBL code.

I tend to think it is not the GRBL code because it seems to be tied to baud rate with far fewer of these errors with higher baud rates. I would think that higher baud rates would yield more errors if the problem was within the GRBL code, but I guess it is possible.

coco243, 2016-09-03:
I don't think that the UBRR formula is correct in the serial.c

`uint16_t UBRR0_value = ((F_CPU / (8L * BAUD_RATE)) - 1)/2 ;`

it shouldn't be?

`uint16_t UBRR0_value = ((F_CPU / (16L * BAUD_RATE)) - 1) ;`