grbl/grbl #838: program hangs on particular slope angle

Shanjaq, 2015-11-06:
I've been trying to cut a particular design for days, every time it reaches a particular slope the program hangs for about a second, then appears to drop several move instructions and continues along an incorrect path.  Here is a screenshot of what I'm trying to cut:
[image]

and here is what I'm seeing (multiple tests on same sheet but rotated, problem-spot highlighted.)
[image]

Aside from the general crapulence of optical drive steppers, these "hangs" seem to be a software issue as they're 100% reproducible, even with a regenerated gcode file using a tweaked tool diameter.

Here is the gcode made with FlatCAM 8.4:
https://dl.dropboxusercontent.com/u/13046780/test1_Bottom_laser.txt

I'm using an RBBB Arduino with ATMega328p and 2x Easy Driver 4.4 boards for the CDROM steppers:
http://www.schmalzhaus.com/EasyDriver/

chamnit, 2015-11-06:
This looks like a lost steps problem still. Make sure your junction deviation setting is about 0.01mm or even less if your system is particularly sensitive.

euchcat, 2015-11-06:
Hello
It seems that he lacks the coordinates for the Z axis in the file you
mention.
Maybe....
Good evening everyone.

Shanjaq, 2015-11-07:
I'm using the default arc tolerance of 0.002mm, and the Z movements have been replaced with the M03 and M05 commands to turn the laser on and off.  The way execution just hangs in that particular spot for exactly 1 second every time really does not look or feel like a missed step.  Could it be that the slope is edging too close to the minimal X value and the software "predicts" a collision because of the tool diameter and skips all instructions that would take it too close?

TadyTheFish, 2015-11-07:
I have a laser cnc. With M3 and M5 commands and it works perfect. As for the tool diameter i always set it ti minimum as possible (0.05mm).

chamnit, 2015-11-07:
@Shanjaq : Arc tolerance isn't junction deviation. These are two separate parameters. The latter is an acceleration setting. Also, you have no G2 or G3 arc commands in your g-code program. These looks to have been translated to G1 commands by your CAM program. 

Grbl does not skip any instructions and always execute valid g-code. Given that there is nothing out of the ordinary jn your g-code program. That leaves only your steppers are losing steps. Please post your Grbl settings to see if there is anything wrong there. Otherwise, it's still something in your machine (backlash, friction) or your stepper drivers.

Shanjaq, 2015-11-08:
Ok, I've set Junction Deviation to 0.001, it still "pauses" for exactly 1 second at that particular spot.  I'll have to conclude that CDROM steppers are just plain crap and will never work.  Thanks for all the help anyway.

Shanjaq, 2015-11-08:
Is there a way to upload v6.0b using the "library method"?  I'd like to try the same version used by the instructable I followed:
http://www.instructables.com/id/Pocket-laser-engraver/

Shanjaq, 2015-11-08:
Alright, with a bit of hacking I got v0.6b onto the arduino, but bCNC doesn't seem to recognize that it's connected; I'm guessing since it's older and didn't have the expected status/reporting code written yet?  Is there a recommended gcode sender I can use for 0.6b on Linux?

chamnit, 2015-11-08:
@Shanjaq : v0.6b is very old and not recommended for use. Most Grbl senders will only support v0.8 and after.

closed 2015-11-08 by chamnit