grbl/grbl #840: Little question about spindle control

onekk, 2015-11-06:
I have wired my grbl 0.9j with the inverted Z limit and variable spindle option.

But i have to drive a SSR with the D11 to control the spindle, how to output at his pin a clean 5V to turno o the relais without a PWM on it?

Regards

Carlo D.

chamnit, 2015-11-06:
@onekk : You have a few options. 
- You can comment out VARIABLE_SPINDLE in the config.h and re-compile. This will swap D11 and D12 back to the old Grbl pin diagram. 
- If you can't swap the pins easily, you can play with the SPINDLE_MAX_RPM and MINIMUM_SPINDLE_PWM config options to output 0V or 5V a very limited range of PWM. You may need to set the S words to something like 1 rpm to turn on to 5V though.
- You can use v1.0 in the edge branch, which has rpm max and min settings. If rpm max is less than or equal to rpm min (both zeros), the PWM pin will output 0V or 5V only with the need for the S word (spdinle speed).

onekk, 2015-11-07:
Thanks,  @chamnit, if i set the output to 5V it will output a pwm or a "clean"  voltage,  it is not very clear this point,  when VARIABLE_SPINDLE is active the pin will output a pwm or a voltage. For now my SSR is working,  but someone affirm that it will not like a PWM and need a "clean" voltage. 

TIA and Regards

Carlo D.

chamnit, 2015-11-07:
Variable spindle will always output PWM. But it will output a 100% duty cycle at 5v when configured. This should be a relatively clean signal. If your system can't handle that for some reason, you'll either need to disable variable spindle or modify the spindle control source, which is pretty simple.

TadyTheFish, 2015-11-07:
If i understand this correctly you need a clean voltage from 0-5v? If this is the case all you need is a low-pass filter.

chamnit, 2015-11-07:
@TadyTheFish : Yep, that will work too. You don't want the capacitance to be too big. Just enough to get rid of noise.

TadyTheFish, 2015-11-07:
To big capacitance wil cause the spindle controller to have a slow response on changing. It depends on input resistance of the controller which is probably in mega-ohm range (opamp at input).  I'm thinking nF range. What is the frequency of PWM? 490Hz?

EliteEng, 2015-11-08:
I think the question should be "why use variable spindle (PWM) if you only want a digital output?"

If you want a digital output on D11 you can disable variable spindle and change the pin configurations in the cpu map.

This is the reason when I did the Mega2560 pin map I left the spindle enable pin intact to allow for turning on relays etc. ( The Mega has spare pins to allow for it too.)

onekk, 2015-11-08:
The question is right, but i prefer not to modify the grbl code, as i thinks my skill aren't good enough, without some sort of guidance.

For now I have left all the thing the same and set an appropriate S value to turn the PWM at 100%, some preliminary test make me believe that the SSR is working fine but i didn't have an oscilloscope to test it in a proper way.

The SSR is driven through a 2n3904 transistor for now that switch to GND - pin of the SSR (the + pin is connected to 12V)

If someone could write some advice on the grbl wiki I think it will be a great help for many users.

Many thanks to all, and in particular to @chamnit for Grbl.

Carlo D.

TadyTheFish, 2015-11-08:
What kind of spindle are you using? AC or DC?

TadyTheFish, 2015-11-08:
Why not make it a variable spindle? If you are using a DC motor use a mosfet driver and a mosfet. Like this http://www.eevblog.com/forum/projects/pwm-mosfet-driver/?action=dlattach;attach=15891 (just put you motor where the load is and don't forget the free wheeling diode)

And if it is a AC make a circuit like thishttp://pcbheaven.com/circuitpages/PIC_DCV_Controlled_AC_Dimmer/

onekk, 2015-11-10:
The dimmer don't work the SSR is a zero crossing type so it is not dimmerable, plus my Spindle (Kress 1050 FM-1) have a speed regulator in it with a soft start function, many thanks for the info it will be useful maybe when I attach a laser to the CNC.

TadyTheFish, 2015-11-11:
I have a laser cnc. 5W infra red with a microscope okular focus lense. I have a relay for laser cobtrol. Because i set the current of the laser with a potentiometer. 
I don't need to set the current with program i set it based on the type of material i want to cut. So that is why i have manual current control :)