grbl/grbl #841: Z probe not working on 0.9J (even with variaIble spindle disabled)

Audio-Rochey, 2015-11-08:
Hi folks,

I've got a system set up, variable spindle disabled, using an uno and a CNC Shield 3.0

I have the following settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=19 (status report mask:00010011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=50.000 (homing seek, mm/min)
$26=25 (homing debounce, msec)
$27=5.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

When the machine is unlocked, ($X), I can get alerts on XYZ just fine, I can also disable hard limits and see the switches when I do a ?
But -- when I try homing, the system completely ignores the Z switch and goes crashing into the top.
I've added a 0.1uF cap accross the switch terminals and a 390Ohm resistor to 5V to try that with no luck.

Any ideas? I feel like I'm going nuts here.

Cheers

Rochey

Audio-Rochey, 2015-11-11:
Is there a better forum to ask this question?

chamnit, 2015-11-11:
@Audio-Rochey : No asking here is fine. Forgot to respond to your first post.

Not exactly sure what's going on with your system, but it's most likely something with the wiring. What I would do in your situation is to remove as many external variables as possible. Then test and validate them. And add one variable at a time and retest. I would start by disconnecting everything from the Arduino except the limit switches (if you can't do that, just a wire connected to GND will work fine).

It would also help if you explained how you wired your switches, which pins they are wired into, and if you made any modifications to the source code/enabled non-default config.h options.

PicEngraver, 2015-11-11:
Not sure just how you have caps installed, but they do need to be located AT the Arduino input pins, not at the switches.  Are you sure you have the Z switch wired to the correct pin?  Does tripping the switch manually trigger a limit condition?  I am using 9J set up like you describe (but without soft limits), and it works fine, so I suspect a switch/wiring issue also.
John

eduardsik, 2016-02-18:
Same problem here, like Audio-Rochey. I trying with hard limits to get error from switches by manually pushing them. working on XY, but on the Z not. I swapped X and Z switch and working good on the "wrong" switch. When I treggered multiple times Z switch, i get hard limit error, but need to push like 10 times. Its for sure not debounce or noise error, I measured. 

For me it seems software error.

chamnit, 2016-02-18:
@eduardsik: It's not a software error. It's either a wiring error, damaged Arduino pin, broken switch, or most often electrical noise. There is no debounce logic for hard limits and the z axis works exactly the same as the x and y axes. Please try the recommendations posted in these treads first. 

Also, please don't post multiple times about your issue. Once is sufficient to get a response.

closed 2016-02-18 by chamnit

wslewis, 2016-12-17:
I'm having the same problem with z limit switch and its NOT the wiring or a broken switch or a damaged arduino pin or electrical noise. Please advise

chamnit, 2016-12-17:
@wslewis : Please open another issue next time. Also, can you explained in more detail of what your problem is?  Tell us what your settings are, how you wired your z-limit (should be on pin 12), and what version of Grbl you are using.

wslewis, 2016-12-17:
@chamnit Thanks.. that was fast! I found your post elsewhere and yes z limit switch is now on pin 12 and now working properly. I still don't understand why a homing cycle is necessary or desirable - yes, I'm that new to cnc. Just got my machine up and running a few days ago. The learning curve really applies to the software and feeds and speeds as I'm finding out. Any tips or good sources of info?

wslewis, 2016-12-17:
One more thing.. I just want to say thanks to those who put grbl together because not only is it free but despite this it actually works very well. It gets the job done. Thanks again. 👍 x 💯

jahnj0584, 2016-12-17:
join the #CNC Router Tips facebook group and learn about chipload

www.daycounter.com/Calculators/GCode/Feed-Rate-Calculator.phtml

chamnit, 2016-12-18:
@wslewis : Homing gives you a fixed reference point in 3d space. From it, you can restart a failed job, test if you exceed machine travel by running your job in check mode, or do duplicate parts by moving the work origin (part zero) around the machine space and resending the job.