grbl/grbl #843: Homing/Z Limit Issue with 0.9j. (working on 0.8c)

icetea0, 2015-11-12:
Homing worked fine on 0.8c but after I updated to 0.9j and rewired the z limit switch to pin 12 accordingly, I started to get "ALARM: Homing fail" messages. This happens after the the z limit switch has been tripped(wired correctly to NO and C). I even re-flashed 0.8c and re wired the NO connection to pin 11 to confirm that its working earlier today. But 0.9j simply refuses to work. I believe I had the same issue with 0.9i. After confirming that the limit switches work with 0.8c, I re compiled and flashed 0.9j onto the arduino uno(//'d out #define INVERT_SPINDLE_ENABLE_PIN). But I still got the same error message. 

>>> $H
ALARM: Homing fail
ok

Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]  

I have some one way resistors connected to the NO outputs from the x,y and z limit switches on one side and ground on another exactly as shown here: http://www.instructables.com/id/End-Stop-Limit-Switch-Problems/ 
I have searched through all the related posts but could not find an answer.

chamnit, 2015-11-12:
@icetea0 : I recall altering how NC switches behave by default in Grbl v0.9. You may just need to invert the limit pins via $5. Or you may not be clearing the limit switch with the homing pull-off value. Try increasing this to 5mm or so.

icetea0, 2015-11-12:
@chamnit : I tried setting $5=1, the behavior changed and every time I try to home, the machine wouldn't move and just throws the same "ALARM: Homing fail" error (It would make contact with the switch before throwing the error with #5=0). I am certain that we can rule this out as a problem since I wired all the limit switches to be normally open and setting $5=1 would simply trip the switch constantly. 

The console inputs and feedback are as follows:

> $5=1
ok
> G10 P0 L20 X0 Y0 Z0
ok
> $22=1
ok
> $21=1
ok
> $20=1
ok
> $h
ALARM: Homing fail
ok
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
> $h
ALARM: Homing fail
ok
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
> G21 G91 G0  Z-2
error: Alarm lock
> $X
[Caution: Unlocked]
ok
> G91 G0  Z-2
ok
> G91 G0  Z2
ok
> G91 G0  Z2
ALARM: Soft limit
[Reset to continue]
> $X
ok
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
> $H
ALARM: Homing fail
ok
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]

As you can see, I tried manually jogging the z axis and it would give me the soft limit alarm. This is to be expected since $5=1.

I then set $5=0 and $27=6. When I homed, I could clearly see that the machine would correctly pull-off after it has tripped the z limit switch. The same Homing fail error is thrown once the machine successfully pulls off.

I'm all out of ideas as I have tried nearly all the solutions you suggested to people who had similar issues but could still not resolve my issue. I first encountered this issue since a month ago after flashing grbl 0.9i onto the arduino. And after many failed attempts to troubleshoot, I simply gave up homing altogether. But I am working on a major project that needs the repeat accuracy so I desperately need to fix this issue now.

My GRBL settings are as follows.

> $$
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=3 (dir port invert mask:00000011)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=750.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=6.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=188.947 (z, step/mm)
$110=8000.000 (x max rate, mm/min)
$111=8000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=500.000 (x accel, mm/sec^2)
$121=500.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=290.000 (y max travel, mm)
$132=100.000 (z max travel, mm)

109JB, 2015-11-12:
I don't think you can blame GRBL this early in the game since many people are using the 0.9j homing cycle without problems, myself included.

I am totally unclear as to how your system is wired. You state:

"I have some one way resistors connected to the NO outputs from the x,y and z limit switches on one side and ground on another exactly as shown here:"

First of all, there is no component that I know of called a one-way resistor. Could you be thinking diode? The link you included in your first post only talks about installing a small capacitor between the limit pin and ground. Nothing is mentioned about resistors except another link to a bunch of forum posts. If you are using a capacitor, and you have it installed at the switch as you describe, then you have it in the wrong place. It needs to be at the arduino. A drawing of the components in the switch circuit and how they are connected would be helpful.

For testing, I would say just remove everything except the actual NO switch and test. It should work fine without anything else, and then you can start adding back in the filter components to locate the problem.

chamnit, 2015-11-12:
@icetea0 : The Homing Fail error is thrown in a limit set of circumstances. 
- The limit switch is not triggered in the allotted travel during the search phase. The search distance is computed as (1.5x)axes max travel. This isn't your problem, as it finds the switch fine, and goes onto the locate phase with that initial pull-off.
- A soft-reset or safety door (when enabled in config.h) are issued during a homing cycle. This will exit the homing cycle as an alarm fail. Again, this is not likely your problem since your error is generated without your input.
- The limit switch is not released after the pull-off motion. Or the pin still reads as triggered at the end of the pull-off motion. This is likely the cause. Either your limit switch is still not cleared at 6mm, or you have something wrong with your wiring that is causing a slow decay in the pin. Meaning you may have too large of capacitor that is discharging too slowly. Typically you see nano or pico-farad sized capacitors for limit switches.

So I would echoes @109JB suggestion, please draw up a diagram of your limit switch circuit. I think this is cause and will likely be able to help you with this.

MaxLRCNC, 2016-03-24:
I had similar issues resolved when Arduino EEPROM was completely erased and grbl re-loaded.
There is a small issue though. I include it in this thread as it seems related and relevant.
I want and have used in v0.8 - (A) #define HOMING_FORCE_SET_ORIGIN -, but two things I have managed to prove using a fresh install with fresh grbl, and only stepper parameters, home cycle enabled and proven limit switches enabled. When I removed the // from (A), the homing works: Z to limit, then backs off,  X to limit SW trigger, then Y to limit SW and back off, then... X backs off. By backing off I mean that it respond to the homing pull-off. I tried to manipulate the stepper before X "backed off" and it was energized. The Stepper de-energised after backing off. I also tested the limits switches. I also recompiled with the only factor being the (A) was commented out. Now same thing and now the routine run with Z, X, Y and they back off in the order that they are called. 
Lastly, (kind of important) the (A) doesn't actualyl work. No change in behaviour nor coordinate..nothing. 
Thanks for grbl !!!

iw2fvo, 2016-06-13:
I am not able to get rid  of homing fail using version 0.9j.
I am using NO swithes but no resistors and no caps.
Any help please ?
Thanks , regards

chamnit, 2016-06-13:
@iw2fvo : Grbl v0.9 switched pins 11 and 12 to access the hardware PWM on pin 11 for variable spindle speed output. So the z-limit pin is now on 12.

closed 2016-06-13 by chamnit

iw2fvo, 2016-06-14:
It seems to be ok on my wiring.
I am not able to override homing fail at all.
I would like to remove the control of Z axe since I am using a LASER source: Could you please tell me the sequence to do that ? How to modify config.h > compile > load on my arduino Please ?
Could I have the modified  config .h ? .How can be the laser ON/OFF be controlled please ?
Thanks
for all
regards,