grbl/grbl #845: Sent lines different from GRBL echoes

109JB, 2015-11-14:
First. let me appologize for the length of this post, but I feel this is a very important issue.

I have been doing some more testing to verify GRBL's running of g-code files. I decided to get a serial port sniffer program to see exactly what is being sent, and what is being received to verify if any of these uncovered errors are from my GUI, or something else. What I have determined is that the data sent to GRBL is correct, but the echoed return is not correct, meaning the error happens somewhere either in the arduino, or in the GRBL code. See the following description to see how I came to this conclusion.

As further testing, I made a small program to use for testing. This program has streaming code that is very streamlined with nothing extra except what is absolutely necessary for streaming and for detecting GRBL error messages. Nothing else. Both the send-response and the character counting protocols can be used and both have generated errors. The only other thing the program does is record the GRBL echoes so they can be compared to the original code for discrepencies. The files can be streamed to GRBL without any timed status requests and the following errors were encountered with timed status responses turned off.

 I am testing with a CAM generated g-code modified to have only G0 commands. This results in very fast streaming. If I slow the stream down a little by placing an intentional delay in the streaming code, it works as expected without errors, but at full speed GRBL doesn't always echo back the same line that was sent and/or throws errors. I have verified that the lines sent are correct by using a serial port sniffer program. 

For example, in a recent run using the character counting streaming, the following lines were sent at the very start of the run:

G01G21G90X-0.5375Y3.5375Z3.3598F4000
X3.8999Z1.0234
X3.896Z1.0262
X3.8645Z1.0475
X3.8566Z1.0527
**X3.8212Z1.0742**
X3.8173Z1.0765

The above first 7 lines were immediately sent by the GUI and the character count, including a LF at the end of each line coincidentally totals 127 characters which filles the RX buffer, and the sent lines were verified using the serial port sniffer. The response from GRBL was:

[echo: G01G21G90X-0.5375Y3.5375Z3.3598F4000]
ok
[echo: X3.8999Z1.0234]
ok
[echo: X3.896Z1.0262]
ok
[echo: X3.8645Z1.0475]
ok
[echo: X3.8566Z1.0527]
ok
**[echo: X3.8999Z1.0742]**
ok
[echo: X3.8173Z1.0765]
ok         

Note the 6th line. The echo from GRBL has the wrong X value but the correct Y value. Remember that all of the above information is from a serial port sniffer and not from my own program. However, my test program shows the exact same thing when comparing the echo from GRBL to the original G-code file, showing that the 6th line does not match. A second run of the same file got through this point fine without an error.

==================================================================================

During another test run, the lines sent were:

X2.8811Z1.167
X2.889Z1.1702
X2.9362Z1.1889
X2.9441Z1.192

with these responses from GRBL

[echo: X2.8811Z1.167]
ok
[echo: X2.889Z1.1702]
ok
[echo:]
ok
[echo: ]
ok
[echo: .9362Z1.1889]
error: Expected command letter
[echo: X2.9441Z1.192]
ok

This was toward the end of a 90,000 line G-code file run using the send-response protocol during check mode. Like before, the serial port sniffer confirmed that the line that generated the error is sent correctly with a LF used as an end of line character. but note that GRBL echoes 3 times for that single line and drops the "X" in the final echo.  The serial port sniffer detected no extra end of line characters in the sent data stream.

==================================================================================

I'm not sure what to make of this.  The above errors, especially the first example is weird and makes me think it may be a GRBL issue, but I am not sure of that.  I am running these tests at the absolute maximum speeds that can be run and running files longer than probably anyone else has before, so maybe I am just exceeding the capability of GRBL, or perhaps this is still related to the baud rate issue I reported. As I said, if I slow the streaming loops down with an inserted delay, but with the baud rate kept the same, the errors go away.

Last week I had experienced errors that seemed to be tied to baud rate.  Indeed those errors did go away when only the baud rate was changed, and this was verified using a simple code within my GUI to compare the original file to the GRBL responses. My new test program uses the same comparison, with the main difference when running being that the new code doesn't include all of the stuff that My GUI implements for canned cycles, tool change support, etc, so the run loop is as streamlined as possible. 

On my Arduino Nano clones I still haven't been able to generate errors on 250000 baud or 76800 baud, but on my Aruino Mega clone, the opposite is true and the errors appear at 250000 baud, but not at 115200 baud. The Nano uses a ch340G USB to Serial chip but the Mega doesn't, so perhaps the USB chip is the culprit.  I'm not sure how to test further, but based on the serial port sniffer program showing that the lines are being sent correctly it seems to me that the error must be either related to the Arduino hardware, or to GRBL code. I am at a loss at this point as to how I can determine which it is, but  adding in a slight delay in the streaming loop seems to work ok, but slows things down a bit.

Anyone have any ideas?

kfoltman, 2015-11-14:
How about writing a simple Arduino sketch to echo a character, in order to find out if it's a grbl issue or a wider issue affecting those platforms?

chamnit, 2015-11-14:
@109JB : Interesting results. Especially the difference between the Mega and your Nano. The Mega data sheet show the same serial timing error as the Uno's 328, so that "should" rule out timing error, but I think there are a lot more variables here than we can isolate that for sure.

Just curious, how does lowering the step per mm by half or more change the behavior? I wonder if the stepper ISR is taking more time than expected. In theory, the execution time should be the same, but the stepper ISR will fire at half the rate. This could help rule out the stepper ISR.

I would also do what @kfoltman suggested and try the Arduino sketch. This might tell you something, but it's hard to replicate the exact timing of everything. Perhaps put a random delay in there that is about 1-4ms since that is about the time it takes Grbl to parse and plan a g-code motion.

109JB, 2015-11-15:
I was gone most of the day today, but I did write, or rather modify, an arduino code that simply receives a serial string and then echoes it back with an "ok" afterward to simulate the responses from GRBL. I will describe in detail how the Arduino code works:

1 - The arduino detects a serial event and reads a single character and buffers it in a string variable. 
2 - The above is repeated until a "\n" is detected
3 - Once the \n is detected, the stored string has "[echo: " put on the front, and "]" put on the back to simulate the GRBL response
4 - The string is then sent using the Serial.println() command
5 - An additional Serial.println("ok") command is used to simulate the GRBL "ok" response

Here is an example of the first few lines of a 10,000 line g-code file followed by what was returned (stripped of the "[echo: " and "]")

G01G21G90X-0.5375Y3.5375Z3.3598F4000
X3.8999Z1.0234
X3.896Z1.0262
X3.8645Z1.0475
X3.8566Z1.0527
X3.8212Z1.0742

G01G21G90X-0.5375Y3.5375Z3.3598F4000
X3.8999Z1.0234X
3.896Z1.0262X
3.8645Z1.0475X
3.8566Z1.0527X3.8212Z1.07
42

I also found that I can eliminate all errors whether using the Arduino test code or GRBL firmware and using either character counting or send response, by inserting a small delay into the loop that sends data to the Arduino.

A delay in the Arduino test code does not eliminate the errors. Also, as requested by @Chamnit , changing the steps/mm in GRBL doesn't affect whether errors are generated, but changing the max rate does. The steps/mm would affect the GRBL code, but not how fast GRBL needs to stream the data. Changing the max rate will however affect how fast data needs to be streamed. so the errors are tied to how fast the arduino needs to get data from the serial stream. Also, using a code with realistic move lengths eliminates errors, also indicating that the streaming speed is the issue.

Based on getting the errors using either GRBL firware, or using the Arduino test code, I believe the problem is related to the Arduino hardware I have. All of my testing has been with Arduino clones, not with genuine Arduinos. I plan to get a genuine Arduino Uno in the future to see if the same holds true for the genuine Arduinos, but it may be a while before I can get one.

It seems that the delay required is also dependent on the baud rate I choose. For some baud rates using send-response, I don't need any delay for them to work correctly. For other baud rates, a delay is needed even with send response, but only a very small one (1 to 2 uS). For character counting, all baud rates need this delay, but some only need 1-2 uS while other baud rates need about 5-6uS, and yet others don't ever work right.

I do want to point out though that just because GRBL doesn't throw an error doesn't mean there isn't one. For example, my previous post highlited one example where a mismatch in what was sent and what GRBL echoed back did not register an error from GRBL because the line was still in proper G-code format, but the coordinate value was wrong. Some of the errors I have found were only discovered by using code that does a line by line comparison of the GRBL echo to the original G-code.

euchcat, 2015-11-15:
I had this kind of behavior with a AT90S2313 and max232 long ago but look on
the side of your oscillator on the 16U2 and Capa 1uF (Ucap / C8 in the
diagram of the Arduino Uno)
On clones of Arduino, it is common for the nearest scrap components are
mounted to lower the manufacturing cost.
Anyway, the best solution is to run on a GRBL official Arduino board.

109JB, 2015-11-17:
@euchcat - Thanks for the info. The clone boards don't use the 16U2 and I think it may be a bit premature to blame it on just the clone boards since as far as I know nobody has done any testing such as this on the genuine boards. Until there is testing on a genuine board, they are still suspect too.

I have done some more testing and have made some more determinations in regard to my clone boards. I made some different Arduino codes and some simple visual basic test programs to examine various things. Here are the highlites:

1: I tried two Arduino codes for receiving the serial data. The first code reads the data one character at a time until a "\n" is received using the Serial.read() function, while the seond reads the data as an entire string using the Serail.readStringUntil() function. Although functionally equivelant, the second code that used Serial.readStringUntil() results in far fewer errors in reading and can tolerate faster streaming. This is an Arduino code and not GRBL, but I don't know how GRBL works in this regard. 

2: The above codes also echoed back the received line. In code #1 I used the Serial.println() function to send the string back, relying on the println function to append the \r\n to the end. In the second code I added a line to append the \r\n to the string first and then used the Serial.print() function to send it. This resulted in fewer errors. I found this because I was seeing that the \r\n in the responses could be in different spots than where they should have been. For instance, a couple characters late, or even missing with 2 lines of response before a \r\n.

3: Although not sure, I have read some stuff on the internet that indicates that to go from Serial.read() to Serial.print() requires a bit of time and can result in errors. Some of the posts I have read suggest the need for an added intentional delay in the Arduino code if going immediately from read to print. I have not really found this to be true but thought I should mention it.

4: From my testing, It is apparent to me that for whatever reason, the Arduinos I have need a period of time between successive serial events.  I have found this to be true with GRBL and with my test Arduino codes. GRBL seems to need to have a longer delay than the test codes that put \r\n on the end before sending. 

5: The send-response protocol is 100% fine, sort of.  What I have found is that I can't generate any errors using send-response as long as the timed status requests are turned off. With the timed status requests turned on, what I believe can happen is that the timed request can be sent such that it results in two serial events being too close together for the Arduino to handle. This is a very rare event. I would estimate that I have been able to do this about once every 300,000 lines of code or so while using totally unrealistic g-code programs and fast timed requests. With realistic programs and slower requests (0.2 sec) , I don't think this would really even be an issue.

6: The character counting protocol needs about a 4 mS delay with GRBL between send events to prevent sending faster than that.  In most cases, this won't be an issue, but in the beginning of a send, the first few lines can definitely be sent faster than this, resulting in errors. This delay does slow things down a bit but in practical use I don't think it would really affect run times for realistic G-codes. The delay just works in cases where it needs a slow down to prevent errors is needed.  The type of errors that can be generated are the type that won't result in a error from GRBL. My GUI appeared to be working fine in Character counting mode, but in fact it wasn't because lines with subtle errors were being sent. It wasn't until I created a routine to do lie by line comparison of the sent vs echoed lines was this discovered. With the delay the errors are gone

7: In all cases, the errors are dependent on baud rate selected. The Mega clones and the Nano clones are a bit different in regard to this too, but my Mega clones use the Atmel 16U2 usb to serial chip and the Nanos use the ch340g chip. I think this could be the reason for the differences here. I do note that the Nanos have 2 crystals, one for the USB and one for the MCU (12 MHz vs 16 MHz), whereas the Mega only has one 16 Mhz oscillator. Not sure what to make of this, but suffice to say they are different in USB to serial hardware used.

chamnit, 2015-11-17:
@109JB : I thought I posted this comment, but I must have not. One way to isolate the USB-serial chip is to sniff the RX/TX pins on the Arduino directly. On the Uno, pins 0 and 1 are tied to both the USB-serial and 328p. So you'll be able to see the traffic between the two. If Grbl is receiving data through the TX line correctly, this means it's something to do with the serial comm to the 328p and/or Grbl. If it's getting garbled before the 328p receives it, then it's the USB-serial chip or host-side USB.

109JB, 2015-11-17:
@chamnit 
Are you referring to the test where the TX and RX pins on the Arduino are shorted?  It seems that this would be a pseudo test of the USB to Serial chip and is definitely something I can do.

It seems to me that a better test would be to use the sniffer program to monitor a second com port  that is hooked to the RX/TX pins of the Arduino. This would enable running GRBL as normal,but I'm not sure how I would do this with the hardware I have since my computer doesn't have an actual serial port on it. I would have to use another USB to Serial adapter which would introduce another adapter in the loop. Then if errors were detected I would not know if it was the second USB adapter or not. 

Any advice you can give on how to test would be appreciated.

Thanks

chamnit, 2015-11-17:
@109JB : Not sure what you mean, but the Arduino Uno pins 0 and 1 are hardwired to the USB-serial chip and the 328p. All data traffic should be able to be read from those pins. I was simply thinking that you could another Arduino there to read the incoming TX data and just spit it out to a terminal on your computer. Disclaimer: I've never done this, but it seems like something that can be done.

109JB, 2015-11-19:
I was able to do some testing of the hardware. I did this using the loopback procedure recommended at the Arduino forums. Basically the procedure is to short the TX and RX pins on the arduino and also place a jumper between the reset pin and ground. This holds the MCU in reset mode and thereby holds the pins low. This takes the 328P out of the loop so that the test is checking the host computer hardware driver, USB cable, and USB to serial converter only. What is happening is because the TX and RX pins are shorted together, the received data stream is simply sent back through the USB/Serial converter and back to the computer.

On my Mega clone that uses the same Atmel 16u USB to serial converter that the genuine boards do, the test was immediately successful and the data stream was echoed back to the host computer as expected. 

On the nano clones I have this test was a little more involved. It didn't simply work with a couple jumper wires like the mega did. The Nano clones I have use the ch340g USB/serial converter. I did find that shorting the pins directly at the ch340g worked, so I figured there must be something in the circuit making the loopback test not work. I couldn't find a schematic for this clone board, but a little poking around with a meter showed that the TX and RX pads from the ch340g were connected to the RX/TX pins of the 328P with a 1 k-ohm resistor in each line just like the 16U on genuine boards. However, the 16U has separate pads to power the indicator LED's. On the ch340g, the LED indicator lamps were being powered from the TX/RX lines with an LED and 1k-ohm resistor.  I suspected that the LED/resistor circuit might be pulling just enough power that just shorting the TX and RX pins together wouldn't work. So, out came my pocket knife and I broke the TX and RX LED's off to eliminate that circuit. It worked and the loopback test was working.

I then wrote a test program  and found that on the MEGA 2560 clone with the Atmel 16U USB/Serial chip, sending lines faster than about 2.9 milliseconds would result in errors. The Nano with the ch340g USB/Serial chip did a bit better. The Nano clone didn't show errors until lines were sent about every 2.2 milliseconds. It is interesting though that the ch340g does better the faster the baud rate. At 2,000,000 baud you can send at 1.7 mSec intervals without errors. The Mega was pretty constant at all baud rates. So these delay times are what the USB/Serial hardware requires in order to work properly.

Since I had to "modify" the one Nano for the testing, I used an unmodified nano with a Arduino test code and with the same test program on the host computer found the required delays to be the same across the baud range tested (76800, 115200, 230400, 250000, 1M, 2M), with the exception of 115200, and 230400. 115200 required a longer delay than the USB/serial hardware alone required, and 230400 would not work at all without errors. So there is a portion of this that is tied to the baud rate matching between the USB/Serial chip and the 328P. This testing confirmed though that even thought the LED circuits were interfering with the loopback test, they are not interfering with normal operation.

The testing also indicates to me that simply blaming the errors on the clone boards isn't correct. The Mega clone I have uses the same Atmel 16u USB chip that the genuine UNO, Mega and Nano boards use and it showed performance similar to, but actually a bit worse, than the ch340g chip on many of the clone boards.

What does all this mean?  Because of this, I have changed my opinion and now believe that the character counting protocol should not be used unless the host program incorporates some method to insure that commands can't be sent faster than 1 line every 3 mS. This is because the error introduced may simply be a missed character in the data stream. For example, X12.3456 might come through as X1.3456.  The safest method is to use the send-response protocol since it will not be sending faster than this since GRBL can't receive and parse a line faster than about once every 4 mS anyway

I will leave it to @chamnit  to decide if this should be noted in the WIKI.

AlexHolden, 2015-11-19:
Thanks for your diagnostic work, John. I wonder if the FTDI chip they used on older Arduinos suffers from this problem.

I imagine if one was to bypass the USB stack entirely and use a hardware UART and level converters instead (e.g. the RPi's on-chip UART connected to an Arduino Pro Mini), you would be unlikely to encounter the corruption problem, and would probably benefit from slightly lower latency too.

P.S. As a side-note, 76800 baud doesn't work with the ch340g Mac driver either (requesting 250000 returns an error status; 76800 returns success but won't actually connect, so I'm guessing it configures the baud rate generator wrongly). The driver is binary only and it's not clear who wrote it or how to contact them, so there's not much prospect of getting it fixed. I think in future I'm going to be avoiding clones with ch340g's on them.

chamnit, 2015-11-19:
@109JB : Agreed. Thanks for digging so deep into this. And good to know that the 328p and Grbl is not the cause. I think we can definitively state this now with your results.

However, I'm still uncertain whether it's the UBS-serial chip itself, the host computer/GUI, or the USB protocol itself. Your experiment doesn't quite isolate the cause yet, even though it certainly shed a ton of light on the issue. 

I wonder if this is more related to the amount of data being sent through the USB protocol itself. In my limited understanding of USB, it's surprisingly complicated and sends data in packets. It either waits until a packet is full to send one or it waits for a time-out to send an incomplete one. Could you vary the message size from a few bytes to 256 bytes and redo your test? I think this may perform differently. 

Also, I wonder if the computer has something to do with it. Is it possible to use another computer and/or OS? If behaves the same, this would rule out OS and host (but not GUI).

109JB, 2015-11-19:
@AlexHolden - I think you are correct that using the 328P's built in UART directly would work for getting rid of the latency from the USB hardware. I was actually thinking the same thing and was looking last night into getting an Arduino Pro Mini and a MAX232 chip so that it could be connected to an actual RS-232 port on a computer rather than a USB to serial conversion. Don't know if I will actually do it, but I gave it a thought. Certainly using something like the RPi to communicate directly could be done as well and would also eliminate the USB to serial hardware.

@chamnit - Yeah. It definitely eliminates the 328P and GRBL firmware and places this somewhere upstream. I see your point regarding whether it is the USB chip, USB in general, or I guess it could even be associated with the drivers for the chips. I can try it on a different computer to see if it is something there. I can easily do it on another Windows machine as I have several that are different make/models and OS. My main computer I run Windows 7, but have several running Windows 10. I'm not sure if running on my old XP machine will tell us anything since it is soooo sloooow to begin with.  As for the GUI causing it, unless it is something inherent with how Visual Studio or the OS handles USB or serial port stuff, I think the coding of the GUI can be ruled out. The test program I wrote just uses an ultra simple loop to send a static string to the serial port. Here is a snippet from the code showing that loop:

For i As Integer = 1 To 5000
            SerialPort1.WriteLine("0123456789ABCDEF")
            For j As Integer = 0 To delayParam
                Application.DoEvents()
            Next
Next

The "j" loop is just used for an adjustable artificial delay and I get the computer clock time before and after the "i" loop to calculate the average time for each line sent. In the serial port read event, the incoming string is compared to make sure it matches "0123456789ABCDEF". If it doesn't, an "errorCount" variable is incremented. Also here a "linesReceived" counter is incremented and, at the end of the "i" loop a comparison makes sure that 5000 lines were received and no mismatches were detected. Pretty simple. I agree though that it could still be something with how the serial port events are handled when the visual studio programs are built, or with how the OS handles them.

I can easily modify the test code to send shorter or longer strings to see what affect it has. I used the 16 character string because I thought this was somewhat close to the average string length for a typical G-code command (X1.2345Y1.2345)  I had the same thought regarding string length last night but it was too late to test it. I may have time this evening. 

I also had some other thoughts for additional testing, but at this point, other than the simple additional tests above, I think I will abandon them since simply using send-response works without issue. I may revisit this some time down the road though.

109JB, 2015-11-20:
@chamnit - I didn't have a whole lot of time last night, but was able to do a little testing and I think I may have to change my approach a little for testing longer strings.  For example, at 115200 baud, and 10 bits/character (1-start, 8 data, 1 stop), it takes (number of characters)/11,520 seconds just to send the data stream. So a 256 character line takes 22 mSec just to send the string if my math is correct. So my test program method won't be too good at calculating a delay. However, I did find that some kind of delay is indeed still needed. If I just let it loop through without an artificial delay it will result in occasional errors.  I need to look into how to do a microsecond delay in Visual Studio. There are built in functions for millisecond delay, but these are only for whole milliseconds, so I would not get the resolution I would like using the built in functions, which is why I used the wasted loop and calculate method I described.  I have found some reference to commercially available functions to do this but the price is too steep so I may have to figure it out myself.

I am thinking now that the delay needs to be not just seconds from sending one line to sending the next, but a delay from when the first line completes then a delay, then send the next, making it even more complex. If this is the case then it even further supports the need to use the send-response protocol since the time to actually stream a line of data is built in to that protocol. For example to send consecutive 256 character lines would be Send(22mS) - Delay(x mS) - Send(22mS)....

More testing to come.

109JB, 2015-11-20:
I just ordered a MAX232 Serial to TTL converter. Once it arrives I should be able to find out a bit more info about this.

AlexHolden, 2015-11-23:
Following further tests, I've gone back to 115200 on my milling machine because I'm seeing definite corruption at 250000 that I haven't encountered at 115200. It happens far more often when using character counting but it occasionally happens with call-response too. This is a RPi 2 running my own GUI connected to an Uno clone with an Atmega 16u2 converter.

If our working hypothesis is that the USB-serial converter sometimes corrupts the data when two writes happen too close together, that could happen even with call-response if a block happens to be sent too close to a status request realtime command

Interestingly I turned on echoing and when using call-response the corruption appears to usually take the form of one of the characters in the block being replaced with ">\r\n", which looks suspiciously like the end of a status reply. For example (typing from memory because I don't have Internet in the workshop), the GUI sent a command like:
G0X1Y2Z3
and the echos I got back looked something like:
echo:G0>
echo:
echo:1Y2Z3

chamnit, 2015-11-23:
@AlexHolden : Interesting result. It definitely looks like the `>\r\n\` end of a status report there. Not exactly sure why that would be there. There are a few things we could try to further isolate this. 

- Increase the serial write buffer size or reduce the rate of status reports. The write buffer isn't very big (64 bytes) and status reports have been getting ever so larger over time. This might be a symptom that the buffer needs to be increased again. This can be changed in the config.h. Maybe up to 96 or 128 bytes.

- Remove the USB-serial converter from the equation. With an RPi, you should have 5V tolerant serial RX/TX pins available on it. (Double check that 5v tolerant before connecting though). You should be able to wire these directly into the Arduino's RX TX pins 0 and 1 and bypass the atmega16U2. I do know of one GUI, GrblWeb, will run on the RPi2 and access these pins.

AlexHolden, 2015-11-24:
To take some of the guesswork out of diagnosing the problem, I've implemented a verification routine in my GUI that immediately stops sending and sends an abort command if it receives an echo from grbl that doesn't match the block it sent.

Increasing grbl's transmit buffer size doesn't appear to have made a difference. The results of changing the status update frequency are so far inconclusive.

Officially the RPi doesn't have 5V tolerant I/O (some people have reported getting away with it but I'd rather not take the risk), so I've ordered a level converter module and will try that when it arrives. It should also be possible to use it to snoop on the data coming out of the 16u2.

109JB, 2015-11-24:
@AlexHolden - Out of curiosity, you currently using a genuine Arduino, or a clone. From my testing I don't think it really matters since one of my clones has the same Atmel 16U2 USB/Serial chip as the genuine ones, but thought I would ask. I see that you have a board with the Atmel chip and your results fit with mine well.

I came up with the same conclusion as you. running a code with lots of short segments at high speed, or fast status requests, or a combination of these can result in more errors. On my testing with the ch340g chip though it seems that running at 250000 results in fewer errors whereas on my Mega board with the Atmel chip 250000 results in more errors. The errors I encountered on the ch340G @ 115200 is what started this whole deal to start with. 

Indeed, as I indicated in a previous post above, I can't get an error out of GRBL if I just stream G-code using send-response and turn off status requests. If I turn on the status requests I can get an occasional error. It is rather random, but it appears to me that slowing down the timed requests results in less errors, which would make sense.

I have been reading a lot about how the USB/Serial chips respond and it seems that a common thing that occurs when they send data back to the PC is that these chips wait for a certain number of bytes in the send buffer, or for a timeout period before actually sending the data. On some chips the timeout period can be set in the driver on the PC. However, this doesn't appear to be the case with either the Atmel 16U/8U chips or the ch340g. Not sure if this has anything to do with it or not.

AlexHolden, 2015-11-24:
The one out in the workshop connected to my mill is an Uno clone with a 16U2. Most of my GUI development is done using a Nano clone with a ch340g (not connected to a real machine).

I haven't actually tried disabling status requests entirely, however after further testing this evening it definitely looks like the errors in call/response mode happen more frequently if I increase the status request rate.

AlexHolden, 2015-11-27:
I believe I might have solved the problem with the 16U2 corrupting data at 250000 baud.

The executive summary is that the firmware in the 16U2 is buggy and I've put a fixed version here:
http://www.alexholden.net/misc/USBtoSerial_arduino.zip
You can install it using the procedure given here, except substitute USBtoSerial_arduino_MEGA2560.hex or USBtoSerial_arduino_UNO.hex as appropriate from my package:
https://www.arduino.cc/en/Hacking/DFUProgramming8U2

Please give it a try and let me know if it helps.

The slightly longer version. The firmware that Arduino (and the clone makers) put on the 16U2 is a fork of the USBtoSerial demo that comes bundled with LUFA, the open source USB stack for AVR micros. It was forked from an obsolete version of the library that no longer compiles with recent versions of avr-gcc, and the code contains several bugs and inefficiencies that the LUFA developers have since fixed.

What I did was to re-fork USBtoSerial from the latest version of LUFA, applying the various Arduino-specific changes where relevant. If you want to try compiling it yourself, unpack my USBtoSerial_arduino.zip into the Projects folder of LUFA 151115 from here:
http://www.fourwalledcubicle.com/LUFA.php
Edit the makefile and uncomment the relevant ARDUINO_MODEL_PID line, then run make (make sure avr-gcc and friends are in your path). It should build a new USBtoSerial_arduino.hex that you can install using ISP, Flip or DFU.

I'm still getting occasional weirdness when sending a large file in check mode using character-counting at 250000 baud, however the symptoms look quite different from what I was seeing before (e.g. complete lines not being echoed back), and I suspect it may have a different cause.

AlexHolden, 2015-11-29:
Using the new 16u2 firmware, I have successfully sent a ten million line test file (with grbl in check mode) at 250000 baud using the call-response protocol, while simultaneously requesting status updates at 20Hz. Echoing was turned on and every echo matched what was sent.

It took 1 day 8 hours; that works out to an average of 87 lines per second.

I did increase grbl's TX_BUFFER_SIZE to 128; not sure if that was necessary or not, but the comments in config.h suggest increasing it when echoing is enabled.

I now feel confident enough to turn echoing off and use 250000 baud call-response as the default on my machine.

109JB, 2015-12-01:
@AlexHolden - I attempted to update the firmware on my Mega2560 clone to your new firmware using the Atmel Flip programmer. Everything seemed to go fine, but upon plugging the Arduino back in it was not properly recognized by Windows. It shows up in device manager, but says the drivers for the device were not installed. I tried re-installing the drivers to no avail. I then downloaded and re-installed the "stock" 16u2 firmware for the Mega and it was working again.  Any ideas what could be wrong?

AlexHolden, 2015-12-02:
Hi John, I think the problem was that at the time Arduino released the source code for the firmware they were 'borrowing' a USB vendor ID from Atmel. I copied the vendor and product IDs from the released source, not knowing that they later got their own Vendor ID and created a bunch more product IDs without updating the source in their github repository. I think the OSX and Linux drivers are just looking at the fact that it claims to be a CDC class device and ignoring the VID/PID, whereas Windows needs to find a matching VID/PID in the driver .inf file.

Please try again with this updated version:
http://www.alexholden.net/misc/USBtoSerial_arduino.zip

Note that there are several more hex files for different boards. The only difference between them is the USB PID and the product name string; you could use the wrong one and it should still work fine other than perhaps being described as the wrong type of board in the Arduino GUI.

If it still doesn't work, the only other difference I can think of that Windows might care about is that I increased the version number from 0.01 to 0.02.

chamnit, 2015-12-02:
@AlexHolden and @109JB. Huge thanks to you guys for working so hard to get to the bottom of this. This is a very interesting development. I wonder if there is an easy way to diagnose this problem for a user that runs into this problem. And if there is a quick solution. I do know that flashing the 16U2 isn't straightforward for most, since on older boards, you have to solder a wire.

AlexHolden, 2015-12-02:
In my testing with the official Arduino 16u2 firmware, it never happened at 115200 but it always happened sooner or later at 250000. Using the command/response protocol and lowering the status update rate just made it happen less often. I would recommend not using 250000 baud without the updated 16u2 firmware.

The issue basically is that due to an inefficiency in the way the main event loop works, if there's a lot of data going back and forth at 250000 baud, it sometimes runs out of space in the Arduino->PC buffer (and it doesn't have any way to ask the Arduino to stop sending, so the data keeps coming in). To make matters worse, when this happens the buffer overflows and corrupts the contents of the PC->Arduino buffer.

109JB, 2015-12-03:
@chamnit - It sounds like @AlexHolden has the issue solved as it relates to the 16U2 equipped Arduinos. The easiest solution seems to be to just use a 16U2 Arduino or clone and set 115200 baud, or if you want to run at 250000 baud, then load up his updated firmware for the 16U2.  I will be loading up the new firmware and giving it a try later this evening.

As for the ch340g equipped clones, I am working on a GUI based work around and I think I have it. Since I believe the problem relates to PC to Arduino serial events packed to close together, I have implemented some checks for this and if necessary, rather than sending a G-code line and a status request in 2 separate events my test code concatenates the g-code line and the status request in one event. My other computer is currently running a 1,000,000 line g-code for a first test. I will perform several more before I declare it works though. Once I have time to do more testing I will report back. 

I'm hopeful that my workaround will work, but if a gui doesn't implement something like this then I think the best solution is to just stick to the 16U2 equipped boards. I have already ordered and UNO clone with the 16U2 for myself.

109JB, 2015-12-03:
Did some testing of the ch340g equipped nano using my work-around scenario.  I am still getting echo mismatches, but no grbl error reports. Examining the echo mismatches leads me to believe that GRBL is now getting the correct data, but the echos are getting messed up. I say this because of the nature of the echo mismatches. For example, for these 2 lines of g-code sent:

X3.1486Z1.331
X3.1643Z1.3342

the expected echos would be:

[echo: X3.1486Z1.331]
[echo: X3.1643Z1.3342]

but what I got was:

[echo: X3.1486Z1.ho: X3.1643Z1.3342]

which is the first part of the first echo with the last part of the second echo. This did not result in an error by GRBL. This looks to me like GRBL got the data correctly and echoed twice but possibly due to a buffer problem in the ch340g to PC reply buffer it got corrupted.  Once I get the MAX232 chip I ordered I will set up to test this by monitoring the grbl replies at the TX pin on the arduino on a second serial channel using a "real" serial port and comparing this to what the ch340g is sending. My suspicion is there will be a mismatch. It sounds like the problem is very similar to what @AlexHolden found with the 16U2, but we have no firmware to modify for the ch340g. At least not that I can find.

109JB, 2015-12-03:
I just finished trying out @AlexHolden new firmware on my Mega2560 clone. The most recent firmware loaded fine with the Atmel Flip programmer and is quite easy to do. I just got done running 1,020,001 lines of g-code through grbl using my test streaming program with no artificial delays. The results were:

Cycle time (sec) = 9321.464 (2 hr 35 min 21.464 sec)
Lines/sec = 109
mSec/line = 9.13868123658702
Total lines = 1020001
Echo fails = 0 
grbl errors = 0

Everything worked great. Thanks Alex.

AlexHolden, 2015-12-04:
That's great, thanks for letting me know it worked for you.

I have moved the firmware from my own server to a fork of the Arduino github repo:
https://github.com/AlexHolden/Arduino/tree/master/hardware/arduino/avr/firmwares/atmegaxxu2/arduino-usbserial
I'll try sending them a pull request; you never know, it might find its way onto future official boards.

P.S. I only this morning realised that on boards that have an 8u2/16u2, they have the same oscillator frequency and baud rate generator hardware as the main CPU, so as long as grbl is using the same algorithm as USBtoSerial to configure the BRG, they should match each other very closely at every supported baud rate. Even if 115200 baud is theoretically out by a few percent in absolute terms, both ends of the serial link will be wrong in the same way.

longtolik, 2016-10-26:
Hello, I've faced this issue just a day ago, so, I'm sorry if I'm wrong.
But my question is - what if to try to implement the parity control?

109JB, 2016-10-28:
@longtolik : I could never get parity to work on the ch340 chip.

longtolik, 2016-10-28:
@109JB : Ok, then the **checksum** also can be implemented as a comment to G-code block.
For instance, a lot of sensors are designed so.
But it seems to me that **it is not a transmission error** at all.
Otherwise, how putting a delay can affect?  We send, say: G01 X25.2876348 Y56.8765439 F500 - this line is transferring anyway, do we add a delay or don't and errors must occur in both cases.
I resolve my problem reducing lines, instead of above one, I send: X25.28Y56.87 (in my case one hundred's of millimeter accuracy is enough). Such short lines let me put some delay after sending. Result packets sending frequency remains the same. 
I use FTDI USB-serial chip, connected to Rx/Tx pins of ATmega328P. Connecting HC-05 Rx to FTDI Tx, I was monitoring G-code receiving by 328P in Terminal window. They always were very clean, without artefacts. So, now I've decided that Grbl controller just cannot do its job when the data are going too fast or G-code lines are too long and complicated to be analysed. And Grbl skips some bytes...
Thanks.

Strooom, 2016-10-28:
@longtolik "But it seems to me that it is not a transmission error at all."
I can confirm that characters **can** be lost in the downstream from PC to GRBL, and when that happens GRBL may rise an error or not.. (the modified command may still be valid for GRBL, but may still destroy your job...)
When doing 3D-milling you typically have a large file with many small moves. A transmission error can result in a move being lost, and this typically means subsequent moves can be offset..
I've had this problem appear 3 times over 6 months, but every time it ruined the piece. 

I've decided that this unprotected comms link, is one of the more important things to improve and so I am adding error-detection on the serial communication.
* add a checksum/CRC to each message.
* have the receiver ACK message
* use a 1-bit sequence number to protect for lost messages/acks

Here is a good/simple explanation of how it can be done in a very simple way : 
http://classes.engineering.wustl.edu/cse473/images/e/e5/Lec09.pdf

When ready I will share - to be seen if we can add it to GRBL on the Arduino, or in the future ARM version. When it is available in GRBL, you also need a Sender aware of it:
* adding the checksums and sequence number
* doing a re-transmit if no ACK
Which GUI team is volunteering ?

gerritv, 2016-10-28:
I think you meant a 1 byte sequence number?
Your source though is only a partial implementation and is really a re-statement of SDLC. You might want to take a look at SDLC protocol for something more complete. E.g. RNR is missing (receiver not ready) which would cover the 'buffer full' scenario.
A gui implementation could discern from what Grbl sends back and determine automagically if the protocol is implemented or not.

I would be interested for Grbl Panel if it is not needed too soon :-)

chamnit, 2016-10-28:
@longtolik : FWIW, the majority of communication problems are from three sources: a poorly implemented GUI streamer, a USB-serial chip that is dropping characters from a buffer overflow, and USB cable noise, particularly when close to the spindle power line. 

I also noticed that you have values with a precision of 7 after the decimal point. You need to reduce this to 4, or at most 5. Machines can't physically resolve sub-micron motions. If you are sending G2/3 arcs with a precision of 7, then you may be exceeding the max 79-80 character limit Grbl accepts for a g-code line.

Please don't discount the fact that there are thousands of machine that run Grbl successfully without issues. Grbl OEMs consistently report that Grbl is never the cause of any communication issues, even when running jobs that are several million lines long, deployed across thousands of their machines.

@Strooom : I agree about adding CRC checks. Unfortunately, the 328p 8-bit processor can't handle both Grbl and computing good CRC calculations at the same time. I have already planned this for the ARM version of Grbl. It'll feature a totally different way of communicating with Grbl and eliminating a lot of current bottlenecks/limitations (though not a problem with most CNC applications) and improving GUI communication even more.

Strooom, 2016-10-28:
@gerritv 
> I think you meant a 1 byte sequence number?

No, take a look in the .pdf, it's with a simple 1-bit SeqNmbr.
I am thinking about overflow, but basically the sender doesn't send a next command, until ACKed by the receiver, and that means the message has been received and removed from RxBuffer.
I'll keep you posted for adding to GRBL Panel
@chamnit 
> Please don't discount the fact that there are thousands of machine that run Grbl successfully without issues

I don't, and I value GRBL a lot. On the other hand, some users may be experienceing communication errors (especially the ones due to RFI, for which there is no exact science to solve them) without knowing. They either don't do 3D milling (which results in large files and statistically more change to hit the error) or they just may think there is an error in their gCode.

About the CRC and ARM : I just bought some Teensy 3.2 boards, powered by an ARM Cortex M4. This has the CRC16/32 supported by HW. I will run my Comms code with CRC32 on that and keep you posted of the progress!

chamnit, 2016-10-28:
@Strooom : Carbide3d does a lot of 3D milling on their Nomad machines, primarily because they come with MeshCAM. The company has two very good electrical engineers that have designed their electronics to be highly resistant to EMI. I often hear users running jobs exceeding several million lines of g-code and completing without a problem.

If EMI is a problem, it really should be addressed at a hardware level whenever possible. Software CRC helps deal with and diagnose an EMI issue, but doesn't solve it. It simply band-aids it. Like I said before, it's on the list of things to do, and I do have an interesting "gap-filler" project in the works to help address this before the ARM version is ready for release.

Strooom, 2016-10-28:
@chamnit OK, Good to know!
I agree about EMI. I have had my learning curve with that as well :-)
* I have my limit switches wired to GND and 5V, and low-pass filters on them. Got the inspiration here : http://www.shapeoko.com/forum/viewtopic.php?f=4&t=1290&start=70
* Then I replaced the mechanical relay (switching the spindle = Makita RT0700) by a solid state Fotek SSR-25 (switching at the AC zero-crossing)
* Separate power-supply for the steppers and Arduino/GRBL
* proper star-based grounding

Still, when milling Aluminium and lubricating with WD40, I do get false limit-switch errors...
Would be helpful if those Carbide3D engineers wrote an inspiring blog-post so we can learn from them :-)

gerritv, 2016-10-28:
@Strooom ok, 1 bit so you can have 1 outstanding message. Personal experience (going back to a failed project I was handed to fix in 1985) leads me to say that creating a comms protocol is not as easy as it sounds, regardless of their impressive math and useful diagrams etc. I would start with an established one and trim it a bit before starting from a blank sheet. E.g. without RNR their protocol has no way to say I am not ready for more. It can only go dark or NACK, both of which are not indicative of what is needed from the sender.

cri-s, 2016-10-30:
First you have to enable parity. This detect errors.
I see it a big fault that CNC controller like grbl don't use the principal error detection method of serial transmission line specially on heavy disturbed environment like CNC.
Second use fec, example one xor after 5 or 10 bytes transmitted. This allows discard the single faulty byte reconstructing it from xor.
It could be made simple and compatible too.
As example having gcode $ with two hex number following or even more redundant $nnxy where n is hex number, x is xor of hex number and y is sequence number or similar.
It should be obvious that controller should get out the number of errors seen, errors recovered, unrecovered ...
As extreme measurement it could send char twices, one time normal and then inverted. As common is 7 bit there should be filtered out nicely and, eventually adding xor send two time inverted. Sending xor two time inverted allows inserting it automatically and transparent.
The inverted chars could even be not follow immediately and phased by 1, 3, 5, 7, 13 byte as example.
There could be control sequences, like 5 times command send with msb set. If 1 or 2 control byte, this are padded.
If 3 control byte 4th contains checksum inverted.
4 control byte, 5th is checksum. This control sequence can be used to send CRC, sequence number, retransmission request, ... And implementation is transparently to existing implementation.

gerritv, 2016-10-30:
There are many existing reliable protocols that can be used, there is no need to invent something that then needs to be debugged. And is unique in the world. I am certain that Sonny will choose something appropriate for the next version, on a chip that has room for it.

cri-s, 2016-10-30:
The examples are protocols from 1971 to 1981 and still in use. Hovewer
parity bit including overall and burst error counter should be included in
all versions without any error or discard of data, just as way to know if
there was errors and how many.
Operating a ladder type plasma cutter with THC, there is no room for < 3v
CPU and communication works only with true rs232 driven from optoisolated
USB rs232 converter just after the laptop.
Arm CPU with MB USB interface and plently ram for buffer, that don't have
problems.
This my 2ct

gerritv, 2016-10-30:
@cri-s I fully agree on the party bit, it seems developers in general don't bother for whatever reason. It adds very little code to handle that, same with framing erros etc. Generally ppl just hope for the best.

chamnit, 2016-10-30:
First, I'm not an electrical engineer and depend on others to help me out. I do know a few but they haven't requested alterations to how things currently operate. 

AFAIK, the USB protocol has integrated CRC of the data packets sent to and from devices. If I'm wrong, please correct me. This would suggest that USB-serial chips also do CRC to and from the host computer. So, data corruption would have to occur between the USB-serial chip and the 328p AVR that Grbl runs on. Usually these are right next to each other, localized shielding and buffering of the data lines should take care of most of EMI issues, which good electronics design usually has. 

FWIW, I haven't changed the serial interface in Grbl, because there hasn't been a real pressing need. Otherwise, I would have heard something from OEMs long ago. That said, data error checking is on the list of things to do, among other things, but I think this are best suited for the ARM version, where CPU, flash, and RAM restrictions are no longer a problem.

cri-s, 2016-10-30:
On serial.c after this code, there should be else command that checks if tx
buffer  is not full and send a char if configured, example CAN, that
clearly signalise this event, in addition to increment one counter.

if (next_head != serial_rx_buffer_tail) {
          serial_rx_buffer[serial_rx_buffer_head] = data;
          serial_rx_buffer_head = next_head;
        }

Same here.
ISR(TIMER1_COMPA_vect)
{
  if (busy) { return; } // The busy-flag is used to avoid reentering this
interrupt

Instead of just returning it should update one saturating 8bit counter or
if you prefer two saturating 8bit counter.

If using echo back feature, tx buffer should be increased at least for
linesize and planner buffer blocks decreased at least for line size
bytecount, planner block buffer is approx 40 byte, otherwise RX buffer
overflow could be possible.

cri-s, 2016-10-30:
To my knowledge, I could be wrong, only usb3 have CRC over payload data,
usb2 don't have it.

Parity bit is supported too on USB .
It cannot correct anything , it just signals the problem. if classical xor
checksum is available, one error byte could be recovered without any
retransmission.
Sometimes just knowing the real problem can help as example to add
ferrites. As example motion controller on really fast PNP need 8 ferrite
over USB cable in order it don't have spurious disconnect or wrong move on
mega controller. It have emi problems, limit switches are disabled on code,
homing needed special code to be repeatable. The user have trayed to
isolate every input and separate grounds and to add rc filter, without
success. However now it work without any issues. Slow or fast rs232 don't
changed anything in this case.
Il 30/ott/2016 21:13 "Cri S" <phone.cri@gmail.com> ha scritto:

> On serial.c after this code, there should be else command that checks if
> tx buffer  is not full and send a char if configured, example CAN, that
> clearly signalise this event, in addition to [...]

109JB, 2016-10-31:
So I am just seeing these responses from the past few days.  I am an amateur programmer, so I tend to think in simple terms.  With all of this talk about error checking, parity bits, CRC, etc., I can't help but wonder why not just use something already built into GRBL to perform error checking.  I have written several test streaming programs that simply compare what the host computer GUI has sent to what GRBL echoes back with the echo function enabled in config.h.  It looks like GRBL only does 2 things before echoing back a response, and those are to change lower case to upper case, and deleting whitespace.  The order of commands is not altered. 

As I have said, I have written several GUI type streaming programs that implement this and simply compare the string sent to the string received. The string sent has the same 2 operations applied (uppercase and delete whitespace). There is a bit of a performance penalty in implementing the echo, but a simple test program I just wrote shows that using this method of error checking works well, and the test program can stream at a rate exceeding 10,000 lines per minute even with echoes turned on.

I personally have run enough testing that I am confident that an Arduino equipped with a 16U2 USB to serial chip with the modified firmware is just fine. I have written several testing programs and I have run tests where I have streamed as much as 200,000,000 lines through a 16U2 equipped UNO without any anomalous behavior, I personally just switched my CH340G equipped NANO for a 16U2 equipped UNO clone and haven't looked back. The UNO clone cost me $4 including shipping, so it wasn't a cost issue.

If you are set on using a CH340G equipped Arduino then I would agree that some type of error checking is needed, but not many people need more than 10,000 lines/min, so I would say a GUI compare of sent to echo should be fine.

arkypita, 2017-11-02:
Hi @109JB (and hi all) 
I have read all your test and result with a lot of interest. Thank you for this hard work.

In the sofware [LaserGRBL](https://github.com/arkypita/LaserGRBL) i have implemented a streaming thread able to send up to 800 gcode lines/sec. https://www.youtube.com/watch?v=pK9GnzNFpHM
This thread is a strict loop that send each gcode line one next to the other without delay (sleep only when character-counting buffer is full).

I have tested it with grbl v1.1f on a [nano clone with CH340](https://user-images.githubusercontent.com/8782035/32319645-93610646-bfbb-11e7-9435-8994a346a986.png), this was my result:
- $C check mode (no motor movement, only TX-RX and formal check)
- 1.25 million of line
- 30 minutes of execution needed (about 700 line/sec)
- no error, no issue, no problems at all
- I have tested also with 3 others type of nano/uno clone without issue.
Sounds good!!! Would you like to test it?

However some LaserGRBL user report some problem of streaming hang-up. 
What is your final consideration about delay between commands? It is really needed? Why? Does they solve the problem or they only reduce it? Did you do some specific test about cables quality?

109JB, 2017-11-02:
@arkypita - A couple things.  

1. If you are only relying on Grbl to respond with "ok" as verification that a line was sent properly then you may have errors going on that you don't know about.  The ch340 transmission errors are insidious and can result in a valid g-code line received by Grbl even though it was not exactly as sent.  I now use the simple send-response protocol with Grbl echoes turned on. My streaming program sends a line of g-code, waits for the echo to return, and then compares the echo to what was sent to make sure it matches.

2. The ch340 errors are also not predictable. sometimes the errors would appear almost immediately, and other times it would not appear for close to a million lines into the run. 

3. As far as the delay between sending commands, I don't think this is an issue and it did not stop the errors from occurring.  I believe that the error has to do with timing when the ch340 is sending to Grbl and also receiving from the GUI.

arkypita, 2017-11-02:
Thank you @109JB for your suggestion.

Use simple send response + echo check is the most safe way to prevent error (but what do you do if you receive back a non corresponding echo?). However it is not usable for a public use gcode sender, because it is slow but more important because it require to recompile/flash grbl to activate the echo.

One of the LaserGRBL user told me that with sourcerabbit he can stream millions of line without error, while with LaserGRBL he always has errors. The error is a hang in grbl (grbl stop send back "ok" and only respond to ctrl-x).

The difference i see between LaserGRBL and sourcerabbit is that LaserGRBL query for position ("?") at 5Hz and sourcerabbit query at 1Hz. Could be a problem for CH340/grbl?

Also sourcerabbit send a "~" before begin streaming and LaserGRBL does not, but i don't really thing could matter.

chamnit, 2017-11-02:
@arkypita : Laser jobs are usually very demanding on Grbl, particularly raster jobs and running at high step rates. A status query at 5Hz is fairly fast. Each query, Grbl will have to take several milliseconds to produce the report and queue for transmit. This can cause communication flow to stutter (stop-start), where Grbl is executing its planner buffer faster than it is receiving new motions from the GUI.

High report rates can also cause systems near the max 30kHz step rate to start to become unstable. The only way to tell is if you test with slower refresh rates.

BTW, it's quite possible that the CH340 firmware has been update to address this problem since we first noticed it, which was couple years ago. Newer Arduinos with a CH340 and possible firmware update may no longer be effected.

arkypita, 2017-11-02:
You are a genius @chamnit: with few words you drive me in the right direction!

I discover that, that user, is using a very high step/mm rate (320 step/mm) in $100,101,102. I test with this values and I see that query for report @ 5Hz and stream quickly will cause grbl hang.
Vice versa query at 1Hz or lower, and push some delay between each gcode does not hang grbl.

I'm almost certain that the problem is explained. Tomorrow we will have confirmation with some tests from the user.

arkypita, 2017-11-02:
Sorry if I take advantage of your availability... what is in your opinion the best approach for streamer thread? Sleep (add a delay) or don't sleep between each consecutive code send? What is a good freq for query position?

ps. in my opinion what grbl does is something magical, it only surrend when you push it to the limit... but it's limit is very very high considering the limited HW!

109JB, 2017-11-05:
Sleep between code sends doesn't work, so there is no reason to do it.

arkypita, 2017-11-06:
Dear @109JB @chamnit for your information here is what I have discovered.
The user that report problems with LaserGRBL but can work sourcerabbit ignore that sourcerabbit use simple send-respond instead of character-count as LaserGRBL.

I guess that the causes of hang are due a **very poor quality arduino clone**.
The poor quality clone together with high speed streaming and frequent status request/response cause the character count process misalign. I guess that some character/commands is miss during the streaming/interpretation flow and LaserGRBL think to don't have space while Grbl has space.

I will need to test this hypothesis together with this user by enabling the grbl buffer report.

However guided by this work I took advantage of some improvements aimed at making LaserGRBL more flexible (making the query status and other timings more configurable) and more fast.

The latest version can stream [up to 3000 lines/sec](https://youtu.be/Y4CZTe8XKaM)
I really like what do you think about this performance, compared with other streamer you know

109JB, 2017-11-07:
@arkypita
What you are finding is what I found when I did my testing.  I initially encountered this problem when testing my GUI using the character counting protocol. What happened was exactly as you describe. The streaming would stop at non-repeatable locations in the same g-code file. This was when I started checking the echo responses from Grbl and found that the echoes did not match what was sent. Many times what would happen is the end of one line would get truncated and the EOL character would be lost. This would result in as you described with a misalignment in the character counting protocol.

I tried many many times to correct this, and on the CH340 equipped boards I was never able to eliminate the problem using either character counting, or send-response protocols.  Slower streaming rates don't really solve the problem. I feel with my GUI it was due to the fact that the stream and the status requests happened in different threads, so the timing between status requests and g-code wasn't able to be regulated and the errors could still occur.

As a side note, I don't remember how many tests I ran, but I don't think I had any errors using either protocol on a ch340 board if I turned off status requests completely,  

In the end, I decided to use the 16U2 equipped boards and not look back. I also decided that for my use a send-respone protocol was fast enough. I know that the laser folks need faster streaming than I do. I use Grbl with a milling machine, and milling doesn't require the streaming speeds of the laser stuff.  For this reason, I would highly recommend simply using a 16U2 equipped Uno or clone. I was able to buy 16U2 Uno clones for about $4 each.

arkypita, 2017-11-07:
Thanks a lot for your detailed report.

> I feel with my GUI it was due to the fact that the stream and the status requests happened in different threads, so the timing between status requests and g-code wasn't able to be regulated and the errors could still occur

In LaserGRBL I send GCodes and query for status in the same TX thread. However even doing this from two threads should not be different (unless you have forgotten to synchronize sending on the serial).

The user who is experiencing this issue report me that he can work well with send-response and query @1Hz, but he always has problems with character-count also with low step/mm rate regardless of query status. I think this can only limit the problem, almost completely, but may occur sporadically again.

Finally the user decide to buy an original Arduino UNO from a local store and problems now disappear. I ask him to send me the buggy board, so I can test it well.

109JB, 2017-11-07:
In my gui there is a timer thread which sends the "?" status query at the selected time interval. This timer is its own thread and is separate from the G-code sending thread, but both threads use the same serial port, so the serial port control for this single serial port handles the queuing and synchronizing of sending to grbl.

What I am saying is that since the status request and the g-code sending are from different threads the timing of each send can and does vary.  Even if I were to put a delay in the g-code sending part, of my GUI, the status timer could send a request from its thread.  So you could have a time gap between sends, or possibly no time gap between send. For example if the serial port is in the process of sending a line of g-code which take say 10mSec, and the status report thread triggers a request during that time, the status request will send immediately after the g-code line is finished being sent with essentially no time gap.  Sounds like you have your GUI set up different, but this is how mine works.

We don't know exactly what causes the problem, but something like the above scenario is one of the possibilities that was discussed.  Also, while all of this sending is going on Grbl is sending back responses to the GUI. All of this has to go through the CH340 chip, so it could also be that the send and receive of the CH340 is experiencing some kind of "collision" that it can't handle.  No way to know since the firmware for the CH340 isn't available to examine.  

I tried so many things related to this because of the prevalence of the CH340 chips on the clone boards, but one thing is certain to me. It is a problem with the CH340, not the 328P or Grbl. Here are some of the things I tried:

1. CH340: Delay between sending (only in my g-code thread) - Didn't work
2. CH340: Tried just about every baud rate there is - Didn't work
3. 16U2: Tried clones with 16U2 @ 115200 baud without firmware mod -  Works fine no errors encountered
4. 16U2 Tried clone with 16U2 @ 250000 baud without firmware mod - Very rare, but still ahd a few errors
5. 16U2: Tried clone board with 16u2 @ other baud with firmware mod - works fine no errors encountered

Desuza, 2017-12-16:
People what this error, and how to fix it? 
In file included from C:\Users\acc\Documents\Arduino\libraries\grbl/config.h:30:0,

                 from C:\Users\acc\AppData\Local\Temp\arduino_modified_sketch_71775\grblUpload.ino:1:

C:\Users\acc\Documents\Arduino\libraries\grbl/grbl.h:68:4: error: #error "Required HOMING_CYCLE_0 not defined."

   #error "Required HOMING_CYCLE_0 not defined."

    ^

C:\Users\acc\Documents\Arduino\libraries\grbl/grbl.h:108:4: error: #error "WCO refresh must be greater than one."

   #error "WCO refresh must be greater than one."

    ^

C:\Users\acc\Documents\Arduino\libraries\grbl/grbl.h:111:4: error: #error "Override refresh must be greater than zero."

   #error "Override refresh must be greater than zero."

    ^

Multiple libraries were found for "config.h"
 Used: C:\Users\acc\Documents\Arduino\libraries\grbl
 Not used: C:\Program Files (x86)\Arduino\libraries\grbl
 Not used: C:\Program Files (x86)\Arduino\libraries\grbl
 Not used: C:\Program Files (x86)\Arduino\libraries\grbl
 Not used: C:\Program Files (x86)\Arduino\libraries\grbl
exit status 1
Error compiling for board Arduino/Genuino Uno.

109JB, 2017-12-19:
@Desuza 

What you are posting about has absolutely nothing to do with the thread that you posted it in.

[11 more replies]