grbl/grbl #846: RUN TIME STEPPER MOTOR ALWAYS VIBRATING..

sebindavis, 2015-11-14:
Sir,
	my home maid cnc machine Gcode read but stepper motor not properly rotate sometimes motor vibrate ,i am use easydriverv4.4 use for driver ic ...some time motor rotate only forward direction.am change so many grbl version  but all the time has same effect ,i think that MY Driver is bad, i also change driver motor vice versa but no improvement .could u help me

chamnit, 2015-11-14:
@sebindavis : It sounds like either your motors are not receiving enough power and/or your Grbl settings need to be lowered. Particularly acceleration and max speed.

sebindavis, 2015-11-16:
sir how to improve motor power?....my project is to make cheap cost
3Dprinter also my aim make pcb machine .....we study so many design and
deferent software........and we are make 3D print using "reprap"
technic.....but hardware to expensive so we plan to reduce cost of over all
system frame make wood ,electronic use arduino uno board , software
grbl -simple method .....software side and frame we are ok now....main
trouble we are facing design cheap cost stepper motor driver,but some
electronics item not available in my country........wht i do sir?
😅

closed 2016-01-11 by chamnit