grbl/grbl #847: Hardware or Software issue?

vlachoudis, 2015-11-14:
Hi, I recently encounter the following problem twice. For some strange reason the '?' forces a reset. I am using the latest 0.9j on Arduino UNO with the only change that I've set 12000rpm on the config for the spindle.

Initially I though that for some reason my atmega was fried. So I've changed it with a brand new one, flash it then after a short run I faced the same problem again.

This time I did  a $RST=# it corrected the problem.
I still have the spare ATMEGA328 (without the RST that has the problem)

(output from miniterm.py, with bold my keystokes)
**?**<Idle,MPos:0.000,0.0
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
**?**<Alarm,MPos:0.000,0.
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
**$$**
$0=10 (step pulse, usec)
$1=10 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=1 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=11 (homing dir invert mask:00001011)
$24=25.000 (homing feed, mm/min)
$25=1000.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=2.000 (homing pull-off, mm)
$100=320.000 (x, step/mm)
$101=320.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=0.000 (x max rate, mm/min)
$111=4000.000 (y max rate, mm/min)
$112=1200.000 (z max rate, mm/min)
$120=150.000 (x accel, mm/sec^2)
$121=150.000 (y accel, mm/sec^2)
$122=25.000 (z accel, mm/sec^2)
$130=370.000 (x max travel, mm)
$131=205.000 (y max travel, mm)
$132=100.000 (z max travel, mm)
ok
ok
**?**<Alarm,MPos:0.000,0.
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]

chamnit, 2015-11-14:
Interesting. Since the clear parameters $RST=# command fixed it, it looks like the parameters data was corrupted somehow. I'm not sure why've l would crash with a status query but I suppose it's possible that when applying the offsets for work position may have caused it (serial TX is buffered). Maybe it's read as a NaN or something. 

Did you check the $# printout when this problem happens? My guess would be that it also crashes there.

vlachoudis, 2015-11-14:
You are perfectly right $#  is corrupted and causes as well a reset

?<Alarm,MPos:0.000,0.
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
$#
[G54:-1000276.160,-1000048.000,-68.200]
[G55:-368.000,-203.000,-42.000]
[G56:-358.000,-198.000,-27.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:-335.952,-111.976,-2.000]
[G30:0.000,0.000,0.000]
[G
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]

chamnit, 2015-11-14:
@vlachoudis : G54 does indeed look weird, but I'm surprised that `$#` crashes when it prints G92 or TLO. I'm not sure why it would do this, since both G92 and TLO are in RAM, not EEPROM. 

If you re-wrote the G54 to all zeros, does this fix the crashing problem? This would confirm that the work coordinate systems in EEPROM are getting corrupt somehow due to a way bCNC might be sending the G10 command?

vlachoudis, 2015-11-14:
By reseting the G54 everything works fine.
I've understood the origin of the problem. I was setting the WCS as the lower limit automatically given by the margins of the gcode in bCNC with a function evaluation
`set [xmin] [ymin]`
however when I didn't activate any gcode the margins where invalid and I was setting a crazy value causing the strange behaviour on grbl.
I will protect from the bCNC side, but maybe it would be nice to be protected on grbl.

chamnit, 2015-11-14:
@vlachoudis : I think that's doable, but not sure how to implement it on Grbl's side. Work coordinate offsets are not defined to have a limit. There are instances you can have offsets that exceed the travel of the machine, but the g-code program itself operates within it. This is a rare case though.

I suppose there can be a check for egregious values that are being programmed, but I'm not sure where the cutoff should be or whether it should be configurable.

vlachoudis, 2015-11-14:
At least to avoid the reset. bCNC was triggering 10 times per second the position and I was in an endless loop, resetting immediately.

chamnit, 2015-11-14:
Hmm. I think we would need to isolate what values cause the reset and why. It might the print float function or something else.

vlachoudis, 2015-11-14:
I will try to rest the above valves to see if I can reproduce it

vlachoudis, 2015-11-15:
Indeed by setting this large value it happens again
ok
g10l20p1x1000276.160y1000048.000
ok
?<Idle,MPos:0.000,0.0
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]

vlachoudis, 2015-11-15:
I believe that in the "printFloat()" routine you run out of the buf[10];
long can up to +/-2 147 483 647 which needs: sign + 10 digits + dot + '\0'
you need at least 13 bytes!

vlachoudis, 2015-11-15:
After changing the unsigned char buf[10] to buf[13] it doesn't reset anymore.

chamnit, 2015-11-15:
@vlachoudis : Awesome. Thanks for looking so deep into this. I'll push a fix later this week. Still have some day job stuff that is taking up all of my time.

closed 2020-11-25 by vlachoudis