grbl/grbl #849: svn1829 : no "Ctl::0000bbbb"

LETARTARE, 2015-11-18:
with  svn1829  and Uno328p :
1- $10=127 
2- and config.h -> //#define REPORT_ALL_PIN_STATES 
the response to '?'  -> "< ......,Lim:zyx>"

While in svn1816 had
 "< ......,Lim:zyx,Ctl:0000bbbb>"

I forgot something ?

chamnit, 2015-11-18:
This is fixed in v1.0 edge branch

closed 2015-11-18 by chamnit

LETARTARE, 2015-11-18:
Excuse me insist, I just checked the edge svn1829 is 1.0c.
"< ......,Lim:zyx,Ctl:0000bbbb>" does not appear if config.h -> "// #define REPORT_ALL_PIN_STATES" ,
one simply receives  "< ......,Lim:zyx>".

chamnit, 2015-11-18:
@LETARTARE : You'll need to enable the report by the $10 setting. (Just set $10=255 for now.) It's not written about what bit flags are associated with what yet. IT will be done before official release to master.