grbl/grbl #852: Touch plate error.

candelo, 2015-11-18:
Hi, 
I have an error using GRBL0.9j touch plate function. It´s only go down Z axis 2.978mm always. I have not connected probe pin, the probe pin have 5V.﻿
https://youtu.be/KZSQqkgIHcc

chamnit, 2015-11-18:
@candelo : This post is still active.

Not sure why github doesn't have my response here, but I did write one earlier this morning. Basically this isn't a Grbl problem. It's a Chilipeppr implementation issue, as Grbl doesn't have a touch plate capability in the strictest sense. It only has a probe command that a GUI would use for handling a touch plate. Please see the Linux CNC g-code descriptions on how the G38.x commands work. Perhaps this will help you understand the problem a little more.

candelo, 2015-11-18:
I think that is not a chilipeppr problem, Because I try to send a G38.x
command to GRBL using Universal G-code sender and the error occurs too.
I already checked the voltage on A5 (probe pin) with a Oscilloscope and it
is correct with no fluctuations. I tried too, change A5(probe pin) polarity
and the error is the same.

chamnit, 2015-11-18:
Can you provide a step by step set of g-code to help me reproduce the problem? It's not clear from the video what you are doing exactly.

candelo, 2015-11-18:
For example when I send the G38.2Z-15F10 command,
the Z axis move starts at 10mm/min, but it stops always at 1.889mm from
start point.

chamnit, 2015-11-18:
@candelo : It's still not clear what you are doing. Are you starting from a fresh reboot and your origin is 0,0,0? Does the probe cycle give you an error, report a successful probe by printing out a [PRB:] message, does it just stop, and then allow you do something else? Did you make any alterations from the default build options in Grbl?