grbl/grbl #856: mega 2560 shield, features.

cri-s, 2015-11-24:
I'm making a shield for mega2560, current work in progress is this.

What is missing and need to be added:
- state ample , error/run/idle
- master/slave sync
- watchdog
- spi quadrature encoder, external board, just connector (need to reassign some inputs)
- rpi connector
- debug interface

Now i'm asking if state ample, watchdog, master/slave interface could be interesting in order to integrate it into official branch.
What is the stable version for cartesian motion. 
If someone want pcb, hi have other wishes for adding features or different things that could coexist ?

cri-s, 2015-11-25:
I was not aware that the attached image was not published. Link to it here:
http://s10.postimg.org/tp5lf04ix/mega2560_grbl.png

chamnit, 2015-11-25:
@cri-s :  Thanks but Grbl does not officially support the Mega2560 at this moment. The current Mega pin diagram is subject to change and will likely be altered.

cri-s, 2015-11-25:
The D50-53 input must go to D10-13 for freeing SPI functionality.
It's working in progress.
I have version 9g running with ramps, but need build machine with more
axis and additional
relays, that's why i'm making new shield.

closed 2015-11-25 by cri-s