grbl/grbl #860: help to control Tb6560 4axies with arduino uno

ramihossam, 2015-11-30:
i got drive TB6560 4axis blue board and arduino uno board 
i connect the drive with arduino and start to test X axi 
>>> this is the data sheet for the DB25 Connector
http://postimg.org/image/yi8kwxhyp/
http://postimg.org/image/5mwhwojmx/
i connect DB25 to arduino like this
pin 1 in DB25 to pin 5 in Arduino
pin 16 in DB25 to pin 2 in Arduino
pin 10 in DB 25 to pin 9 in Arduino
pin 4 in DB 25 to pin 8 in Arduino
>>Pin 2 in DB 25 to pin 12,13 in Arduino <---- i dont know what is pin For
Pin 25 in DB 25 to GND in Arduino
and Start to test this Axi X with UniversalGcodeSender-v1.0.7 
with Machine control X- X+ there's no move in stepper motor
when i remove The GND from the Arduino motor move with non Stop like carzy when i press X- its Stop and when i press X+ its move and dont Stop 
im new in this world and sorry for my bad English language :( hope one understand me and help me to solve this Issues

109JB, 2015-12-02:
Try only connecting the following wires first to see if you can get the X-axis working.

DB25 ------- Arduino
Pin 1 -------- Pin 5    (X-direction)
Pin 16 ------ Pin 2    (X-Step)
Pin 4 -------- Pin 8    (X-Enable - optional)
Pin 25 ------ GND    (Ground)

I am almost positive that the TB6560 controllers use a high signal to DIS-able the drives, so you should not really need to connect the enable pin. With only the above 3 or 4 wires connected between the arduino and the TB6560 controller, you should be able to move the X-axis. 

Please report back what happens.

ramihossam, 2015-12-06:
i connected pin 4 to 5 volt from the Power and it Work but with noise in the motor The problem now with Hex File :D

gerritv, 2016-09-04:
You need to find/read the board manual, to reduce/remove motor noise requires changing current and/or micro stepping settings. This has nothing to do with Grbl itself. The blue boards are very common and there should be lots of forums that can help you.

jahnj0584, 2016-09-04:
You need to set $4=1. Kyo made videos about this on youtubes for his Sphinx machine