grbl/grbl #862: Why grbl after completion HOME, machine position is not cleared to 0

viewsat, 2015-12-02:
Why grbl after completion HOME, machine position is not cleared to 0.
After LinuxCNC finish HOME, machine position will be cleared to 0.

chamnit, 2015-12-02:
@viewsat : Grbl sets home to be at the trigger point of the homing switch. Since Grbl moves off by a pull-off distance, your home position will reflect this pull-off distance. Grbl also sets the machine volume in negative space (traditional in production machines), based on max travel settings. 

If you don't like this behavior, you can disable it by uncommenting the HOMING_FORCE_ORIGIN define in config.h and re-compiling Grbl.

closed 2015-12-02 by chamnit

109JB, 2015-12-02:
It is also not necessarily true that LinuxCNC sets the machine position to 0. That is LinuxCNC's default behavior, but it can actually be set to whatever the user desires by changing the "home switch location" and "Home location" parameters in LinuxCNC.

Protoneer, 2015-12-02:
Silly question, but what G-Code would be needed to Zero the machine after the Home Cycle?

chamnit, 2015-12-02:
@Protoneer : The g-code standard doesn't have a "zero machine coordinates" command. Technically. The 3d printing community added a set of them as M commands, I believe. IMHO, it's a risky proposition to give users and GUIs direct access to them, as this can lead to gross misuse, especially when machine coordinates change and work coordinate offsets don't. 

Often you'll see options in a machine setup or controller setup that defines where the machine origin is relative to home as a single startup setting, but almost never something that is a command that makes it easy to alter it. The reason is that once setup, you don't need to change it again. 

Grbl doesn't give you too many options, other than the default behavior or changing it to have home be the origin always. It's been enough up till now. It'll definitely have more options when things move to ARM.